open62541/open62541 #704: Namespace compiler does not create children elements of Datatypes

FlorianPalm, 2016-07-18:
The generated C-Code does not include any children elements of DataType Nodes. For Example, in namespace 0 there are no children nodes of "ServerStatusDataType".

``
  <UADataType NodeId="i=862" BrowseName="ServerStatusDataType">
    <DisplayName>ServerStatusDataType</DisplayName>
    <References>
      <Reference ReferenceType="HasSubtype" IsForward="false">i=22</Reference>
    </References>
    <Definition Name="ServerStatusDataType">
      <Field Name="StartTime" DataType="i=294" />
      <Field Name="CurrentTime" DataType="i=294" />
      <Field Name="State" DataType="i=852" />
      <Field Name="BuildInfo" DataType="i=338" />
      <Field Name="SecondsTillShutdown" DataType="i=7" />
      <Field Name="ShutdownReason" DataType="i=21" />
    </Definition>
  </UADataType>
''

Stasik0, 2016-07-18:
@ichrispa

ichrispa, 2016-08-26:
@FlorianPalm: You are perfectly right. The namespace compiler currently does not handle fields. Only reference based structures are parsed and encoded.

This can be considered a 0.3 target issue.

ichrispa, 2016-09-28:
To elaborate: Exporting fields requires dataType structs to be encoded. It is certainly possible to create your own encodable structs, but at the moment the namespace compiler does not make use of this.

I classified this as a 0.3 feature.

Pro, 2017-09-07:
@FlorianPalm @ichrispa is this still a required feature?
The Fields are used to encode the created object, but they are not explicitly created for the dataType node.

I just checked the Prosys demo server and it results in the exact same data model for `ServerStatusDataType` as it is currently created in #739

FlorianPalm, 2017-09-07:
I am okay with it. Can be closed.

closed 2017-09-07 by Pro