open62541/open62541 #705: Question: What's the difference between UA_DataSource and UA_ValueCallback?

markusgraube, 2016-07-18:
During reading the tutorials I asked myself, when to use `UA_ValueCallback` with `UA_Server_setVariableNode_valueCallback` and when `UA_DataSource` with `UA_Server_addDataSourceVariableNode` for creating a variable node with callbacks. From the type defintions, both look like the same. Both have a handle and a callback for read and write

`examples/server_variable.c` use the first one and `examples/server_datasource.c` the latter one.

Stasik0, 2016-07-18:
these are 2 ways for doing the same `UA_Server_addDataSourceVariableNode ` creates all in one go.

`UA_Server_addVariableNode `+ `UA_Server_setVariableNode_valueCallback` allows to create a variableNode at first (e.g. by a parsed XML nodeset) and add a callback in a later step (in some custom C code).

markusgraube, 2016-07-18:
Okay, I understand.

But is there a reason why `UA_Server_addDataSourceVariableNode` is using  `UA_DataSource` instead of `UA_ValueCallback`? Would it be possible to use the same struct for both methods?

jpfr, 2016-07-18:
> But is there a reason why UA_Server_addDataSourceVariableNode is using UA_DataSource instead of UA_ValueCallback? Would it be possible to use the same struct for both methods?

:+1:

Stasik0, 2016-07-18:
should we break the api? or maybe just define a precompiler alias?

jpfr, 2016-07-18:
Let's not break the API on v0.2 please.

Stasik0, 2016-07-19:
okay, there is a semantic difference - DataSource is called while reading and should manipulate the data, while ValueCallback elements are called after the value has been set. By a coincidence the structs are equal, I'd keep it in 0.2 as is and think about a better naming in 0.3

jpfr, 2016-07-19:
closing the issue here as the question is resolved.
we have the reference in #710, so we won't forget the proposed API change.

closed 2016-07-19 by jpfr