open62541/open62541 #710: API flaws to be fixed in 0.3

Stasik0, 2016-07-19:
* #705
*  UA_Server_addMethodNode needs a possibility to define node ids of "InputArguments" and "OutputArgument" nodes: https://github.com/open62541/open62541/commit/6d23db1acf0bd77416b2bfa68fe8c858f12be5af#diff-3e72cb3692072ede081976a22332e59aR882

HolgerJeromin, 2016-07-19:
You can change the list to a [task list](https://github.com/blog/1375-task-lists-in-gfm-issues-pulls-comments). With this the progress would be visible as a progress bar:
`- [ ] `

jpfr, 2016-07-19:
Change `void * UA_Array_new(size_t size, const UA_DataType *type)` to
```c
UA_StatusCode UA_Array_new(void **outArray, size_t *outSize,
                           UA_Int32 requestedSize, const UA_DataType *type)
```
That lets us handle edge cases, such as NULL arrays, "inside".

jpfr, 2016-08-09:
The Server_addXXXNode function can be simplified/improved.
For example, addObjectTypeNode might take a `subTypeOf` argument instead of parent and parentReference.

uleon, 2016-08-10:
Not exactly an API change, but how about getting rid of urcu with 0.3 ?

Stasik0, 2016-08-10:
@uleon in favor of what?

uleon, 2016-08-10:
The last time I investigated this issue I had the impression that liburcu is used exclusively for the nodestore. I'd suggest to design a nodestore-plugin concept that made it possible to provide architecture/os/environment specific implementations. In that respect this topic would be in fact an API topic, wouldn't it?

jpfr, 2016-08-10:
+1 for nodestore plugins as a **new** public facing API

jpfr, 2016-11-22:
Add a server-wide node creation callback (suggested in https://groups.google.com/forum/#!topic/open62541/wugJCNEu_n8)

jpfr, 2017-01-26:
Add more limits to the server, e.g. maxsubscriptions, a.s.o.

jpfr, 2017-01-27:
Forward the session handle (userdata) with the method callback.

Pro, 2017-09-07:
Related: #1168

ichrispa, 2018-02-01:
So to summarize:
- [ ] Remove liburcu from multithreaded versions
- [ ] Attach sessions to method callbacks
- [ ] Add more limits to the server, e.g. maxsubscriptions
- [ ] Add a server-wide node creation callback

closed 2026-02-06 by andreasebner