open62541/open62541 #712: Generating Namespace 0 does not work

Torben-D, 2016-07-19:
That's the error code:

INFO:__main__:Linking namespace nodes and references
INFO:__main__:Building datatype encoding rules
INFO:__main__:Allocating variables
WARNING:__main__:Can't ignore node, Namespace does currently not contain a node with id i=2292
WARNING:__main__:Can't ignore node, Namespace does currently not contain a node with id i=274
INFO:__main__:Generating Header
Traceback (most recent call last):
  File "/home/xxx/xxx/tools/pyUANamespace/generate_open62541CCode.py", line 176, in <module>
    generatedCode=ns.printOpen62541Header(ignoreNodes, args.suppressedAttributes, outfilename=basename(args.outputFile))
  File "/home/xxx/xxx/tools/pyUANamespace/ua_namespace.py", line 697, in printOpen62541Header
    code = code + n.printOpen62541CCode(unPrintedNodes, unPrintedRefs, supressGenerationOfAttribute=supressGenerationOfAttribute)
  File "/home/xxx/xxx/tools/pyUANamespace/ua_node_types.py", line 679, in printOpen62541CCode
    code = code + codegen.getCreateNodeNoBootstrap(self, parentNode, parentRef, unPrintedNodes)
  File "/home/xxx/xxx/tools/pyUANamespace/open62541_MacroHelper.py", line 152, in getCreateNodeNoBootstrap
    if r.target().value() != None:
AttributeError: opcua_node_object_t instance has no attribute 'value'

Stasik0, 2016-07-22:
for me it works with the current version

ichrispa, 2016-09-28:
Cannot verify :/

@Torben-D: Is this still a problem?

TorbenD, 2016-09-28:
No problem anymore

closed 2016-09-29 by markusgraube