open62541/open62541 #715: UA_Server_addMethodNode is not working when the parent node has a string id

markusgraube, 2016-07-19:
When you want to add a method node with UA_Server_addMethodNode below a parent node that has a string node id, the method node won't be created.

See also my minimal example under: https://github.com/markusgraube/open62541/blob/issue_add_method_string_id_parent/examples/server_method.c

The first call in line 86 does not work since it uses a string node id as parent.
The second call in line 94 works with a numeric node id for the parent.

jpfr, 2016-07-20:
Is this separate from #717?

markusgraube, 2016-07-20:
@jpfr Yes, it is.
#715 is about the c function in open62541
#717 is about the namespace compiler

jpfr, 2016-07-20:
fixed in a4bead3

closed 2016-07-20 by jpfr

markusgraube, 2016-07-20:
Perfect. Works for me.
@jpfr Thank you.

commit a4bead3e (2016-07-20): fix #715
commit f9c5ef96 (2016-08-01): fix #715