open62541/open62541 #716: Namespace Compiler: Duplicate references

markusgraube, 2016-07-19:
I really like the namespace compiler :)
Here is a small flaw.

We have a simple node xml: two objects which have both a third object as component.
``` xml
<?xml version="1.0"?>
<UANodeSet xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns="http://opcfoundation.org/UA/2011/03/UANodeSet.xsd">
  <NamespaceUris>
    <Uri>http://test.test</Uri>
  </NamespaceUris>
  <Aliases>
    <Alias Alias="HasComponent">i=47</Alias>
  </Aliases>
  <UAObject NodeId="ns=1;i=1" BrowseName="Node_1">
    <DisplayName>Node_1</DisplayName>
    <References>
      <Reference ReferenceType="HasComponent" IsForward="false">i=85</Reference>
    </References>
  </UAObject>
  <UAObject NodeId="ns=1;i=2" BrowseName="Node_2">
    <DisplayName>Node_2</DisplayName>
    <References>
      <Reference ReferenceType="HasComponent" IsForward="false">i=85</Reference>
    </References>
  </UAObject>
  <UAObject NodeId="ns=1;i=3" BrowseName="Node_3">
    <DisplayName>Node_3</DisplayName>
    <References>
      <Reference ReferenceType="HasComponent" IsForward="false">ns=1;i=1</Reference>
      <Reference ReferenceType="HasComponent" IsForward="false">ns=1;i=2</Reference>
    </References>
  </UAObject>
</UANodeSet>
```

The namespace compiler will generate following code
``` c
/* WARNING: This is a generated file.
 * Any manual changes will be overwritten.

 */
#include "nodeset.h"
UA_INLINE void nodeset(UA_Server *server) {
UA_Server_addNamespace(server, "http://test.test");

do {
// Referencing node found and declared as parent: i=85/Objects using i=47/HasComponent
// Node: opcua_node_object_t(ns=2;i=1), Node_1
UA_ObjectAttributes attr;
UA_ObjectAttributes_init(&attr);
attr.displayName = UA_LOCALIZEDTEXT("", "Node_1");
attr.description = UA_LOCALIZEDTEXT("", "");
UA_NodeId nodeId = UA_NODEID_NUMERIC(2, 1);
UA_NodeId typeDefinition = UA_NODEID_NULL;
UA_NodeId parentNodeId = UA_NODEID_NUMERIC(0, 85);
UA_NodeId parentReferenceNodeId = UA_NODEID_NUMERIC(0, 47);
UA_QualifiedName nodeName = UA_QUALIFIEDNAME(0, "Node_1");
UA_Server_addObjectNode(server, nodeId, parentNodeId, parentReferenceNodeId, nodeName
       , typeDefinition
       , attr, NULL, NULL);
} while(0);

do {
// Referencing node found and declared as parent: i=85/Objects using i=47/HasComponent
// Node: opcua_node_object_t(ns=2;i=2), Node_2
UA_ObjectAttributes attr;
UA_ObjectAttributes_init(&attr);
attr.displayName = UA_LOCALIZEDTEXT("", "Node_2");
attr.description = UA_LOCALIZEDTEXT("", "");
UA_NodeId nodeId = UA_NODEID_NUMERIC(2, 2);
UA_NodeId typeDefinition = UA_NODEID_NULL;
UA_NodeId parentNodeId = UA_NODEID_NUMERIC(0, 85);
UA_NodeId parentReferenceNodeId = UA_NODEID_NUMERIC(0, 47);
UA_QualifiedName nodeName = UA_QUALIFIEDNAME(0, "Node_2");
UA_Server_addObjectNode(server, nodeId, parentNodeId, parentReferenceNodeId, nodeName
       , typeDefinition
       , attr, NULL, NULL);
} while(0);

do {
// Referencing node found and declared as parent: ns=2;i=1/Node_1 using i=47/HasComponent
// Node: opcua_node_object_t(ns=2;i=3), Node_3
UA_ObjectAttributes attr;
UA_ObjectAttributes_init(&attr);
attr.displayName = UA_LOCALIZEDTEXT("", "Node_3");
attr.description = UA_LOCALIZEDTEXT("", "");
UA_NodeId nodeId = UA_NODEID_NUMERIC(2, 3);
UA_NodeId typeDefinition = UA_NODEID_NULL;
UA_NodeId parentNodeId = UA_NODEID_NUMERIC(2, 1);
UA_NodeId parentReferenceNodeId = UA_NODEID_NUMERIC(0, 47);
UA_QualifiedName nodeName = UA_QUALIFIEDNAME(0, "Node_3");
UA_Server_addObjectNode(server, nodeId, parentNodeId, parentReferenceNodeId, nodeName
       , typeDefinition
       , attr, NULL, NULL);
// This node has the following references that can be created:
//UA_Server_addReference(server, UA_NODEID_NUMERIC(2, 3), UA_NODEID_NUMERIC(0, 47), UA_EXPANDEDNODEID_NUMERIC(2, 2), false);
//  Creating this node has resolved the following open references:
UA_Server_addReference(server, UA_NODEID_NUMERIC(2, 2), UA_NODEID_NUMERIC(0, 47), UA_EXPANDEDNODEID_NUMERIC(2, 3), true);
} while(0);
}
```

This is really cool. However the last command

```
//  Creating this node has resolved the following open references:
UA_Server_addReference(server, UA_NODEID_NUMERIC(2, 2), UA_NODEID_NUMERIC(0, 47), UA_EXPANDEDNODEID_NUMERIC(2, 3), true);
```
seems to be not necessary. It duplicates the reference.

ichrispa, 2016-07-26:
It is quite possible that the problem is that the node actually ends up having a duplicate reference.

You could run the compiler sans the printer interactively and use ns.getNodeById() to take a closer look at the references?

You pretty much do this:
`$ python
>>> from ua_namespace *
>>> ns = opcua_namespace("testing")
>>> ns.parseXML("../schema/namespace0/Opc.Ua.NodeSet2.xml")
>>> ns.linkOpenPointers()
>>> ns.sanitize()
>>> ns.getNodeByBrowseName("...")
`

markusgraube, 2016-07-26:
Hi,

I just had another look on it.
[See gist](https://gist.github.com/markusgraube/fdbc96b7941744afbd8fb89aaf1e025e) for an updated server_example.xml and the resulting nodeset.c.

The output of the namespace compiler seems totally correct to me. Even, if you put the code for example in examples/server_firstSteps.c, there are still all defined references duplicated in the server.

So, it seems to be not a failure of the namespace compiler but of the high level API.

Is this somehow related with #718?

Stasik0, 2016-07-26:
@markusgraube are "HasEffect" relations doubled?

Stasik0, 2016-07-26:
maybe the namespace compiler has to notice when it actually uses high-level api, and mark relations that are implicitly created by the high level api as "created"
or is it already inside @ichrispa, @markusgraube?

markusgraube, 2016-07-26:
@Stasik0 yes, the "HasEffect" relations are doubled.

Okay, the problem is that the reverse references are already be created internally by the high level api:
```
UA_Server_addReference(server, UA_NODEID_NUMERIC(2, 3), UA_NODEID_NUMERIC(0, 54), UA_EXPANDEDNODEID_NUMERIC(2, 2), false);
```
So, it should be simple to remove those lines.

> maybe the namespace compiler has to notice when it actually uses high-level api

Another reason for `--high-level-api` option from #729

Stasik0, 2016-07-26:
will it be handled automatically by #729?

markusgraube, 2016-07-26:
@Stasik0 I just added it with  	aed1af0. Should work for #725, too.

Stasik0, 2016-07-26:
i think it depends on whether we use internal or external api

Pro, 2017-09-07:
This is also fixed in #739 

It generates the following code:

```
/* WARNING: This is a generated file.
 * Any manual changes will be overwritten. */

#include "example_namespace.h"

UA_StatusCode example_namespace(UA_Server *server) {  // NOLINT

UA_StatusCode retVal = UA_STATUSCODE_GOOD;

/* Use namespace ids generated by the server */
UA_UInt16 ns0 = UA_Server_addNamespace(server, "http://opcfoundation.org/UA/");
UA_UInt16 ns1 = UA_Server_addNamespace(server, "http://yourorganisation.org/test/");

/* Node_1 - ns=1;i=1 */
{
UA_ObjectAttributes attr = UA_ObjectAttributes_default;
attr.displayName = UA_LOCALIZEDTEXT("", "Node_1");
attr.description = UA_LOCALIZEDTEXT("", "");
attr.writeMask = 0;
attr.userWriteMask = 0;
retVal |= UA_Server_addObjectNode(server,
UA_NODEID_NUMERIC(ns1,1),
UA_NODEID_NUMERIC(ns0,85),
UA_NODEID_NUMERIC(ns0,35),
UA_QUALIFIEDNAME(ns0, "Node_1"),
UA_NODEID_NULL,
attr,
NULL, NULL);
}

/* Node_2 - ns=1;i=2 */
{
UA_ObjectAttributes attr = UA_ObjectAttributes_default;
attr.displayName = UA_LOCALIZEDTEXT("", "Node_2");
attr.description = UA_LOCALIZEDTEXT("", "");
attr.writeMask = 0;
attr.userWriteMask = 0;
retVal |= UA_Server_addObjectNode(server,
UA_NODEID_NUMERIC(ns1,2),
UA_NODEID_NUMERIC(ns0,85),
UA_NODEID_NUMERIC(ns0,35),
UA_QUALIFIEDNAME(ns0, "Node_2"),
UA_NODEID_NULL,
attr,
NULL, NULL);
}

/* Node_3 - ns=1;i=3 */
{
UA_ObjectAttributes attr = UA_ObjectAttributes_default;
attr.displayName = UA_LOCALIZEDTEXT("", "Node_3");
attr.description = UA_LOCALIZEDTEXT("", "");
attr.writeMask = 0;
attr.userWriteMask = 0;
retVal |= UA_Server_addObjectNode(server,
UA_NODEID_NUMERIC(ns1,3),
UA_NODEID_NUMERIC(ns0,85),
UA_NODEID_NUMERIC(ns0,35),
UA_QUALIFIEDNAME(ns0, "Node_3"),
UA_NODEID_NULL,
attr,
NULL, NULL);
}

retVal |= UA_Server_addReference(server, UA_NODEID_NUMERIC(ns1,3), UA_NODEID_NUMERIC(ns0,54), UA_EXPANDEDNODEID_NUMERIC(ns1, 1), false);
retVal |= UA_Server_addReference(server, UA_NODEID_NUMERIC(ns1,3), UA_NODEID_NUMERIC(ns0,54), UA_EXPANDEDNODEID_NUMERIC(ns1, 2), false);
return retVal;
} // closing nodeset()

```

In UA Expert it looks like the references are correct:

Node_1:
[image]

Node_2:
[image]

And Node_3 (same for both Node_1 and Node_2):
[image]

@markusgraube if that's not the expected result leave any comment directly in the #739 PR.

closed 2017-09-07 by Pro