open62541/open62541 #717: Namespace compiler: Singlequotes vs. Doublequotes for String NodeIDs

markusgraube, 2016-07-19:
When using string NodeIDs in a node XML, there are some problems with quoting

Following works:
```<UAObject NodeId='ns=1;s="Node1"'... ```

This one does not work:
```<UAObject NodeId="ns=1;s='Node1'"...```

The only difference is the order of using singlequotes and doublequotes. Both are valid XML.

It would be really nice if the namespace compiler could handle both kinds of quoting

erossignon, 2016-07-19:
May be 'ns=1;s=Node1' or "ns=1;s=Node1" should be valid too. Why do we need " or ' around the string portion ? How to escape a " or ' character in a string in this case ?

markusgraube, 2016-07-20:
I would also prefer the idea from @erossignon. :+1: 
Escaping of characters is usually done by `\`
For example:  `'ns=1;s=mom\'s node'`
However, if you have quotes in your identifier, probably you have made something wrong during modeling

As far as I know there is no specification of the serialisation of a NodeId. I have not found any definition in the IEC 62541.
In the xsd (https://opcfoundation.org/UA/2011/03/UANodeSet.xsd) it is just a simple string type
``` xsd
<xs:simpleType name="NodeId">
    <xs:restriction base="xs:string"></xs:restriction>
 </xs:simpleType>
```
So, open62541 should support a wide range of appropriate NodeId serialisations.

markusgraube, 2016-07-20:
Perhaps we could use following regular expression
```
import re
re.compile("^(ns=(\d+);)?(i=(\d+)|s=(\w+)|s="([^"]+)"|s='([^']+)'|g=([0-9A-Fa-f]{8}-[0-9A-Fa-f]{4}-[0-9A-Fa-f]{4}-[0-9A-Fa-f]{4}-[0-9A-Fa-f]{12})|b=(.+))$")
```

ichrispa, 2016-07-26:
@markusgraube: You could easily fix this problem using the preprocessor stage.

A quick-and-dirty way would be to stuff this into doc.finalize(). There's already some code there that removes <uax:String> - like namespace qualifiers from value type fields. So one could easily "harmonize" the NodeId string format here using regexps prior to printing the line into the temp doc. 

A more structured approach would be to create a document wide "preprocessor plugin" mechanism which let's us add preprocessing steps to documents in a more universal manner... just a thought though; I definetely don't have any time for it.

Beste regards,

ichrispa

Pro, 2017-09-07:
This is now fixed in #739 
The correct way to go is to use no quotes at all:
`ns=1;s=SomeString`

If there are quotes in the string part, these will then also be part of the final node id (it could be that this is wanted) so no escaping needed.

`NodeId="ns=1;s='Befuellen2'"` 
Results in:
* ns = 1
* s = 'Befuellen2'

`NodeId='ns=1;s="Befuellen3"'`
Results in:
* ns = 1
* s = "Befuellen2"

Any comments on this? Please react directly within #739 to keep the comments collected at one place.

closed 2017-09-07 by Pro