open62541/open62541 #718: Namespace Compiler: Create only one HasTypeDefinition

markusgraube, 2016-07-20:
Currently, the namespace compiler creates for every node a reference `HasTypeDefinition` to `BaseObjectType`.
If there is another HasTypeDefintion in the nodeXML, this reference is appended.

You can see this for example in `server_nodeset` when looking at `testInstance`

However, the spec says for objects (and accordingly for the other nodetypes)
> The HasTypeDefinition Reference points to the type definition of the Object.
Each Object shall have exactly one type definition and therefore be the
SourceNode of exactly one HasTypeDefinition Reference pointing to an
ObjectType .

So, the namespace compiler should omit the `BaseObjectType` and only use the last `HasTypeDefinition` reference.

Stasik0, 2016-07-21:
the problem is in this line: 

https://github.com/open62541/open62541/blob/0.2/tools/pyUANamespace/open62541_MacroHelper.py#L213

a node is cerated without a typedefinition which automatically creates a BaseObjectType definition, a second reference is created afterwards by reference handling rules

Stasik0, 2016-07-21:
I got a fix, the problem is that addVariableNode or addObjectNode methods automatically generate a reference to the most basic object e.g. BaseObjectType since objectType is set to Null. Two solutions are possible
* Set object type accordingly: in this case, however children nodes will be automatically generated with random ids (API flaw #710?)
* Remove the automatically generated references immediately after node creation (this is what I did)

Still fixing the namespace compiler is like http://i.imgur.com/C2NyP.gif

commit b9ff003f (2016-07-21): fixes #718 by removing automatically generated HasTypeReference references
closed 2016-07-21 by Stasik0