open62541/open62541 #728: UAExpert-Message: "Browsing binary DataTypeSystem failed"

NetoLe, 2016-07-25:
I got the mentioned message in my UAExpert (1.4.2), when connecting to a open62541-Server that is inhabited in an embedded device. Besides that the server is working without problems.

In Wireshark there is a browse Request for a Node with the numeric identifier 93. In the XML of the opc foundation it is called: "OPC Binary - A type system which uses OPC binary schema to describe the encoding of data types."

the browse response gives a 0x80340000 (BadNodeIdUnknown), well yeah, because there is no node with Id 93 in there. 
Is this a normal behavior? I have never seen that before with the open62541-Stack.

Stasik0, 2016-07-25:
this seems to be something new, maybe some check in a new version of
UAExpert, afaik CTT does not require this node to exsit.

Pro, 2018-02-01:
The `OPCBinarySchema_TypeSystem` will be added if `UA_ENABLE_FULL_NS0` is set to ON. So this issue should be fixed

closed 2018-02-01 by Pro