open62541/open62541 #731: Reading a value disconnects the client

SvenManier, 2016-07-27:
Hello again,
this is the code im using the read the.answer from your example static-server. 
Before UA_Client_Service_read everything is fine (even subscriptions are possible) but the read-method disconnects me from the server. I get the [0x80ae0000 // The network connection has been closed.] as response (which is empty)

```
UA_Client *c = UA_Client_new(UA_ClientConfig_standard);
	UA_Client_connect(c, "opc.tcp://172.19.7.204:16664");

	UA_ReadRequest rReq;
	UA_ReadRequest_init(&rReq);
	rReq.nodesToRead = (UA_ReadValueId*) UA_Array_new(1, &UA_TYPES[UA_TYPES_READVALUEID]);
	rReq.nodesToReadSize = 1;

	/* define the node and attribute to be read */
	rReq.nodesToRead[0].nodeId = UA_NODEID_STRING_ALLOC(1, "the.answer");
	rReq.nodesToRead[0].attributeId = UA_ATTRIBUTEID_VALUE;

	/* call the service and print the result */
	UA_ReadResponse rResp = UA_Client_Service_read(c, rReq);
	if (rResp.responseHeader.serviceResult == UA_STATUSCODE_GOOD && rResp.resultsSize > 0 && rResp.results[0].hasValue && UA_Variant_isScalar(&rResp.results[0].value)
			&& rResp.results[0].value.type == &UA_TYPES[UA_TYPES_INT32]) {
		int result = *((int*) rResp.results[0].value.data);
		cout << result << endl;
	}

	UA_ReadRequest_deleteMembers(&rReq);
	UA_ReadResponse_deleteMembers(&rResp);
```

Stasik0, 2016-07-27:
Which server?
Wireshark dump?

SvenManier, 2016-07-27:
Its your C-static_server
wireshark dumb will follow tomorrow

Stasik0, 2016-07-27:
well the other code works with current server https://github.com/open62541/open62541/blob/0.2/examples/client.c#L122

SvenManier, 2016-07-28:
Well, it works now. I dont know why, didnt change anything, just captured some data...
Thanks for your time, maybe it will show again then im prepared

closed 2016-07-28 by Stasik0