open62541/open62541 #733: [Client] Browse / Search  nodes by theire types

maljac, 2016-07-31:
Hello, 

this is not a bug but more a general question how to do this with open62541:

I have several hundred objects that are all in the same server and share the same (custom) object type. How can I find all objects that share the same object type? Think of a SQL query: `SELECT nodeid FROM root_node WHERE object_type = 'ns=foo;i=bar'`.

Is this supported by the standard spec or is it only possible when I create a reference manually (on node instanciation) between the object type and the object?

jpfr, 2016-07-31:
OPC UA defines a `query` service for that purpose. open62541 has not implemented that service so far.

There is a TypeDefinitionOf/HasTypeDefinition reference pair between every object and its type. At least when you provide the type-nodeid during instantiation.

So you can just
- use the `browse` service to get a list of all instances of the object type
- do a second `browse` for the sub-nodes in the relevant folder
- take the intersection of the returned sets of nodeids

jpfr, 2016-08-23:
Please reopen if the question is still unresolved.

closed 2016-08-23 by jpfr