open62541/open62541 #734: Support https for website

HolgerJeromin, 2016-07-31:
https://github.com/blog/2186-https-for-github-pages
But we will loose the cool url after redirect to github pages.

jpfr, 2016-11-18:
I like the cool URL we have. Having https *and* the cool URL would be the best. 😄 
Anyway, the homepage needs work. The releases don't load in Firefox.

Firefox doesn't even attempt to load https://api.github.com/repos/open62541/open62541/releases?per_page=1&page=1 in the background.
Maybe it's a security issues for going to a different domain,

Pro, 2016-11-18:
I'm using Lets Encrypt for my domains and these certificates work very well.
The only "problem" is that you need a crontab somewhere, which renews the certificate every 2 months.
Maybe that's also an option for open62541.org

Otherwise, e.g. GlobalSign provides free certificates for open source:
https://www.globalsign.com/en/ssl/ssl-open-source/

closed 2017-05-20 by jpfr