open62541/open62541 #736: [build problem] source code v0.2

ilike28, 2016-08-02:
Hi ~~

I download the  v 0.2 source code today and I got the following issue during building.
https://github.com/open62541/open62541

Thanks ^^

opcua@opcua-VirtualBox:~/test/open62541-0.2$ cmake -DUA_BUILD_EXAMPLECLIENT=On -DUA_BUILD_EXAMPLESERVER=On -DUA_ENABLE_SUBSCRIPTIONS=ON -DUA_BUILD_EXAMPLE=ON -DUA_ENABLE_GENERATE_NAMESPACE0=ON -DUA_ENABLE_METHODCALLS=ON
-- The C compiler identification is GNU 5.3.1
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found PythonInterp: /usr/bin/python (found version "2.7.11") 
-- Found Git: /usr/bin/git (found version "2.7.4") 
fatal: Not a git repository (or any of the parent directories): .git
-- Git failed (not a repo, or no tags). Build will not contain git revision info.
-- Git version: unknown--git-commit-id-unknown
-- CMAKE_BUILD_TYPE not given; setting to 'Debug'
-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    UA_BUILD_EXAMPLE
    UA_BUILD_EXAMPLECLIENT
    UA_BUILD_EXAMPLESERVER

-- Build files have been written to: /home/opcua/test/open62541-0.2
opcua@opcua-VirtualBox:~/test/open62541-0.2$ make
[  2%] Generating src_generated/ua_nodeids.h
[  5%] Generating src_generated/ua_types_generated.c, src_generated/ua_types_generated.h, src_generated/ua_types_generated_handling.h, src_generated/ua_types_generated_encoding_binary.h
[  7%] Generating src_generated/ua_transport_generated.c, src_generated/ua_transport_generated.h, src_generated/ua_transport_generated_handling.h, src_generated/ua_transport_generated_encoding_binary.h
[ 10%] Generating src_generated/ua_namespaceinit_generated.c, src_generated/ua_namespaceinit_generated.h
INFO:__main__:Preprocessing /home/opcua/test/open62541-0.2/tools/schema/namespace0/Opc.Ua.NodeSet2.xml
INFO:__main__:Parsing /tmp/Opc.Ua.NodeSet2.xml_preProcessed-qlZoRt.xml
INFO:__main__:Linking namespace nodes and references
INFO:__main__:Building datatype encoding rules
INFO:__main__:Allocating variables
WARNING:__main__:Can't ignore node, Namespace does currently not contain a node with id i=2292
WARNING:__main__:Can't ignore node, Namespace does currently not contain a node with id i=274
INFO:__main__:Generating Header
INFO:__main__:Namespace generation code successfully printed
Scanning dependencies of target open62541-object
[ 12%] Building C object CMakeFiles/open62541-object.dir/src/ua_types.c.o
[ 15%] Building C object CMakeFiles/open62541-object.dir/src/ua_types_encoding_binary.c.o
[ 17%] Building C object CMakeFiles/open62541-object.dir/src_generated/ua_types_generated.c.o
[ 20%] Building C object CMakeFiles/open62541-object.dir/src_generated/ua_transport_generated.c.o
[ 23%] Building C object CMakeFiles/open62541-object.dir/src/ua_connection.c.o
[ 25%] Building C object CMakeFiles/open62541-object.dir/src/ua_securechannel.c.o
[ 28%] Building C object CMakeFiles/open62541-object.dir/src/ua_session.c.o
[ 30%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_server.c.o
In file included from /home/opcua/test/open62541-0.2/src/server/ua_server.c:10:0:
/home/opcua/test/open62541-0.2/src_generated/ua_namespaceinit_generated.h:16:23: fatal error: open62541.h: 沒有此一檔案或目錄 (no such file or ...)
compilation terminated.
CMakeFiles/open62541-object.dir/build.make:279: recipe for target 'CMakeFiles/open62541-object.dir/src/server/ua_server.c.o' failed
make[2]: *** [CMakeFiles/open62541-object.dir/src/server/ua_server.c.o] Error 1
CMakeFiles/Makefile2:104: recipe for target 'CMakeFiles/open62541-object.dir/all' failed
make[1]: *** [CMakeFiles/open62541-object.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
opcua@opcua-VirtualBox:~/test/open62541-0.2$

Stasik0, 2016-08-02:
Please turn UA_AMALGAMATION on, this is a known bug and is being fixed in #729

ilike28, 2016-08-02:
Thank you very much ^_^

closed 2016-08-02 by jpfr