open62541/open62541 #738: Server internals availability for userland

cbelyaev, 2016-08-03:
Hello,

There is need for interacting with some server internals in userland:

- UA_Session
- UA_Subscription
- UA_MonitoredItem

In OPC UA.NET stack examples (https://github.com/OPCFoundation/UA-.NET/blob/master/SampleApplications/SampleLibraries/Server/Server/IServerInternal.cs) there is **IServerInternal** with **SessionManager** and **SubscriptionManager**. Maybe this can be starting point...

Stasik0, 2016-08-03:
could you give a more specific example of your use-case? E.g. "I need to now the list of MonitoredItems for a subscriptions"?

jpfr, 2016-08-03:
We are required to implement some of that as part of the CTT validation of the upcoming v0.2.
There will be a node created for every session in a special folder.

We have sessionmanager and subscriptionmanager internally. But they are not exposed to users atm.

cbelyaev, 2016-08-03:
For `UA_Session` I need to know the list of sessions and `clientDescription`, `sessionName` of every session.
For any `UA_VariableNode` I need to know `subscription` (for `state`), `monitoredItemType`, `samplingInterval` of every `UA_MonitoredItem` linked to variable.

If you show me the direction I'd love was involved in the implementation.

jpfr, 2016-08-09:
> If you show me the direction I'd love was involved in the implementation.

Sounds great!

So let's evaluate the possible options. At which point do you need that information?
We can either implement a function that lets users retrieve that information.
Or a callback mechanism that is executed whenever there are relevant changes.

cbelyaev, 2016-08-09:
And what about 'unhide' UA_SessionManager, UA_Subscription etc in amalgamated header?

jpfr, 2016-08-09:
> And what about 'unhide' UA_SessionManager, UA_Subscription etc in amalgamated header?

Then we would have to unhide `ua_server_internal.h` so one can find the sessionManager struct in the server.

But those are APIs we don't want to make public.
- They are not stable (we want to be able to change internals **and** have a stable public API)
- They are not as polished as the public API
- The public API should be "complete" as in "it covers all sensible use cases"

So you can
- Use internal headers as a quick fix. But expect some breakage of the API over time.
- Add some small functions that expose these internals (PRs are welcome)
- Start implementing **correct** rights management. We'd eventually like to be able to register/deregister sessions in userspace. And be able to control the access rights for the current session from userspace. At best with a plugin-struct so that the rights-mgmt is exchangeable. But that will take some time to get right.

cbelyaev, 2016-08-09:
I understand your reasons.

> Use internal headers as a quick fix. But expect some breakage of the API over time.

I am not so much in a hurry.

> Add some small functions that expose these internals (PRs are welcome)

```
UA_StatusCode UA_Server_getSessionNodeIds(UA_Server *server, UA_NodeId **nodeIds, size_t *count);
UA_StatusCode UA_Server_getSessionAppDescr(UA_Server *server, const UA_NodeId *sessionNodeId, UA_ApplicationDescription *copy);
UA_StatusCode UA_Server_getSessionName(UA_Server *server, const UA_NodeId *sessionNodeId, UA_String *copy);
UA_StatusCode UA_Server_getMonitoredItemIds(UA_Server *server, const UA_NodeId *monitoredNodeId, UA_UInt32 attributeID, UA_NodeId **nodeIds, size_t *count)
UA_StatusCode UA_Server_getMonitoredItemSamplingInterval(UA_Server *server, const UA_NodeId *nodeId, UA_Double *samplingInterval);
...UA_Server_getMonitoredItemMonitoringMode...
...UA_Server_getMonitoredItemSubscriptionState...
etc
```
looks ugly :(

maybe `{Session,Subscription,MonitoredItem}{Added,Activated,Closed...}` will be better?

> Start implementing correct rights management...

I don't see any connection between server internals and rights management.

jpfr, 2016-08-09:
> I don't see any connection between server internals and rights management.

Rights management is where userland interacts with sessions.

What you can look into is the `SessionsDiagnosticsSummary` object in the server.
This object will contain an entry for every session that is currently open in the server.
You can find a description in part 4 of the standard. But we haven't come around to implementing it.

GrantEdwards, 2016-12-21:
I need to show a list active connections/sessions on a server diagnostics web page [this is a pretty basic requirement for all the industrial protocols our products support].  Has there been any progress on providing a way for user-application code to find this information?  For me, I would prefer a function that I can call to retrieve a list/array of open connections or sessions.  It would also be nice to have a count of subscribed items for each connection, but I could probably live without that.

Stasik0, 2016-12-21:
For the time being you can use internal headers and dive into internal code. But be aware, that the api is not stable.

closed 2018-02-01 by Pro

Pro, 2018-02-01:
Closed as the proposed solution is to use internal methods

GrantEdwards, 2018-02-01:
Instead of just closing the issue because you don't like the proposed soloution, why don't you explain the proper way to solve the problem without using internal methods?

Pro, 2018-02-01:
@GrantEdwards currently the only way to get the info is by using internal methods and unfortunately there is no plan to implement your requested features in the near future.
As any open source project we can not implement every wish a user has, but are very happy to receive a corresponding pull request or a suggestion to improve it.
To make our developer life easier, we do not want to include every functionality in a public API, that would lead to many API breaks in the future.

I closed the issue since it is quite old and there was a solution proposed to circumvent it.

To still keep track of your requested features, could you please add a new cleaned up issue which lists all the desired features and we can tag it as `help needed` and `good first issue`. So if someone else is interested in implementing it (maybe even you @GrantEdwards?) we can keep track there.