open62541/open62541 #740: [Server] Create a variable type within an open62541 server.

MatthiasSchild, 2016-08-05:
Hello everbody,

I try to write an opc ua server with open62541. Currently I try to add a variable type to the server. It adds the node, but ignores the data type for the variable. Could someone tell me what I did wrong?

```
UA_VariableTypeAttributes attr;
UA_VariableTypeAttributes_init(&attr);
UA_Variant_setScalarCopy(&attr.value, "<unset>", &UA_TYPES[UA_TYPES_STRING]);
attr.description = UA_LOCALIZEDTEXT_ALLOC("en_US", displayName.c_str());
attr.displayName = UA_LOCALIZEDTEXT_ALLOC("en_US", displayName.c_str());
attr.dataType = UA_NODEID_NUMERIC(0, UA_NS0ID_STRING);
UA_NodeId nodeId = UA_NODEID_NUMERIC(ns, id);
UA_QualifiedName qualifiedName = UA_QUALIFIEDNAME_ALLOC(ns, this->qualifiedName.c_str());
UA_NodeId parentNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE);
UA_NodeId parentReferenceNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_HASSUBTYPE);

UA_Server_addVariableTypeNode(server, nodeId, parentNodeId, parentReferenceNodeId, qualifiedName, attr, 0, 0);
```

Language: C++
OS: Windows
IDE: Visual Studio Community 2015

Thank you in advance.

Yours Sincerely,

Matthias

closed 2016-08-05 by MatthiasSchild

Stasik0, 2016-08-05:
The first three lines can also be shortened to:
UA_String value = UA_STRING("<unset>");

MatthiasSchild, 2016-08-05:
@Stasik0 Thank you for this tipp.