open62541/open62541 #741: [question] Can Open62541 SDK ensure the state is consistent with the node class during MonitorItem setting??

ilike28, 2016-08-07:
**I saw there have nodeclass in the  Ua_nodes.c(src/server).**

    switch(node->nodeClass) {
    case UA_NODECLASS_OBJECT:
        break;
    case UA_NODECLASS_METHOD:
        break;
    case UA_NODECLASS_OBJECTTYPE:
        break;
    case UA_NODECLASS_VARIABLE:
    case UA_NODECLASS_VARIABLETYPE: {
        UA_VariableNode *p = (UA_VariableNode*)node;
        if(p->valueSource == UA_VALUESOURCE_VARIANT)
            UA_Variant_deleteMembers(&p->value.variant.value);
        break;
    }
    case UA_NODECLASS_REFERENCETYPE: {
        UA_ReferenceTypeNode *p = (UA_ReferenceTypeNode*)node;
        UA_LocalizedText_deleteMembers(&p->inverseName);
        break;

## But there don't have  nodeclass setting in the ua_client_highlevel_subscriptions.c(src\client)

item.itemToMonitor.nodeId = nodeId;
    item.itemToMonitor.attributeId = attributeID;
    item.monitoringMode = UA_MONITORINGMODE_REPORTING;
    item.requestedParameters.clientHandle = ++(client->monitoredItemHandles);
    item.requestedParameters.samplingInterval = sub->PublishingInterval;
    item.requestedParameters.discardOldest = true;
    item.requestedParameters.queueSize = 1;
    request.itemsToCreate = &item;

How can I ensures the state is consistent with the Server node class?
If I want to monitor Object Type Item.

For example by the OPC-UA Foundation C# SDK

                        MonitoredItem monitoredItem = new MonitoredItem();
                        monitoredItem.StartNodeId = nodes_1[ii];
                        monitoredItem.Handle = controlbinding[ii];

                        //************************
                        **monitoredItem.NodeClass = NodeClass.Object;**
  			//************************

                        monitoredItem.AttributeId = Attributes.EventNotifier;
                        monitoredItem.MonitoringMode = MonitoringMode.Reporting;
                        monitoredItem.SamplingInterval = 0;

jpfr, 2016-08-09:
Short answer. In our implementation, the nodeclass of a node can never change.

I'm not sure why the C# code of the foundation nees that value. Because it is never transmitted in the protocol.
Below is the entirety of the information contained in the request to create a monitoreditem.

Did that answer your question?

```c

typedef struct {
    UA_UInt32 clientHandle;
    UA_Double samplingInterval;
    UA_ExtensionObject filter;
    UA_UInt32 queueSize;
    UA_Boolean discardOldest;
} UA_MonitoringParameters;

typedef enum {
    UA_MONITORINGMODE_DISABLED = 0,
    UA_MONITORINGMODE_SAMPLING = 1,
    UA_MONITORINGMODE_REPORTING = 2
} UA_MonitoringMode;

typedef struct {
    UA_NodeId nodeId;
    UA_UInt32 attributeId;
    UA_String indexRange;
    UA_QualifiedName dataEncoding;
} UA_ReadValueId;

typedef struct {
    UA_ReadValueId itemToMonitor;
    UA_MonitoringMode monitoringMode;
    UA_MonitoringParameters requestedParameters;
} UA_MonitoredItemCreateRequest;
```

ilike28, 2016-08-09:
Thank you ^^

Because the EVENT feature must be setting the nodeclass  when setting monitoritem on the OPC-UA Foundation C# SDK.
I want to use EVENT feature on the open62541.
So , I try to edit SDK by my-self .^^

closed 2016-08-09 by jpfr