open62541/open62541 #743: returned timeout of Server_run_iterate very small

anyn99, 2016-08-09:
Hi there,

I have integrated an open62541 server into an external mainloop, just as in the example server_mainloop.c.
I take the timeout to determine when to call the mainloop again. The issue for me is now, that the returned timeouts are very small, and mostly even 0. Since the timeout is described as maximum possible delay until next _iterate call, this means the server iteration wants to be called without delay!

I tested the returned timeout like this:
`UA_Uint16 timeout = UA_Server_run_iterate(OPCS.Server, OPCS.WaitInternal);
if (timeout < tt)
    {
      tt = timeout;
      DEBUG_TRACE("Timeout decreased to: %u\r\n", tt);
      if (0 == tt) tt = 0xFFFF;
    }`

Here an example with timestamps of the output of an idle server with no connected clients:
(newest message on top)
`2:37:00.748473 Timeout decreased to: 49
2:37:00.748186 Timeout decreased to: 0
2:37:00.747991 Timeout decreased to: 0
2:37:00.747792 Timeout decreased to: 0
2:37:00.747543 Timeout decreased to: 0
2:37:00.747329 Timeout decreased to: 0
2:37:00.746252 Timeout decreased to: 1
2:37:00.745238 Timeout decreased to: 2
2:37:00.744235 Timeout decreased to: 3
2:37:00.743224 Timeout decreased to: 4
2:37:00.742248 Timeout decreased to: 5
2:37:00.741270 Timeout decreased to: 6
2:37:00.740264 Timeout decreased to: 7
2:37:00.739306 Timeout decreased to: 8
2:37:00.738296 Timeout decreased to: 9
2:37:00.737346 Timeout decreased to: 10
2:37:00.736284 Timeout decreased to: 11
2:37:00.735261 Timeout decreased to: 12
2:37:00.734223 Timeout decreased to: 13
2:37:00.733334 Timeout decreased to: 14
2:37:00.732247 Timeout decreased to: 15
2:37:00.731281 Timeout decreased to: 16
2:37:00.730328 Timeout decreased to: 17
2:37:00.729230 Timeout decreased to: 18
2:37:00.728256 Timeout decreased to: 19
2:37:00.727271 Timeout decreased to: 20
2:37:00.726294 Timeout decreased to: 21
2:37:00.725299 Timeout decreased to: 22
2:37:00.724292 Timeout decreased to: 23
2:37:00.723345 Timeout decreased to: 24
2:37:00.722224 Timeout decreased to: 25
2:37:00.721248 Timeout decreased to: 26
2:37:00.720274 Timeout decreased to: 27
2:37:00.719311 Timeout decreased to: 28
2:37:00.718339 Timeout decreased to: 29
2:37:00.717222 Timeout decreased to: 30
2:37:00.716349 Timeout decreased to: 31
2:37:00.715326 Timeout decreased to: 32
2:37:00.714389 Timeout decreased to: 33
2:37:00.713277 Timeout decreased to: 34
2:37:00.712312 Timeout decreased to: 35
2:37:00.711357 Timeout decreased to: 36
2:37:00.710245 Timeout decreased to: 37
2:37:00.709282 Timeout decreased to: 38
2:37:00.708258 Timeout decreased to: 39
2:37:00.707231 Timeout decreased to: 40
2:37:00.706352 Timeout decreased to: 41
2:37:00.705223 Timeout decreased to: 42
2:37:00.704290 Timeout decreased to: 43
2:37:00.703293 Timeout decreased to: 44
2:37:00.702367 Timeout decreased to: 45
2:37:00.701309 Timeout decreased to: 46
2:37:00.700232 Timeout decreased to: 47
2:37:00.699269 Timeout decreased to: 48
2:36:50.748335 Timeout decreased to: 49`

Is this expected behaviour? Am I missing something?

anyn99, 2016-08-09:
... just realized that i forgot to usleep(timeout * 1000)... oh my!

so now this issue boils down to the question: why is the timeout sometimes 0?

0:01:10.752743 Timeout decreased to: 49
0:01:10.752372 Timeout decreased to: 0
0:01:10.752182 Timeout decreased to: 0
0:01:10.751989 Timeout decreased to: 0
0:01:10.751790 Timeout decreased to: 0
0:01:10.748595 Timeout decreased to: 4
0:01:10.735375 Timeout decreased to: 17
0:01:00.752688 Timeout decreased to: 49
0:01:00.752484 Timeout decreased to: 0
0:01:00.752135 Timeout decreased to: 0
0:01:00.751943 Timeout decreased to: 0
0:01:00.751749 Timeout decreased to: 0
0:01:00.751071 Timeout decreased to: 1
0:01:00.748869 Timeout decreased to: 3
0:01:00.736633 Timeout decreased to: 16

jpfr, 2016-08-09:
Difficult to see from here.

Why is you clock running backwards?
And why is it jumping 10 seconds in the middle of the printout?

closed 2016-11-24 by jpfr