open62541/open62541 #744: [build problem] source code v0.2 - sample code [sorry...]

ilike28, 2016-08-09:
sorry >_<"
I change the command...
But It still does not produce the sample app on v0.2 today .

opcua@opcua-VirtualBox:~/new_cc/open62541-0.2$ 
opcua@opcua-VirtualBox:~/new_cc/open62541-0.2$ cmake -DUA_BUILD_EXAMPLECLIENT=On -DUA_BUILD_EXAMPLESERVER=On -DUA_ENABLE_SUBSCRIPTIONS=ON -DUA_BUILD_EXAMPLE=ON 
-- The C compiler identification is GNU 5.3.1
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found PythonInterp: /usr/bin/python (found version "2.7.11") 
-- Found Git: /usr/bin/git (found version "2.7.4") 
fatal: Not a git repository (or any of the parent directories): .git
fatal: Not a git repository (or any of the parent directories): .git
-- Git failed (not a repo, or no tags). Build will not contain git revision info.
-- Git version: commit id unknown
-- CMAKE_BUILD_TYPE not given; setting to 'Debug'
-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    UA_BUILD_EXAMPLE
    UA_BUILD_EXAMPLECLIENT
    UA_BUILD_EXAMPLESERVER

-- Build files have been written to: /home/opcua/new_cc/open62541-0.2
opcua@opcua-VirtualBox:~/new_cc/open62541-0.2$ make
fatal: Not a git repository (or any of the parent directories): .git
fatal: Not a git repository (or any of the parent directories): .git
-- Git failed (not a repo, or no tags). Build will not contain git revision info.
-- Git version: commit id unknown
-- Configuring done
-- Generating done
-- Build files have been written to: /home/opcua/new_cc/open62541-0.2
[  1%] Generating src_generated/ua_nodeids.h
[  3%] Generating src_generated/ua_types_generated.c, src_generated/ua_types_generated.h, src_generated/ua_types_generated_handling.h, src_generated/ua_types_generated_encoding_binary.h
[  5%] Generating src_generated/ua_transport_generated.c, src_generated/ua_transport_generated.h, src_generated/ua_transport_generated_handling.h, src_generated/ua_transport_generated_encoding_binary.h
Scanning dependencies of target open62541-object
[  7%] Building C object CMakeFiles/open62541-object.dir/src/ua_types.c.o
[  8%] Building C object CMakeFiles/open62541-object.dir/src/ua_types_encoding_binary.c.o
[ 10%] Building C object CMakeFiles/open62541-object.dir/src_generated/ua_types_generated.c.o
[ 12%] Building C object CMakeFiles/open62541-object.dir/src_generated/ua_transport_generated.c.o
[ 14%] Building C object CMakeFiles/open62541-object.dir/src/ua_connection.c.o
[ 15%] Building C object CMakeFiles/open62541-object.dir/src/ua_securechannel.c.o
[ 17%] Building C object CMakeFiles/open62541-object.dir/src/ua_session.c.o
[ 19%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_server.c.o
[ 21%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_server_binary.c.o
[ 22%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_nodes.c.o
[ 24%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_server_worker.c.o
[ 26%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_securechannel_manager.c.o
[ 28%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_session_manager.c.o
[ 29%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_services_discovery.c.o
[ 31%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_services_securechannel.c.o
[ 33%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_services_session.c.o
[ 35%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_services_attribute.c.o
[ 36%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_services_nodemanagement.c.o
[ 38%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_services_view.c.o
[ 40%] Building C object CMakeFiles/open62541-object.dir/src/client/ua_client.c.o
[ 42%] Building C object CMakeFiles/open62541-object.dir/src/client/ua_client_highlevel.c.o
[ 43%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_nodestore.c.o
[ 45%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_nodestore_concurrent.c.o
[ 47%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_services_call.c.o
[ 49%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_subscription.c.o
[ 50%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_services_subscription.c.o
[ 52%] Building C object CMakeFiles/open62541-object.dir/src/client/ua_client_highlevel_subscriptions.c.o
[ 54%] Building C object CMakeFiles/open62541-object.dir/plugins/ua_network_tcp.c.o
[ 56%] Building C object CMakeFiles/open62541-object.dir/plugins/ua_clock.c.o
[ 57%] Building C object CMakeFiles/open62541-object.dir/plugins/ua_log_stdout.c.o
[ 59%] Building C object CMakeFiles/open62541-object.dir/plugins/ua_config_standard.c.o
[ 61%] Building C object CMakeFiles/open62541-object.dir/deps/libc_time.c.o
[ 63%] Building C object CMakeFiles/open62541-object.dir/deps/pcg_basic.c.o
[ 63%] Built target open62541-object
Scanning dependencies of target open62541
[ 64%] Linking C static library libopen62541.a
[ 64%] Built target open62541
Scanning dependencies of target server_mainloop
[ 66%] Building C object examples/CMakeFiles/server_mainloop.dir/server_mainloop.c.o
[ 68%] Linking C executable server_mainloop
[ 68%] Built target server_mainloop
Scanning dependencies of target server_variable
[ 70%] Building C object examples/CMakeFiles/server_variable.dir/server_variable.c.o
[ 71%] Linking C executable server_variable
[ 71%] Built target server_variable
Scanning dependencies of target client
[ 73%] Building C object examples/CMakeFiles/client.dir/client.c.o
[ 75%] Linking C executable client
[ 75%] Built target client
Scanning dependencies of target server
[ 77%] Building C object examples/CMakeFiles/server.dir/server.c.o
[ 78%] Linking C executable server
[ 78%] Built target server
Scanning dependencies of target client_firstSteps
[ 80%] Building C object examples/CMakeFiles/client_firstSteps.dir/client_firstSteps.c.o
[ 82%] Linking C executable client_firstSteps
[ 82%] Built target client_firstSteps
Scanning dependencies of target server_method
[ 84%] Building C object examples/CMakeFiles/server_method.dir/server_method.c.o
[ 85%] Linking C executable server_method
[ 85%] Built target server_method
Scanning dependencies of target server_datasource
[ 87%] Building C object examples/CMakeFiles/server_datasource.dir/server_datasource.c.o
[ 89%] Linking C executable server_datasource
[ 89%] Built target server_datasource
Scanning dependencies of target server_firstSteps
[ 91%] Building C object examples/CMakeFiles/server_firstSteps.dir/server_firstSteps.c.o
[ 92%] Linking C executable server_firstSteps
[ 92%] Built target server_firstSteps
Scanning dependencies of target server_repeated_job
[ 94%] Building C object examples/CMakeFiles/server_repeated_job.dir/server_repeated_job.c.o
[ 96%] Linking C executable server_repeated_job
[ 96%] Built target server_repeated_job
Scanning dependencies of target server_readspeed
[ 98%] Building C object examples/CMakeFiles/server_readspeed.dir/server_readspeed.c.o
[100%] Linking C executable server_readspeed
[100%] Built target server_readspeed
opcua@opcua-VirtualBox:~/new_cc/open62541-0.2$

Stasik0, 2016-08-09:
try this one 

cmake -DUA_BUILD_EXAMPLES=ON .

ilike28, 2016-08-09:
Hi @@
I get this log

opcua@opcua-VirtualBox:~/new_cc/open62541-0.2$ make clean
opcua@opcua-VirtualBox:~/new_cc/open62541-0.2$ cmake -DUA_BUILD_EXAMPLES=ON
fatal: Not a git repository (or any of the parent directories): .git
fatal: Not a git repository (or any of the parent directories): .git
-- Git failed (not a repo, or no tags). Build will not contain git revision info.
-- Git version: commit id unknown
-- Configuring done
-- Generating done
-- Build files have been written to: /home/opcua/new_cc/open62541-0.2
opcua@opcua-VirtualBox:~/new_cc/open62541-0.2$ 
opcua@opcua-VirtualBox:~/new_cc/open62541-0.2$ 
opcua@opcua-VirtualBox:~/new_cc/open62541-0.2$ 
opcua@opcua-VirtualBox:~/new_cc/open62541-0.2$ make
[  1%] Generating src_generated/ua_nodeids.h
[  3%] Generating src_generated/ua_types_generated.c, src_generated/ua_types_generated.h, src_generated/ua_types_generated_handling.h, src_generated/ua_types_generated_encoding_binary.h
[  5%] Generating src_generated/ua_transport_generated.c, src_generated/ua_transport_generated.h, src_generated/ua_transport_generated_handling.h, src_generated/ua_transport_generated_encoding_binary.h
Scanning dependencies of target open62541-object
[  7%] Building C object CMakeFiles/open62541-object.dir/src/ua_types.c.o
[  8%] Building C object CMakeFiles/open62541-object.dir/src/ua_types_encoding_binary.c.o
[ 10%] Building C object CMakeFiles/open62541-object.dir/src_generated/ua_types_generated.c.o
[ 12%] Building C object CMakeFiles/open62541-object.dir/src_generated/ua_transport_generated.c.o
[ 14%] Building C object CMakeFiles/open62541-object.dir/src/ua_connection.c.o
[ 15%] Building C object CMakeFiles/open62541-object.dir/src/ua_securechannel.c.o
[ 17%] Building C object CMakeFiles/open62541-object.dir/src/ua_session.c.o
[ 19%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_server.c.o
[ 21%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_server_binary.c.o
[ 22%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_nodes.c.o
[ 24%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_server_worker.c.o
[ 26%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_securechannel_manager.c.o
[ 28%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_session_manager.c.o
[ 29%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_services_discovery.c.o
[ 31%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_services_securechannel.c.o
[ 33%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_services_session.c.o
[ 35%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_services_attribute.c.o
[ 36%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_services_nodemanagement.c.o
[ 38%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_services_view.c.o
[ 40%] Building C object CMakeFiles/open62541-object.dir/src/client/ua_client.c.o
[ 42%] Building C object CMakeFiles/open62541-object.dir/src/client/ua_client_highlevel.c.o
[ 43%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_nodestore.c.o
[ 45%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_nodestore_concurrent.c.o
[ 47%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_services_call.c.o
[ 49%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_subscription.c.o
[ 50%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_services_subscription.c.o
[ 52%] Building C object CMakeFiles/open62541-object.dir/src/client/ua_client_highlevel_subscriptions.c.o
[ 54%] Building C object CMakeFiles/open62541-object.dir/plugins/ua_network_tcp.c.o
[ 56%] Building C object CMakeFiles/open62541-object.dir/plugins/ua_clock.c.o
[ 57%] Building C object CMakeFiles/open62541-object.dir/plugins/ua_log_stdout.c.o
[ 59%] Building C object CMakeFiles/open62541-object.dir/plugins/ua_config_standard.c.o
[ 61%] Building C object CMakeFiles/open62541-object.dir/deps/libc_time.c.o
[ 63%] Building C object CMakeFiles/open62541-object.dir/deps/pcg_basic.c.o
[ 63%] Built target open62541-object
[ 64%] Linking C static library libopen62541.a
[ 64%] Built target open62541
Scanning dependencies of target server_mainloop
[ 66%] Building C object examples/CMakeFiles/server_mainloop.dir/server_mainloop.c.o
[ 68%] Linking C executable server_mainloop
[ 68%] Built target server_mainloop
Scanning dependencies of target server_variable
[ 70%] Building C object examples/CMakeFiles/server_variable.dir/server_variable.c.o
[ 71%] Linking C executable server_variable
[ 71%] Built target server_variable
Scanning dependencies of target client
[ 73%] Building C object examples/CMakeFiles/client.dir/client.c.o
[ 75%] Linking C executable client
[ 75%] Built target client
Scanning dependencies of target server
[ 77%] Building C object examples/CMakeFiles/server.dir/server.c.o
[... 2 lines cut ...]
Scanning dependencies of target client_firstSteps
[ 80%] Building C object examples/CMakeFiles/client_firstSteps.dir/client_firstSteps.c.o
[ 82%] Linking C executable client_firstSteps
[ 82%] Built target client_firstSteps
Scanning dependencies of target server_method
[ 84%] Building C object examples/CMakeFiles/server_method.dir/server_method.c.o
[ 85%] Linking C executable server_method
[ 85%] Built target server_method
Scanning dependencies of target server_datasource
[... cut ...]
[ 92%] Built target server_firstSteps
Scanning dependencies of target server_repeated_job
[ 94%] Building C object examples/CMakeFiles/server_repeated_job.dir/server_repeated_job.c.o
[ 96%] Linking C executable server_repeated_job
[ 96%] Built target server_repeated_job
Scanning dependencies of target server_readspeed
[ 98%] Building C object examples/CMakeFiles/server_readspeed.dir/server_readspeed.c.o
[100%] Linking C executable server_readspeed
[100%] Built target server_readspeed
opcua@opcua-VirtualBox:~/new_cc/open62541-0.2$

ilike28, 2016-08-10:
I saw the server & client app is move to the examples folder?
not in the builder folder.
Do you know what happen @@?

jpfr, 2016-08-10:
We moved it.
That way, we can separate the build machinery of the examples from all the rest.

Are we done here?

closed 2016-08-10 by Stasik0