open62541/open62541 #745: Client memory leak of authenticationToken

kyursen, 2016-08-10:
Valgrind reports memory leaks in client example when it is configured to connect to a sample server provided by OPC Foundation (https://opcfoundation.org/developer-tools/developer-kits-unified-architecture/sample-applications/):

````
==17297== 
==17297== HEAP SUMMARY:
==17297==     in use at exit: 480 bytes in 15 blocks
==17297==   total heap usage: 848 allocs, 833 frees, 4,169,884 bytes allocated
==17297== 
==17297== 32 bytes in 1 blocks are definitely lost in loss record 1 of 11
==17297==    at 0x4C2CC70: calloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==17297==    by 0x407FE6: UA_Array_copy (open62541.c:3845)
==17297==    by 0x407CB9: copyNoInit (open62541.c:3742)
==17297==    by 0x407D5A: UA_copy (open62541.c:3757)
==17297==    by 0x403991: UA_ByteString_copy (open62541.h:2234)
==17297==    by 0x406516: NodeId_copy (open62541.c:3207)
==17297==    by 0x407C40: copyNoInit (open62541.c:3732)
==17297==    by 0x407D5A: UA_copy (open62541.c:3757)
==17297==    by 0x403A10: UA_NodeId_copy (open62541.h:2256)
==17297==    by 0x424DAE: __UA_Client_Service (open62541.c:19940)
==17297==    by 0x40183E: UA_Client_Service_browse (open62541.h:6228)
==17297==    by 0x401F76: main (client.c:75)
==17297== 
==17297== 32 bytes in 1 blocks are definitely lost in loss record 2 of 11
==17297==    at 0x4C2CC70: calloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==17297==    by 0x407FE6: UA_Array_copy (open62541.c:3845)
==17297==    by 0x407CB9: copyNoInit (open62541.c:3742)
==17297==    by 0x407D5A: UA_copy (open62541.c:3757)
==17297==    by 0x403991: UA_ByteString_copy (open62541.h:2234)
==17297==    by 0x406516: NodeId_copy (open62541.c:3207)
==17297==    by 0x407C40: copyNoInit (open62541.c:3732)
==17297==    by 0x407D5A: UA_copy (open62541.c:3757)
==17297==    by 0x403A10: UA_NodeId_copy (open62541.h:2256)
==17297==    by 0x424DAE: __UA_Client_Service (open62541.c:19940)
==17297==    by 0x40507B: UA_Client_Service_browse (open62541.h:6228)
==17297==    by 0x425551: UA_Client_forEachChildNodeCall (open62541.c:20087)
==17297== 
................................................................................
==17297== 
==17297== 128 bytes in 4 blocks are definitely lost in loss record 11 of 11
==17297==    at 0x4C2CC70: calloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==17297==    by 0x407FE6: UA_Array_copy (open62541.c:3845)
==17297==    by 0x407CB9: copyNoInit (open62541.c:3742)
==17297==    by 0x407D5A: UA_copy (open62541.c:3757)
==17297==    by 0x403991: UA_ByteString_copy (open62541.h:2234)
==17297==    by 0x406516: NodeId_copy (open62541.c:3207)
==17297==    by 0x407C40: copyNoInit (open62541.c:3732)
==17297==    by 0x407D5A: UA_copy (open62541.c:3757)
==17297==    by 0x403A10: UA_NodeId_copy (open62541.h:2256)
==17297==    by 0x424DAE: __UA_Client_Service (open62541.c:19940)
==17297==    by 0x404DE3: UA_Client_Service_addNodes (open62541.h:6197)
==17297==    by 0x425FE9: __UA_Client_addNode (open62541.c:20218)
==17297== 
==17297== LEAK SUMMARY:
==17297==    definitely lost: 480 bytes in 15 blocks
==17297==    indirectly lost: 0 bytes in 0 blocks
==17297==      possibly lost: 0 bytes in 0 blocks
==17297==    still reachable: 0 bytes in 0 blocks
==17297==         suppressed: 0 bytes in 0 blocks
==17297== 
==17297== For counts of detected and suppressed errors, rerun with: -v
==17297== ERROR SUMMARY: 11 errors from 11 contexts (suppressed: 0 from 0)
````

It seems like authenticationToken is getting allocated more times than deleted.

We've made some fixes in ua_client.c to eliminate these memory leaks, but I am not sure we did it properly.

kyursen, 2016-08-10:
Here is the fix we have made: https://github.com/kyursen/open62541/commit/02749d6221ee4d78ed266548178bb409f9cfe501

Stasik0, 2016-08-10:
@kyursen could you try my fix please?

kyursen, 2016-08-10:
It is fixed now. Thank you!

closed 2016-08-10 by kyursen