open62541/open62541 #746: UA_Client_connect do not connected

tpiat, 2016-08-11:
UA_StatusCode retval = UA_Client_connect(node.client, (char *)uri.c_str());
returns code 0x80b9000,

uri = opc.tcp://127.0.0.1:4845

I use SIMATIC server.

a437267d13 (21/07/2016)     works 
ca046757e7 (23/07/2016)     and later do not work.

Verion a437267d13 compiled when line 
/* #undef UA_DYNAMIC_LINKING */ uncomment,

Verions ca046757e7 and later not compiled if this line comment, 
and compiled with uncomment line.

Stasik0, 2016-08-11:
@tpiat thank you for testing simatic! Probably *again* some negotiation problems, 

* could you tweak the lines https://github.com/open62541/open62541/blob/0.2/plugins/ua_config_standard.c#L16 and https://github.com/open62541/open62541/blob/0.2/plugins/ua_config_standard.c#L17

e.g. setting maxMessageSize to 65535 and/or maxChunkCount to 1 as it was before the commit.

Could you also capture some traffic so we can see what the server is announcing

tpiat, 2016-08-11:
setting maxMessageSize to 65535 and/or maxChunkCount to 1 as it was before the commit.

Do now works.

But I have problems with copmpiling.
Probably incorrect build.

error: variable 'UA_BYTESTRING_NULL' definition is marked dllimport
& etc.

Stasik0, 2016-08-11:
@tpiat could you still give us some tcp traffc? how do you compile it?

jpfr, 2016-08-11:
Dllimport is related to ms visual studio and your build settings.
Either use static linking or switch compile settings.
You can see the dll import settings in ua_config.h.

Stasik0, 2016-08-11:
@jpfr tpiat is using mingw32

jpfr, 2016-08-11:
Visual studio is overly complicated when it comes to dll building.
You need to (manually) set UA_DYNAMIC_LINKING_EXPORT when building the dll. But not when importing it.

tpiat, 2016-08-11:
I do not build dll.
I include open62541.h and open625641.c to my project.

jpfr, 2016-08-11:
Then just disable dynamic linking in the .h

tpiat, 2016-08-11:
How disable it?

/* #undef UA_DYNAMIC_LINKING */ 
This line is comment and I have 8 errors.
```
error: variable 'UA_BYTESTRING_NULL' definition is marked dllimport
```

Or something else?

Stasik0, 2016-08-11:
try defining UA_EXPORT to empty string

tpiat, 2016-08-11:
Compiled.
But there are error.
If I work under Win32 but do not define UA_DYNAMIC_LINKING, comiler use non win32 section.

```
#if defined(_WIN32) && defined(UA_DYNAMIC_LINKING)
# ifdef UA_DYNAMIC_LINKING_EXPORT /* export dll */
#  ifdef __GNUC__
#   define UA_EXPORT __attribute__ ((dllexport))
#  else
#   define UA_EXPORT __declspec(dllexport)
#  endif
# else /* import dll */
#  ifdef __GNUC__
#   define UA_EXPORT __attribute__ ((dllimport))
#  else
#   define UA_EXPORT __declspec(dllimport)
#  endif
# endif
#else /* non win32 */
# if __GNUC__ || __clang__
#  define UA_EXPORT __attribute__ ((visibility ("default")))
# endif
#endif
#ifndef UA_EXPORT
# define UA_EXPORT /* fallback to default */
#endif
```

Stasik0, 2016-08-11:
so its a logic error, which value of UA_EXPORT worked at the end? 

ths one?

# define UA_EXPORT /* fallback to default */

kyursen, 2016-08-11:
I guess the problem is that `tools/amalgamate.py` (with -DUA_ENABLE_AMALGATION=ON) inserts the following code into open62541.c:

```
#ifndef UA_DYNAMIC_LINKING
# define UA_DYNAMIC_LINKING
#endif
```

While it is likely was assumed to define `UA_DYNAMIC_LINKING_EXPORT` (to force export but not dynamic linking itself):

```
#ifndef UA_DYNAMIC_LINKING_EXPORT
# define UA_DYNAMIC_LINKING_EXPORT
#endif
```

Try to modify `tools/amalgamate.py` like this: https://github.com/kyursen/open62541/commit/fac7a69a9aa9f4819dffe26c6d195404d6e3ef91

tpiat, 2016-08-11:
Yes. Worked.
I wrote about it in first string of previous messge.

And then I begin a long discourse, sorry.

Stasik0, 2016-08-11:
https://github.com/kyursen/open62541/commit/fac7a69a9aa9f4819dffe26c6d195404d6e3ef91 this one?

tpiat, 2016-08-11:
Error in log: GetEndpointRequest failed with statuscode 0x80b90000.

WireShark do not show local tcp traffic.
Can I get local traffic?

Stasik0, 2016-08-11:
https://github.com/open62541/open62541/wiki/Capturing-traffic-with-Wireshark

> If you want to sniff from a a server which is on the same computer (localhost) you have to install Npcap on windows systems (https://wiki.wireshark.org/CaptureSetup/Loopback)

tpiat, 2016-08-11:
> @tpiat could you still give us some tcp traffc? how do you compile it?

Where to send the dump file?

Stasik0, 2016-08-11:
http://www.tinyupload.com/

tpiat, 2016-08-11:
http://s000.tinyupload.com/?file_id=49679280932322873159

closed 2016-08-12 by Stasik0