open62541/open62541 #750: Does anyone have an Linux Debian 7 image laying around? (armhf)

Spezies31, 2016-08-16:
Hello,

after i successfully implemented an OPC UA Server and Client Application with an Windows Server, I am trying again to run OPC UA Application on an Raspberry Pi 3 Model B. Recently i use an Raspbian Jessy Distribution for this. But i have an reasonable suspicion that it would be easier with Debian 7 instead of Raspbian. 

I googled pretty long and found no "normal" Debian 7 Image to flash on my MicroSD Card.

Could anybody tell me if they have an Debian 7 Image laying around? I also take a detailed Tutorial/How-To  Instruction for installing Debian 7. Because in the moment i downloaded a bunch of files but i don't really understand how i set up my raspberry with them because the instructions were not clear for me.

greetings

Spezies31

Stasik0, 2016-08-16:
why not taking raspbian?

just get it, install build-essnetial and cmake and you can compile our stack

Spezies31, 2016-08-17:
First thanks for the reply.

I first wanted to try the stack from Unified Automation and they only distribute an C++ Stack for Debian 7. If i just take it and try to install it i just get the error message that the architecture of my system(armel) does not match the architecture of the stack(armhf).

I already have used the stack from open62541.org but on an Windows Desktop PC i really liked the Unified Automation solution. So i wanted to try it on the Raspberry Pi. 

But apparently it works now after trying it again. I don't really get it why it works. Now i am trying to make C++ Projekts with Geany with the Stack from Unified Automation. I am pretty sure to come back with more questions to ask.^^

greetings 

Spezies31

closed 2016-08-17 by Stasik0