open62541/open62541 #752: Remove `Slow Float Encoding.` warning?

Pro, 2016-08-17:
What's the reason behind the `Slow Float Encoding` warning at https://github.com/open62541/open62541/blob/2fa33020b8e6fbe2a3acbd44b687177b599a9ba3/include/ua_config.h.in#L187

It is always shown when clang is used for build. Is this necessary?

E.g. see:
https://travis-ci.org/open62541/open62541/jobs/152919168

I think this slows down compilation time

jpfr, 2016-08-19:
Clang does not define __FLOAT_WORD_ORDER__. So we cannot reliably detect whether floats can be  memcpy 'd on the network buffer.

Solutions:
- Use a custom program to detect this in cmake. (But not for the single-file release.)
- Make the warning appear in ua_types.c once instead of for every include of the config.

jpfr, 2016-08-22:
fixed #759

closed 2016-08-22 by jpfr