open62541/open62541 #763: OPC-UA for Devices Segmentation Fault

trm1904, 2016-08-23:
Hi,

I want to add an object with the type definition "DeviceType", which is defined in my node2.xml (OPC-UA for Devices). When I ran the code below I get a segmentation fault:

```
// add object node
    UA_ObjectAttributes attr_obj;
    UA_ObjectAttributes_init(&attr_obj);
    attr_obj.displayName = UA_LOCALIZEDTEXT("en_US", "Sensor");
    UA_NodeId objNodeID = UA_NODEID_STRING(1, "VAA.Sensor");
    UA_NodeId parentNodeId = UA_NODEID_NUMERIC(2, 5001);
    UA_NodeId parentReferenceNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES);
    //UA_NodeId objType = UA_NODEID_NULL;
    UA_NodeId objType = UA_NODEID_NUMERIC(2, 1002);
    UA_QualifiedName browseName = UA_QUALIFIEDNAME(1, "Sensor");
    UA_Server_addObjectNode(server, objNodeID, parentNodeId, parentReferenceNodeId,
                              browseName, objType, attr_obj, NULL, NULL);
```

[Opc.Ua.Di.NodeSet2.xml.txt](https://github.com/open62541/open62541/files/432469/Opc.Ua.Di.NodeSet2.xml.txt)
.

If I use the normal object type everything is fine and everything is constructed how it should(checked by UAExpert). I have also tested with an additional node.xml file, where I have added some devices. This also works fine.

[vaa.xml.txt](https://github.com/open62541/open62541/files/432483/vaa.xml.txt)

last commit:
commit f916ba4886b8adf695509e8e7a9e69365cea33eb

jpfr, 2016-08-23:
Hey @trm1904,

please help us help you. :smile: 

Can you
- update to the latest commit in the 0.2 branch and try again
- Show a backtrace of where the crash happens. (Use the Visual Studio Debugger, gdb or similar to see where the crash occurs)

trm1904, 2016-08-23:
with the latest commit 8c4f50e363f2ed4b11d9852c940c44355440622e , I can't build the stack:

> [ 87%] Building C object CMakeFiles/open62541-object.dir/open62541.c.o
/home/trm/Software/open62541/build/open62541.c: In function ‘UA_Server_addExternalNamespace’:
/home/trm/Software/open62541/build/open62541.c:14352:11: error: request for member ‘length’ in something not a structure or union
     if(url.length >= 256)
           ^
/home/trm/Software/open62541/build/open62541.c:14354:26: error: request for member ‘data’ in something not a structure or union
     memcpy(urlString, url.data, url.length);
                          ^
/home/trm/Software/open62541/build/open62541.c:14354:36: error: request for member ‘length’ in something not a structure or union
     memcpy(urlString, url.data, url.length);
                                    ^
/home/trm/Software/open62541/build/open62541.c:14355:18: error: request for member ‘length’ in something not a structure or union
     urlString[url.length] = 0;
                  ^
/home/trm/Software/open62541/build/open62541.c:14365:5: error: implicit declaration of function ‘addNamespaceInternal’ [-Werror=implicit-function-declaration]
     addNamespaceInternal(server, urlString);
     ^
/home/trm/Software/open62541/build/open62541.c: In function ‘Service_AddReferences_single’:
/home/trm/Software/open62541/build/open62541.c:19605:22: error: conversion to ‘UA_StatusCode {aka unsigned int}’ from ‘UA_Int32 {aka int}’ may change the sign of the result [-Werror=sign-conversion]
             retval = ens->addOneWayReference(ens->ensHandle, item);
                      ^
/home/trm/Software/open62541/build/open62541.c:19636:22: error: conversion to ‘UA_StatusCode {aka unsigned int}’ from ‘UA_Int32 {aka int}’ may change the sign of the result [-Werror=sign-conversion]
             retval = ens->addOneWayReference(ens->ensHandle, &secondItem);
                      ^
cc1: all warnings being treated as errors
CMakeFiles/open62541-object.dir/build.make:212: die Regel für Ziel „CMakeFiles/open62541-object.dir/open62541.c.o“ scheiterte
make[2]: *** [CMakeFiles/open62541-object.dir/open62541.c.o] Fehler 1
CMakeFiles/Makefile2:104: die Regel für Ziel „CMakeFiles/open62541-object.dir/all“ scheiterte
make[1]: *** [CMakeFiles/open62541-object.dir/all] Fehler 2
Makefile:83: die Regel für Ziel „all“ scheiterte
make: *** [all] Fehler 2

Stasik0, 2016-08-23:
@trm1904 please disable UA_ENABLE_EXTERNAL_NAMESPACES flag, it is an experimental feature that is broken, but does not affect you

@FlorianPalm @JayGro External Namespaces are broken?

trm1904, 2016-08-23:
gdb says:

Program received signal SIGSEGV, Segmentation fault.
0x0000000000406713 in UA_Array_copy (src=<Fehler beim Lesen der Variable: Cannot access memory at address 0x7fffff7fefd8>, src_size=<Fehler beim Lesen der Variable: Cannot access memory at address 0x7fffff7fefd0>, dst=<Fehler beim Lesen der Variable: Cannot access memory at address 0x7fffff7fefc8>, type=<Fehler beim Lesen der Variable: Cannot access memory at address 0x7fffff7fefc0>) at /home/trm/Software/open62541/build/open62541.c:5176
5176	UA_Array_copy(const void *src, size_t src_size, void **dst, const UA_DataType *type) {

In CLion I have made a screen shot  of the function trace:
[image]

trm1904, 2016-08-23:
I have testing a little bit and choose object types from the opc for device spec, which does not contain "<" or ">" as member and they worked fine. Only object types with members which does contain "<" or ">" still cause a segmentation fault.

trm1904, 2016-08-23:
Another issue is, that the object type of for example BlockType inherits from TopologyElementType. But when I make an object instance from type BlockType only the attributes of BlockType and not from TopologyElementType are considered.

jpfr, 2016-08-23:
Got it.
This issue is now flagged as related to the nodeset compiler and as an issue to be fixed before 0.2.

Don't expect a solution by tomorrow, but this definitely on the radar.

JayGro, 2016-08-24:
Hey @Stasik0 , i just fixed the broken EnableExternalNamespace in Pull Request #765.
Just in case @trm1904 needs it anyway.

ichrispa, 2016-08-26:
@jpfr: Can you elaborate on what you "got" about this and why this a namespace compiler issue?

ichrispa, 2016-08-26:
I inserted a instantiationCallback to follow the instantiation tree and see where the infinite instantiation loop occurs.

The instantiation routine follows the components of DeviceType down to ns2;i=6215 "ImageSet":

Apparently, "<CPIdentifier>" is next, as a ConnectionPointType Object is being instantiated. However at this point, the instantiation only creates more and more folder types. I cannot see a cause for this from the model standpoint, at least not the one that the namespacecompiler created. It seems to be pretty much conformant with what the XML describes.

@jpfr: How did you deduce that there is a model generation error here? Can you determine what causes the infinite Folder-Creation-Loop?

ichrispa, 2016-08-26:
Located it: In the DI Model, the DeviceType is of type ConnectionPointType. The later contains a NetworkAdress, which is also of ConnectionPointType. Hence the instantiation runs in a loop. 

This is a model issue; the spec is pretty clear about loops contained by stuff inheriting from hasChild (includes hasSubType and hasComponent ):

    Starting from Node “A” and only following References of the subtypes of the HasChild ReferenceType shall never be able to return to “A”. 

@jpfr : In my old instantiation code, a visitedTypeNodeList was dragged along to avoid this sort of recursion in invalid models. We could "support" this sort of error by reintroducing it?

Second subissue: There have been multiple tickets reporting that inheritance is not supported by our current stack. This has been pointed out multiple times. I consider the runtime environment responsible for creating attributes/methods of a supertype before creating the type and its subtypes. That is not a namespace compiler issue.

closed 2016-09-09 by ichrispa