open62541/open62541 #769: Wrong NodeIDs offset for BinaryEncoding

Pro, 2016-08-25:
In commit https://github.com/open62541/open62541/commit/952072a3469f12dbb2c406d9dbd8689182fbc2b1 @jpfr  updated the `NodeIds.csv` to the most recent version.

Most of the (old) node IDs look something like this:
```
 FindServersRequest,420,DataType
 FindServersRequest_Encoding_DefaultXml,421,Object
 FindServersRequest_Encoding_DefaultBinary,422,Object
```
So encoding binary is just the main ID+2.
That's also used when constructing the binary message:
https://github.com/open62541/open62541/blob/0.2/include/ua_constants.h#L104
and
https://github.com/open62541/open62541/blob/0.2/src/ua_securechannel.c#L222

Unfortunately this does not work anymore for new types. If you check the new NodeIds.csv, there is for example:
```
ServerOnNetwork,12189,DataType
FindServersOnNetworkRequest,12190,DataType
FindServersOnNetworkResponse,12191,DataType
RegisterServer2Request,12193,DataType
RegisterServer2Response,12194,DataType
ServerOnNetwork_Encoding_DefaultXml,12195,Object
FindServersOnNetworkRequest_Encoding_DefaultXml,12196,Object
FindServersOnNetworkResponse_Encoding_DefaultXml,12197,Object
RegisterServer2Request_Encoding_DefaultXml,12199,Object
RegisterServer2Response_Encoding_DefaultXml,12200,Object
OpcUa_XmlSchema_ServerOnNetwork,12201,Variable
OpcUa_XmlSchema_ServerOnNetwork_DataTypeVersion,12202,Variable
OpcUa_XmlSchema_ServerOnNetwork_DictionaryFragment,12203,Variable
ServerOnNetwork_Encoding_DefaultBinary,12207,Object
FindServersOnNetworkRequest_Encoding_DefaultBinary,12208,Object
FindServersOnNetworkResponse_Encoding_DefaultBinary,12209,Object
RegisterServer2Request_Encoding_DefaultBinary,12211,Object
RegisterServer2Response_Encoding_DefaultBinary,12212,Object
```

Especially look at (https://github.com/open62541/open62541/blob/0.2/tools/schema/NodeIds.csv#L4822)
```
RegisterServer2Request,12193,DataType
RegisterServer2Response,12194,DataType
ServerOnNetwork_Encoding_DefaultXml,12195,Object
```
The node ID 12195 is NOT `RegisterServer2Request_Encoding_DefaultBinary` but `ServerOnNetwork_Encoding_DefaultXml`

So just offsetting the NodeID by 2 to get the ID for Binary Encoding does not work anymore, and wrong node IDs are sent.

Do you have any idea/suggestion how to fix this?

jpfr, 2016-08-25:
Oh boy. Is nothing sacred to the OPC Foundation?
[sarcasm off]

Your PR is the only correct way to fix this.
I close this issue and we continue in #770.

closed 2016-08-25 by jpfr

ichrispa, 2016-09-05:
This is more of a problem then you expect. There are actually parts of the namespace compiler that manually encode data structures to which no DataType has been provided (i.e. the compiler undestands the anatomy of the extensionObject, but the stack doesn't know about it).

The offset has been readded to the generated header... we could add an #undef at the end of the instantiation function if desired.

commit 99522b21 (2016-08-26): Fix #769: Wrong NodeIDs offset for BinaryEncoding (#770)

* Fix #769: Wrong NodeIDs offset for BinaryEncoding

* Avoid double loops for finding correct node id

* Remove findDataType

* Disable xmlEncodingId for now
commit 966e41b9 (2016-08-29): Fix #769: Wrong NodeIDs offset for BinaryEncoding