open62541/open62541 #772: RemoveRepeatedJob call within processRepeatedJob results in invalid memory access

Pro, 2016-08-26:
@jpfr in your recent commit you changed the way how repeated jobs are stored:
https://github.com/open62541/open62541/commit/baa92aea252567e24371512c0a688d1af4f47b21

Especially the following:
https://github.com/open62541/open62541/commit/baa92aea252567e24371512c0a688d1af4f47b21#diff-2c0adffa1ffa262b543b31b8ccec7832R217

The mDNS discovery stuff requires a job which is executed every 10 minutes to register the server. The first call has to be delayed by some milliseconds to wait until the server is fully initialized, especially that the discoveryUrl is set.

I have the following method for that:
`UA_StatusCode UA_Server_addPeriodicServerRegisterJob(UA_Server *server, const UA_UInt32 intervalMs,
const UA_UInt32 delayFirstRegisterMs, UA_Guid* periodicJobId)`
https://github.com/open62541/open62541/blob/feature_mdns/src/server/ua_services_discovery.c#L972

It creates two repeated jobs:
1. The first repeated job with an interval of `delayFirstRegisterMs`. This is a one-shot job and should be deleted after the first processing.
2. The regular 10 Minutes `intervalMs` job with first execution in 10 minutes.

The problem with your changes is the following:
For the `delayFirstRegister` Job, I delete it within the processRepeatedJobs in here:
https://github.com/open62541/open62541/blob/feature_mdns/src/server/ua_services_discovery.c#L925, this means that `rj` is deleted within `processJob` (https://github.com/open62541/open62541/blob/baa92aea252567e24371512c0a688d1af4f47b21/src/server/ua_server_worker.c#L225)
So `rj` is invalid after processJob, which causes the next `LIST_FOREACH_SAFE` call to fail (see valgrind output here: https://travis-ci.org/Pro/open62541/jobs/155271988#L2575)

How can I solve this issue?
- Is there a better way for implementing the One-Shot repeated job? E.g. add an additional variable to RepeatedJob which indicates the number of shots, and then delete the repeatedJob in `processRepeatedJobs`.
- Shouldn't it be possible to remove the repeated job within its `processRepeatedJob` call?

Pro, 2016-08-26:
Also the comment here is then wrong:
https://github.com/open62541/open62541/blob/0.2/include/ua_server.h#L195
```
/* Remove repeated job. The entry will be removed asynchronously during the next
 * iteration of the server main loop.
```

Because the repeated job is removed immediately (at least for the non-multithreaded version).
Or am I wrong?

jpfr, 2016-08-26:
> @jpfr in your recent commit you changed the way how repeated jobs are stored: baa92ae

There were some timing problems with the conformance testing tools. Batching repeated jobs with the same execution interval together is good for performance. But the first delay may be unexpectedly short.

> - Is there a better way for implementing the One-Shot repeated job? E.g. add an additional variable to RepeatedJob which indicates the number of shots, and then delete the repeatedJob in processRepeatedJobs.
- Shouldn't it be possible to remove the repeated job within its processRepeatedJob call? This could be solved by setting a flag in rj and then delete it within processRepeatedJob?

In order to have an event loop also for the client, ua_server_worker should be refactored into a generic implementation that works for server and client. I would add one-shot-jobs once we have that.

Until then, a job that removes itself can be done in very few lines of additional code.
That is now possible. (commit dcf1e52d15b61bde9a0fa0a3f5c262488e49c19a)

> Also the comment here is then wrong: https://github.com/open62541/open62541/blob/0.2/include/ua_server.h#L195

fixed

jpfr, 2016-09-10:
Closing as the main issue is now solved and there is a workaround for the rest.

closed 2016-09-10 by jpfr