open62541/open62541 #775: GetEndpointUrl

AlexSTI, 2016-08-26:
Sorry to disturb again.
I'm engineering sampling our app.
One of our testers underlined a possible issue.
If somebody does GetEndopintUrl without specifiing a "destination" Url the server answers with an empty string.
I've been modified the code (ua_services_discovery.c - line 111 ) as follow:

        // We will give our endpoint descriptor to the requester and not what he sent if it sent 0...
        
        if ( request->endpointUrl.length != 0 )
        {
			UA_String_deleteMembers(&response->endpoints[k].endpointUrl);
			retval = UA_String_copy(&request->endpointUrl, &response->endpoints[k].endpointUrl);
        }

So if the sender didn't specify anything, we will insert our info. Otherwise, everything is as now.
Don't know if this is correct/polite/etc... but it solves...

Hoping to have been helped.

jpfr, 2016-08-30:
Hey @AlexSTI ,

is there a mismatch between what you describe and the code?

> We will give our endpoint descriptor to the requester and not what he sent if it sent 0

Did you mean the following?

```c
if ( request->endpointUrl.length 0= 0 )
```

AlexSTI, 2016-08-30:
No, in effect I ment what I wrote. Infact, if you look at the entire loop code, you can see:

        if(!relevant_endpoints[j])
            continue;
        retval = UA_EndpointDescription_copy(&server->endpointDescriptions[j], &response->endpoints[k]);
        if(retval != UA_STATUSCODE_GOOD)
            break;
        
        // We will give our endpoint descriptor to the requester and not what he sent if it sent 0...
        
        if ( request->endpointUrl.length != 0 )
        {
			UA_String_deleteMembers(&response->endpoints[k].endpointUrl);
			retval = UA_String_copy(&request->endpointUrl, &response->endpoints[k].endpointUrl);
        }
        k++;

So, before, there should be an entire copy of the endpoint from the server and just in case the requester asked something different, I copy what he asked for...

Please tell me if I got too tired...

Pro, 2018-02-05:
I added a unit test checking your described scenario and in current master everything already works as expected.
Probably fixed in e15cdfa

commit a66ebcde (2018-02-05): Add unit test for empty GetEndpoints call

Fixes #775
commit 1cfbcd55 (2018-02-05): Add unit test for empty GetEndpoints call

Fixes #775
closed 2018-02-05 by Pro