open62541/open62541 #780: Empty data when catching UA_ServerStatusDataType

SvenManier, 2016-08-30:
When reading the node 'ServerStatus' my result is slightly empty even though the reading didn't fail.
Client and server are 0.2.
`UA_ServerStatusDataType *serverStatus = (UA_ServerStatusDataType*) readResponse.results[0].value.data;`
I added a picture of the ServerStatusDataType and the ReadResponse. Sorry for the quality, the window closed every time i tried to make a normal screenshot.
[UA_Issue.pdf](https://github.com/open62541/open62541/files/444155/UA_Issue.pdf)

jpfr, 2016-08-30:
Hello @SvenManier ,

reading the serverstatus and buildinfo works fine on my box.
Are you working with a recent commit or v0.2-rc1?

The buildinfo structure you show look freshly initialized (zeroed out).
Can you use wireshark to find out if the empty structure is sent over the network or due to some client error?

https://github.com/open62541/open62541/wiki/Writing-Good-Issue-Reports

SvenManier, 2016-08-30:
Hello,

Im using a recent commit. Here's the wireshark dump

[Server-Capturing.pcapng.zip](https://github.com/open62541/open62541/files/444551/Server-Capturing.pcapng.zip)

jpfr, 2016-09-12:
Hey @SvenManier ,

solved on the 0.2 branch. We just did not copy over the buildinfo content in the callback.

Best, Julius

commit 6a6ebf34 (2016-09-12): fix #780; include complete build info in the server state
closed 2016-09-12 by jpfr