open62541/open62541 #783: C# or VB.NET OPCUA Browser

iquestor, 2016-08-31:
There was discussion in issue #234 ( https://github.com/open62541/open62541/issues/234 )  about using the open62541 library DLL in C# project. I never saw if those changes were included in the libOpen62541.dll and lib. 

I am using the current open62541.dll in my C++ client, but need to write an OPCUA server browser in my .NET configuration software, which is currently VB.NET. I could do either C# or VB.NET. 

Is there an example of how to access the library in either of these, and/or has anyone done so?

ichrispa, 2016-09-02:
Hi @iquestor,

speaking from expirience: wrapping any pure C header file into C# constructs is a pain - even more so for open62541, where lots of headers are (a) huge and (b) autogenerated for the prior reason.

I have two sets of advice for you:
(1) Create a generator script (python or awk) that parses the C files and puts every C definition in C# struct/class tags. (not suggested)
(2) We actually decoupled C# and C using winsocks on an application level, which is far more advisable the solution 1. Determine what C# wants to know from the OPC UA part, then create a small packet library so these queries can be send to an OPC UA Proxy listening on a winsock or unix socket. Let the C Proxy handle the entire OPC UA part, then communicate the necessary info back to C#.

Best regards,

ichrispa

iquestor, 2016-09-02:
Ichrispa

2 sounds reasonable.  I could see how they would work.  I only need recursive node info to fill out a browser tree.  Let me think on it and will report what I'm doing.

Thanks for your reply.

jpfr, 2016-09-03:
Hello @iquestor,

it sounds as if you needed only a fraction of the functionality of open62541.
I would suggest the following to keep complexity to a minimum:

- Build a C++/CLI library that exports only what you need as managed code (that can be called from C#). Internally, the C++ library can use the open62541 library without any constraints.
- Use the the DLLImport function of C# to load functions from the precompiled open62541 library.

I copied the information from the different issues into a github wiki page: https://github.com/open62541/open62541/wiki/Using-open62541-from-C%23

That way, we can keep the information current and in one place.
Thank you for adding you own experience and improvements.
Once we have a wrapper exporting a reasonable portion of functionality, we can think of moving it off the wiki page to a dedicated project in the github open62541 organization.

jpfr, 2016-11-21:
Please reopen if still relevant

closed 2016-11-21 by jpfr