open62541/open62541 #786: Open62541 UA Server with Qt GUI application

jucastel, 2016-08-31:
How to integrate a Open62541 UA Server/Client in Qt GUI application ?

Pro, 2016-08-31:
Do you mean in general?
You can do this as you would include any other C++ library in Qt:
https://doc.qt.io/qtcreator/creator-project-qmake-libraries.html

Then in Qt create a thread which calls `UA_Server_run` or if you want to have more control over the loop something like:
```
    UA_StatusCode retval = UA_Server_run_startup(server);
    if(retval != UA_STATUSCODE_GOOD)
        return retval;
    while(*running)
        UA_Server_run_iterate(server, true);
    return UA_Server_run_shutdown(server);
```

Look into the examples folder for additional examples.

Regarding multithread safety, I'm not completely sure what the status of that is for open62541.
Maybe one of the other project members can tell you, if open62541 is multithread safe.
Especially:
If you have a separate server loop thread, can you call any method on the server from another thread without any issues?

jpfr, 2016-09-03:
An open62541 server can run in a single thread together with a Qt GUI.
The main idea is to call `UA_Server_run_iterate` cyclically in the Qt event loop or to have your own event loop calling both `UA_Server_run_iterate` and `QApplication::processEvents()`.
http://stackoverflow.com/questions/1051333/combing-an-external-event-loop-with-qts

For the client, I would create a separate thread. The current client implementation supports only synchronous service calls. Hence the GUI would block until a response is received (or a timeout occurs).

> If you have a separate server loop thread, can you call any method on the server from another thread without any issues?

If compiled with multithreading enabled, the OPC UA services exposed via userspace functions (such as UA_Server_read) can be called from multiple threads in parallel.

However, even though this is known to work in practice, testing and formal verification of concurrent operation is incomplete. In production, you will be better off protecting the UA_Server structure with a semaphore/lock.

tijo40, 2016-09-07:
I found the below link more useful.
http://stackoverflow.com/questions/18646778/build-a-3rd-party-library-from-source-within-an-existing-qt-project.

jpfr, 2016-09-10:
Great find @tijo40!

I'm closing. Please reopen if more assistance is required.

closed 2016-09-10 by jpfr