open62541/open62541 #787: QNX problem

tpiat, 2016-09-06:
In the latest versions of open62541 structure used addinfo

In file open62541.h used include <gulliver.h> 
which for some reasons undefines constatnt __EXT_POSIX1_200112,
without in does not compile addrinfo structure (netdb.h).

Now each new version I have made follows:
```
#ifdef __QNXNTO__
#define __EXT_POSIX1_200112
#endif
#include <netdb.h>
```

Can anybody solve this problem or insert 3 line to project???

jpfr, 2016-09-10:
Hey @tpiat ,

I'm no expert in QNX.
That looks like the internal macro of some compiler. It would be a lot easier if we could just use
```c
#define _XOPEN_SOURCE 500
```
or a similar POSIX standard definition.

Can you point us to the the file where __EXT_POSIX1_200112 is undefined or where it should be defined?

Here are some ideas on how to find the location of a specific macro: http://stackoverflow.com/questions/290952/how-to-know-in-gcc-when-given-macro-preprocessor-symbol-gets-declared

tpiat, 2016-09-10:
It macro defined in system header files of QNX.
But I do not know, where in undefined.
I insert
```
#ifndef __EXT_POSIX1_200112
#error AAA
#endif
```
before  #include <netdb,h>
and 
```
#ifndef __EXT_POSIX1_200112
#error BBB
#endif
```
after. I have only second error.

This macro appear if defined __QNXNTO__
```
#ifdef __QNXNTO__
#define __EXT_POSIX1_200112
#endif
```
This problem founds not only me:
http://webcache.googleusercontent.com/search?q=cache:bcIwPgyLtR0J:community.qnx.com/sf/go/post102808+&cd=4&hl=ru&ct=clnk&gl=ru

Any idea why "gnu++0x" removes __EXT_POSIX1_200112? How do I fix that?

I can send or list header files with this macro on monday.

tpiat, 2016-09-11:
I foung where this macros is undefined

File platform.h
```#undef __EXT_POSIX1_200112	/* Posix 1003.1-2001 */```
and
```#if defined(_QNX_SOURCE) || (_XOPEN_SOURCE+0 >= 600) || (_POSIX_C_SOURCE+0 >= 200112) || (defined(__EXT) && !defined(_POSIX_SOURCE) && (_POSIX_C_SOURCE+0 <= 0) && !defined(_XOPEN_SOURCE))
#define __EXT_POSIX1_200112
#endif
```

I define _XOPEN_SOURCE as 600 and now it compiles

Thanx!!!

closed 2016-09-11 by tpiat