open62541/open62541 #788: error/client	GetEndpointRequest failed

kennyc0522, 2016-09-07:
Hi,

I encountered an issue in connecting with OPC server from Siemens 840D controller.
but I can make connection with simulation server.
As the title, the error messages showed in the shell are "GetEndpointRequest failed" and "No suitable endpoint found".

The OPC client was compiled and  ran on the VMware machine, and the OS is Ubuntu 15.
I have added some debug messages into "open62541.c", recompiled client and tried again and again., 
it seems to stuck in function "UA_Client_connect".

I have no idea how to solve this issue. 
Does anybody has any advice? 
Hoping to have been helped.

Regards,
Kenny

kennyc0522, 2016-09-07:
Thread #676 

I will try the following way to test connection.
> tpiat commented on 12 Jul
> I use SIMATIC server.
> 
> I have commented all lines with UA_SecureChannel_processSequenceNumber call.
> (contain retval = UA_SecureChannel_processSequenceNumber)
> Client works now.

jpfr, 2016-09-08:
Hey @kennyc0522 ,

which version of the SDK are you using? v0.2-rc1 or more recent from the github repository?
It's quite possible your problem was solved in the meantime.

We can help you if you give us more information, e.g. as in here:
https://github.com/open62541/open62541/wiki/Writing-Good-Issue-Reports

kennyc0522, 2016-09-09:
Please see the attached file

[OpenSecureChannel.zip](https://github.com/open62541/open62541/files/463157/OpenSecureChannel.zip)

kennyc0522, 2016-09-10:
[image]

jpfr, 2016-09-10:
Hey @kennyc0522 ,

ok. We are getting close to a solution. Can you

- Try with a nightly build from here: http://open62541.org/releases/ Just replace the source and header file.
- Give the statuscode in ¨GetEndpointRequest failed with statuscode 0xXXXXX¨

xty19, 2016-11-07:
i have the same problem.
and i am testing siemens 828D device to connect via open62541 code (.c and .h files).
but i also meet the same "GetEndpointRequest failed" and "No suitable endpoint found" question.

xty19, 2016-11-07:
I can connect the siemens device ! :laughing: 
just change the **UA_ClientCOnfig_standard** !

- `const UA_EXPORT UA_ClientConfig UA_ClientConfig_standard =
-  {
-     .timeout = 5000,
-     .secureChannelLifeTime = 600000,
-     .logger = Logger_Stdout,
-     .localConnectionConfig =
-  {
- //original setting
- 	/*.protocolVersion = 0,
- 	.sendBufferSize = 65536,
- 	.recvBufferSize  = 65536,
- 	.maxMessageSize = 65536,
- 	.maxChunkCount = 1*/
- //changed 
- 
- - setting

		.protocolVersion = 0,
		.sendBufferSize = 16384,
		.recvBufferSize = 16384,
		.maxMessageSize = 65536,
		.maxChunkCount = 0 },
		.connectionFunc = UA_ClientConnectionTCP
};`

kennyc0522, 2017-02-09:
Recently, I've got a new siemens 828D, and had opportunity to work on OPC test.

[ Issue ]
"No suitable UserTokenPolicy found for the possible endpoints".

[ Environments ]
The OPC client was compiled and ran on the VMware machine.
OS is Ubuntu 15.
SDK version is : Git-Revision ad31a12b6d.

Hi @jpfr ,
I've just tried the latest nightly version: Git-Revision ad31a12b6d.
When the client was testing, I got an error message which was "No suitable UserTokenPolicy found for the possible endpoints".
But I'm sure that username and password is correct.

Hi @xty19,
Even if I changed the buffer size of UA_ClientConfig_standard as your setting,
client was still responding the same error message : "No suitable UserTokenPolicy found for the possible endpoints".
Would you please tell me which nightly version that you're using?

Thanks for your help.

xty19, 2017-02-09:
@kennyc0522 
friend ,you'd better paste the code and we can find the problem . that's what you can do now

kennyc0522, 2017-02-09:
Hi @xty19 ,

The code is very simple, only make connection with device.

[ Environments ]
The OPC client was compiled and ran on the VMware machine.
OS is Ubuntu 15.
SDK version is : 
  1. Git-Revision ad31a12.
  2. Git-Revision v0.2-rc1 (open62541.c)

    #include <stdio.h>
    #include <inttypes.h>
    #include "open62541.h"

    int main(void) {
    UA_Client *client = UA_Client_new(UA_ClientConfig_standard);

    UA_StatusCode retval = UA_Client_connect_username(client, "opc.tcp://192.8.4.248:4840", "OpcUaClient", "OpcUa");

    if(retval != UA_STATUSCODE_GOOD) {
        UA_Client_delete(client);
        return (int)retval;
    }

    UA_Client_disconnect(client);
    UA_Client_delete(client);
    return (int) UA_STATUSCODE_GOOD;
    }

xty19, 2017-02-09:
did you change the code 
UA_ClientConfig_standard ?

kennyc0522, 2017-02-09:
Using default setting which can't connect to device with default setting, and changing it as following setting either.

    .protocolVersion = 0,
    .sendBufferSize = 16384,
    .recvBufferSize = 16384,
    .maxMessageSize = 65536,
    .maxChunkCount = 0

xty19, 2017-02-09:
I suggest you provide 
some informations from Opcua Binary Protocol.
just like the pic above . you paste before and we can analyse the problem with you

jpfr, 2017-02-09:
> When the client was testing, I got an error message which was "No suitable UserTokenPolicy found for the possible endpoints". But I'm sure that username and password is correct.

The error was generated before the username and password are checked.
My guess is, the server does only allow encrypted communication. Or requires at the password to be encrypted, even when the securechannel is not encrypted. open62541 does currently not support encryption.

We can say for sure with a wireshark capture. Or you try login with an anonymous user.

Best regards,
Julius

ku3bl3r, 2017-04-13:
Hi there,

I ran in the same problem when trying to connect my open62541 client to a Siemens SINUMERIK 840D sl OPC UA Server. The _ActivateSession**Response**_ is _0x80200000 BadIdentityTokenInvalid_.
My _ActivateSession**Request**_ looks like this: 

>No.     Time           Source                Destination           Protocol Length Info
   3846 3.136040       141.58.103.181        141.58.102.66         OpcUa    237    UA Secure Conversation Message: ActivateSessionRequest
Frame 3846: 237 bytes on wire (1896 bits), 237 bytes captured (1896 bits) on interface 0
Ethernet II, Src: Dell_6e:f6:8f (ec:f4:bb:6e:f6:8f), Dst: SiemensS_23:5c:27 (00:1f:f8:23:5c:27)
Internet Protocol Version 4, Src: 141.58.103.181, Dst: 141.58.102.66
Transmission Control Protocol, Src Port: 23802, Dst Port: 4840, Seq: 432, Ack: 6891, Len: 183
OpcUa Binary Protocol
    Message Type: MSG
    Chunk Type: F
    Message Size: 183
    SecureChannelId: 131202
    Security Token Id: 1
    Security Sequence Number: 4
    Security RequestId: 4
    OpcUa Service : Encodeable Object
        TypeId : ExpandedNodeId
            NodeId EncodingMask: Four byte encoded Numeric (0x01)
            NodeId Namespace Index: 0
            NodeId Identifier Numeric: ActivateSessionRequest (467)
        ActivateSessionRequest
            RequestHeader: RequestHeader
                AuthenticationToken: NodeId
                    .... 0010 = EncodingMask: Numeric of arbitrary length (0x2)
                    Namespace Index: 1
                    Identifier Numeric: 1059961410
                Timestamp: Apr 13, 2017 14:53:06.057000000 Mitteleuropäische Sommerzeit
                RequestHandle: 4
                Return Diagnostics: 0x00000000
                    .... .... .... ...0 = ServiceLevel / SymbolicId: False
                    .... .... .... ..0. = ServiceLevel / LocalizedText: False
                    .... .... .... .0.. = ServiceLevel / AdditionalInfo: False
                    .... .... .... 0... = ServiceLevel / Inner StatusCode: False
                    .... .... ...0 .... = ServiceLevel / Inner Diagnostics: False
                    .... .... ..0. .... = OperationLevel / SymbolicId: False
                    .... .... .0.. .... = OperationLevel / LocalizedText: False
                    .... .... 0... .... = OperationLevel / AdditionalInfo: False
                    .... ...0 .... .... = OperationLevel / Inner StatusCode: False
                    .... ..0. .... .... = OperationLevel / Inner Diagnostics: False
                AuditEntryId: [OpcUa Null String]
                TimeoutHint: 600000
                AdditionalHeader: ExtensionObject
                    TypeId: ExpandedNodeId
                        EncodingMask: 0x00, EncodingMask: Two byte encoded Numeric
                        Identifier Numeric: 0
                    EncodingMask: 0x00
                        .... ...0 = has binary body: False
                        .... ..0. = has xml body: False
            ClientSignature: SignatureData
                Algorithm: [OpcUa Null String]
                Signature: <MISSING>[OpcUa Null ByteString]
            ClientSoftwareCertificates: Array of SignedSoftwareCertificate
            LocaleIds: Array of String
                ArraySize: -1
            UserIdentityToken: ExtensionObject
                TypeId: ExpandedNodeId
                    EncodingMask: 0x01, EncodingMask: Four byte encoded Numeric
                    Namespace Index: 0
                    Identifier Numeric: 324
                EncodingMask: 0x01, has binary body
                    .... ...1 = has binary body: True
                    .... ..0. = has xml body: False
                UserNameIdentityToken: UserNameIdentityToken
                    PolicyId: http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15
                    UserName: OpcUaClient
                    Password: 6f70637561
[... 57 lines cut ...]
            ClientSoftwareCertificates: Array of SignedSoftwareCertificate
            LocaleIds: Array of String
                ArraySize: 1
                [0]: LocaleIds: en
            UserIdentityToken: ExtensionObject
                TypeId: ExpandedNodeId
                    EncodingMask: 0x01, EncodingMask: Four byte encoded Numeric
                    Namespace Index: 0
                    Identifier Numeric: 324
                EncodingMask: 0x01, has binary body
                    .... ...1 = has binary body: True
                    .... ..0. = has xml body: False
                UserNameIdentityToken: UserNameIdentityToken
                    PolicyId: http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15
                    UserName: OpcUaClient
                    Password: a8644f682650cef4d1ca2eafa1bd2bed3590ab6b3ba9294a...
                    EncryptionAlgorithm: http://www.w3.org/2001/04/xmlenc#rsa-1_5
            UserTokenSignature: SignatureData
                Algorithm: [OpcUa Null String]
                Signature: <MISSING>[OpcUa Null ByteString]

only major difference I can spot is the encryption of the sent password, as mentioned by @jpfr.

Are there a fix available yet?
Thanks in advance!

Pro, 2017-11-23:
@Infinity95 this is related to encryption.
In the new encryption implementation do we support encrypting the password without an encrypted secure channel?

Infinity95, 2017-11-23:
No, that's not yet implemented. Having that requires a rework of the accesscontrol plugin and maybe some slight modifications to the plugin api. I haven't touched that yet, since we first need the working policy for that.

sirop, 2018-01-17:
@ku3bl3r 

Hello.

Have you found any work around for Siemens 840D?
The logs you presented - are these Wireshark dump?

Thanks in advance.

ku3bl3r, 2018-01-23:
@sirop 
Hi, no I didn't find any workaround. In that specific project I switched the stack for a commercial one. Still, I think a lot of applications would benefit from the open62541 stack with encryption.
The dumps are created with wireshark, yes. It has a built-in opcua filter.

Cheers

Pro, 2018-02-05:
This is a duplicate of #934 and will be fixed as soon as we have client side encryption.

closed 2018-02-05 by Pro