open62541/open62541 #789: Question about browsing

iquestor, 2016-09-08:
I am working on a simple server endpoint browser and am trying to figure out how to browse a given child node. Should be easy, but I cant seem to find the right way. I am connecting to KepServEx 5.9 OPCUA. 

From the client example, I am working with a callback function. 

static UA_StatusCode nodeIter(UA_NodeId childId, UA_Boolean isInverse, UA_NodeId referenceTypeId, void *handle) 
{
  UA_NodeId *parent = (UA_NodeId *) handle;
  if(!isInverse) {
    printf("nsIndex %d, ParentiD %d --- reftype %d %-16.*s ---> child NSIndex %d, Child id %d name %-16.*s\n",
           parent->namespaceIndex, parent->identifier.numeric,
           referenceTypeId.identifier.numeric,(int)referenceTypeId.identifier.string.length, referenceTypeId.identifier.string.data,childId.namespaceIndex,
		childId.identifier.numeric, (int)childId.identifier.string.length, childId.identifier.string.data);
 }
  return UA_STATUSCODE_GOOD;
}
 
**from the first example, its called like so:** 

UA_NodeId *parent = UA_NodeId_new();
    *parent = UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER);
    UA_Client_forEachChildNodeCall(client, UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER),
                                   nodeIter, (void *) parent);
    UA_NodeId_delete(parent);

results: 
nsIndex 0, ParentiD 85 --- reftype 40 (null)           ---> child NSIndex 0, Child id 61 name (null)
nsIndex 0, ParentiD 85 --- reftype 35 (null)           ---> child NSIndex 0, Child id 2253 name (null)
nsIndex 0, ParentiD 85 --- reftype 35 (null)           ---> child NSIndex 2, Child id 7 name _System
nsIndex 0, ParentiD 85 --- reftype 35 (null)           ---> child NSIndex 2, Child id 8 name Channel1
nsIndex 0, ParentiD 85 --- reftype 35 (null)           ---> child NSIndex 2, Child id 18 name Data Type Examples
nsIndex 0, ParentiD 85 --- reftype 35 (null)           ---> child NSIndex 2, Child id 19 name Simulation Examples

What I want to do is now call  UA_Client_forEachChildNodeCall() to get the list of nodes for the node "Data Type Examples".  How do I do this? There are folders and items under this node in Kepware. 

I have tried using the NodeID 18 and also changing the NODEID types, etc. I looked for different flags other than "UA_NS0ID_OBJECTSFOLDER";  I cannot find anything in the documentation that would suggest how to proceed. any help would be greatly appreciated.

jpfr, 2016-09-08:
Have you tried to use the `browse` service directly?
https://github.com/open62541/open62541/blob/0.2/include/ua_client.h#L225

iquestor, 2016-09-08:
Using that browse service would seem to give me the same issue as I don't know how to configure the UA_BrowseRequest to ask for a given Node.  I am trying to write a tree based browser so first I would get all root nodes, then a user would select a node, and i would make the call to browse that node, and so on. I need to do it this way as I am doing the Browse code in a C++ DLL which is called from a C# Application, which displays the current state of the tree. 

So my issue is in how do I set up UA_BrowseRequest object set up to view a particular Node. ? The following configuration gives me the root namespace nodes, as per the OP results. 

UA_BrowseRequest bReq;
    UA_BrowseRequest_init(&bReq);
    bReq.requestedMaxReferencesPerNode = 0;
    bReq.nodesToBrowse = UA_BrowseDescription_new();
    bReq.nodesToBrowseSize = 1;
    bReq.nodesToBrowse[0].nodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER); /* browse objects folder */
    bReq.nodesToBrowse[0].resultMask = UA_BROWSERESULTMASK_ALL; /* return everything */

**consider this result :**
nsIndex 0, ParentiD 85 --- reftype 35 (null) ---> child NSIndex 2, Child id 18 name Data Type Examples

So, using the info above, How would I set up the UA_BrowseRequest object to return browse results of "Data Type Examples"? 
I tried the following, none of which returned results
UA_NODEID_NUMERIC(18, UA_NS0ID_OBJECTSFOLDER);   // ChildID of node I want
UA_NODEID_BYTESTRING_ALLOC(18, "Data Type Examples"); // string type node

My issue is I can't find documentation on how to set this up. 
Any help would be greatly appreciated.

jpfr, 2016-09-12:
Hello @iquestor ,

it seems you are doing the right thing already.
The full definition of the browsedescription type is here:
http://open62541.org/doc/current/types_generated.html#browsedescription

Nodes in an OPC UA information model are connect with types, directed references.
For browsing, you need to specifiy
- where to start (you can only browse one level deep)
- which reference types to follow (usually, you will want UA_NS0ID_HIERARCHICALREFERENCES and subtypes enabled to follow all hierarchical reference types)
- which direction (forward/backward)
- which node classes (can be left as zero)
- which parts of the refernce to return (UA_BROWSERESULTMASK_ALL is good)

For a complete description, download part 4 of the standard from the OPC UA foundation homepage.
https://opcfoundation.org/

jpfr, 2016-11-17:
Please reopen if questions remain.

closed 2016-11-17 by jpfr