open62541/open62541 #790: Decode problem @@

ilike28, 2016-09-10:
Hello

I Use the following code from Check_builtin.c

 size_t pos = 0;
 UA_Byte data[] =
 {'A', 'C', 'P', 'L', 'T', ' ', 'U', 'A' };
 UA_ByteString srcF = { 8, data };
 UA_String dstF;
  retval = UA_String_decodeBinary(&srcF, &pos, &dstF);

printf("	FFFFFFFFFFFFFFFFFF	 [UA_String decode :%s]   %d \n\n",dstF.data,retval);

But the result is  -> FFFFFFFFFFFFFFFFFF	 [UA_String decode :(null)]   -2147024896  
Does that the right result??

Thanks ^^

jpfr, 2016-09-10:
The encoded buffer needs to contain the length of the string. The length is always encoded as a 32bit signed integer in little-endian encoding. Try
```c
UA_Byte data[] = {8, 0, 0, 0, 'A', 'C', 'P', 'L', 'T', ' ', 'U', 'A' };
UA_ByteString srcF = { 10, data };
```

ilike28, 2016-09-11:
Thank you@@

If I had a ByteString
				 size_t pos = 0;
				 UA_Byte data[]  = {8, 0, 0, 0,8, 0, 0, 0, 'A', 'C', 'P', 'L', 'T', ' ', 'U', 'A' };
				UA_ByteString srcF = { 16, data };
				UA_String dstF;
				 retval = UA_String_decodeBinary(&srcF, &pos, &dstF);

				printf("	FFFFFFFFFFFFFFFFFF	 [UA_String decode :%s]   %d \n\n",dstF.data,retval);

Could you tell me how can I decode this ByteString?

Pro, 2016-09-26:
@ilike28 does this solve your issue? If not, please reopen this issue and add your comment.

closed 2016-09-26 by Pro