open62541/open62541 #791: [Question] Variant Type Value change to the UA_String Type value

ilike28, 2016-09-11:
Hello @@

I had a variant type value "&(eventNotificatinList->events[j].eventFields[xx])" .
I want to change type to the UA_String type value.
So,  I change variant type to the ByteString ,the change the ByteString type to the UA_String

**[change variant type to the ByteString]**
 UA_ByteString dst0 ;//= { 32, data };
 UA_ByteString_allocBuffer(&dst0, eventNotificatinList->events[j].eventFields[xx].type->memSize);

 UA_Variant_encodeBinary(&(eventNotificatinList->events[j].eventFields[xx]),&dst0,&offset);
 printf("**Xavier  ( %lu ) __ offset:%lu __ memsize:%d __ DataType:%s**\n"
 ,xx
 ,offset
  ,eventNotificatinList->events[j].eventFields[xx].type->memSize
 ,eventNotificatinList->events[j].eventFields[xx].type->typeName
   );

 dst0.length=offset;

and get the following log.
**Xavier  ( 3 ) __ offset:10 __ memsize:16 __ DataType:String
c 5 0 0 0 41 42 43 44 45** 

The variant is send by the server which contain UA_String "ABCDE".
So, I got the ByteString "c 5 0 0 0 41 42 43 44 45".

**[ change the ByteString type to the UA_String]**
 UA_String dst2;
 size_t offset_string=0;
 UA_StatusCode retval  = UA_String_decodeBinary(&dst0,&offset_string,&dst2);
 
  printf("  [UA_String decode :%s]   %d \n\n",dst2.data,retval);

and get the following log.
 [UA_String decode :(null)]   -2147024896

Could anyone tell me how can change variant type value to the UA_String type ??

thanks ^^.

jpfr, 2016-09-12:
Where do you want to change the type?
- Before encoding
- In the encoded bytestring
- After decoding

Note that changing the type in a variant is generally unsafe!
You can however do it for String / ByteString since both have essentially the same content and differ only in semantics.

You could just do
```c
variant->type = &UA_TYPES[UA_TYPES_STRING]
```

ilike28, 2016-09-12:
I get the *Varient from Server.
The Varient's array is 9.  (Varient[0]~Varient[8])
I saw the Varient[3] is String type.
But I don't know how to change Varient[3] to the UA_String type.
So, I Change the Varient[3] to the ByteString,and get the information is **c 5 0 0 0 41 42 43 44 45**

Then I know Varient[3] is String "ABCDE";
But I don't know how to decode Varient[3] to "ABCDE".

Could you please tell me how can decode Varient @@?

Thanks ^_^

ilike28, 2016-09-12:
Hello ~
I found the method @@
UA_String *test = (UA_String *)Variant[3];

Thank you @@

closed 2016-09-12 by ilike28