open62541/open62541 #793: Transmit a whole structure.

MatthiasSchild, 2016-09-12:
Hello everybody,

I have a question about writing data to an open62541 opc server.
I would like to know if it's possible to write a whole structure with a single request. For example a structure `Axis` with the variables `X`. `Y` and `Z`. Now instead of sending three requests I would prefer to build a `Axis` structure instance and write it with a single request.
Is it even possible, and how?

Thank you in advance.

Matthias

FlorianPalm, 2016-09-12:
Hi Matthias,
the write service allows to bulk write. You can write several nodes and their attributes at once. Writing means that you just change the existing values (check out the write service in the IEC 62541 specs).  In case you want to create the structure, I would suggest to create a specific "factory" method (call service) on server side rather than using the standard addNodes/addReferences services.

MatthiasSchild, 2016-09-12:
Thank you for your response.
So instead of sending the structure, I have to disassemble it to its components (primitives like float, string, etc) and send them in a single write request which contains those `nodesToWrite`.

FlorianPalm, 2016-09-12:
If your variableNode value is a structure (DataType) it should be possible to write that complete structure too. At the moment, I am not sure if the XML-parsing/encoding/decoding of that structure is possible.

jpfr, 2016-09-12:
Hey @MatthiasSchild, 

it is possible to define custom data structures in OPC UA that are handled as a single scalar value.

Look at how we autogenerate the data type descriptions.
It is all done in the CMakelists.txt using generate_datatypes.py.

But it is still rough to do for your own types.
In the end, you can do sth like

```c
struct myStruct x;
x.a = 1;
x.b = 2;

UA_Variant var;
UA_Variant_init(&var);
UA_Variant_setScalarCopy(&var, &x, &UA_MY_TYPES[UA_MYSTRUCT]);
```

The encoding is then handled automatically, given that `UA_MY_TYPES[UA_MYSTRUCT]` contains the generated (or hand-written) description of your data type.

But on the receiving end, you will still end up with an ExtensionObject with an un-decoded BinaryString. That's because the encoding function cannot know about your personal data types in our current implementation...

As I said, its possible. But still kinda rough.
If you really really want this to work smoothly, we can give you some pointers for the necessary improvements of the stack.

Best regards,
Julius

jpfr, 2016-09-12:
As @FlorianPalm said, the alternative is to have three separate variables (that can be written independently in a single write request) or to use a method with three arguments to set all values at once.

MatthiasSchild, 2016-09-13:
@FlorianPalm @jpfr Thank you for your help.

Sending a structure seems to be to complex for my project.
I will write the primitives instead.

closed 2016-09-13 by MatthiasSchild