open62541/open62541 #794: Memoryleaking while adding/removing nodes

nikita158, 2016-09-12:
Valgrind reports zero memleak, but memory usage is continuously growing while nothing is actually happening.

```C
#include "open62541.h"

UA_NodeId addNode(UA_Server *server) {
    UA_NodeId resultNodeId = UA_NODEID_NULL,
          targetNodeId = UA_NODEID_STRING_ALLOC(0, "testNodeId");

    UA_ObjectAttributes attr;
    UA_ObjectAttributes_init(&attr);
    attr.displayName = UA_LOCALIZEDTEXT_ALLOC("en_US", "displayName");
    attr.description = UA_LOCALIZEDTEXT_ALLOC("en_US", "description");
    UA_QualifiedName qualifiedName = UA_QUALIFIEDNAME_ALLOC(1, "qualifiedName");

    UA_Server_addObjectNode(server, targetNodeId, UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER),
                UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES),
                qualifiedName, UA_NODEID_NULL,
                attr, NULL, &resultNodeId);

    UA_QualifiedName_deleteMembers(&qualifiedName);
    UA_ObjectAttributes_deleteMembers(&attr);
    UA_NodeId_deleteMembers(&targetNodeId);

    return resultNodeId;
}

void removeNode(UA_Server *server, UA_NodeId nodeId) {
    UA_Server_deleteNode(server, nodeId, true);
    UA_NodeId_deleteMembers(&nodeId);
}

int main() {
    UA_ServerNetworkLayer nl = UA_ServerNetworkLayerTCP(UA_ConnectionConfig_standard, 16664);
    UA_ServerConfig config = UA_ServerConfig_standard;
    config.networkLayers = &nl;
    config.networkLayersSize = 1;

    UA_Server *server = UA_Server_new(config);

    UA_NodeId nodeId = UA_NODEID_NULL;

    while (1) {
        nodeId = addNode(server);
        removeNode(server, nodeId);
    }

    return 0;
}
```

jpfr, 2016-09-12:
Hey @nikita158,

hmm. If more memory is used but does not leak, maybe we don't remove some references.

Can you run your server with valgrind/massif for a while and tell us which code paths allocate most memory?
http://valgrind.org/docs/manual/ms-manual.html

nikita158, 2016-09-13:
```console
--------------------------------------------------------------------------------
  n        time(i)         total(B)   useful-heap(B) extra-heap(B)    stacks(B)
--------------------------------------------------------------------------------
 90 24,409,260,197        2,143,336        1,853,621       289,715            0
 91 24,625,814,409        2,152,488        1,861,541       290,947            0
 92 24,703,531,569        2,175,512        1,884,101       291,411            0
86.60% (1,884,101B) (heap allocation functions) malloc/new/new[], --alloc-fns, etc.
->75.11% (1,634,000B) 0x41AF75: addOneWayReference (in /mnt/DRIVE_C/dev/open62541-0.2/out)
| ->75.11% (1,634,000B) 0x417298: UA_Server_editNode (in /mnt/DRIVE_C/dev/open62541-0.2/out)
|   ->74.23% (1,614,800B) 0x41B23C: Service_AddReferences_single (in /mnt/DRIVE_C/dev/open62541-0.2/out)
|   | ->37.23% (810,000B) 0x4188EE: Service_AddNodes_existing (in /mnt/DRIVE_C/dev/open62541-0.2/out)
|   | | ->36.59% (796,080B) 0x41A315: Service_AddNodes_single (in /mnt/DRIVE_C/dev/open62541-0.2/out)
|   | | | ->36.59% (796,080B) 0x41A587: __UA_Server_addNode (in /mnt/DRIVE_C/dev/open62541-0.2/out)
|   | | |   ->36.59% (796,080B) 0x401231: UA_Server_addObjectNode (in /mnt/DRIVE_C/dev/open62541-0.2/out)
|   | | |     ->36.59% (796,080B) 0x401416: addNode (in /mnt/DRIVE_C/dev/open62541-0.2/out)
|   | | |       ->36.59% (796,080B) 0x40159C: main (in /mnt/DRIVE_C/dev/open62541-0.2/out)
|   | | |         
|   | | ->00.64% (13,920B) in 1+ places, all below ms_print's threshold (01.00%)
|   | | 
|   | ->36.66% (797,600B) 0x41987A: instantiateObjectNode (in /mnt/DRIVE_C/dev/open62541-0.2/out)
|   | | ->36.66% (797,600B) 0x418A7A: Service_AddNodes_existing (in /mnt/DRIVE_C/dev/open62541-0.2/out)
|   | | | ->36.65% (797,360B) 0x41A315: Service_AddNodes_single (in /mnt/DRIVE_C/dev/open62541-0.2/out)
|   | | | | ->36.65% (797,360B) 0x41A587: __UA_Server_addNode (in /mnt/DRIVE_C/dev/open62541-0.2/out)
|   | | | |   ->36.65% (797,360B) 0x401231: UA_Server_addObjectNode (in /mnt/DRIVE_C/dev/open62541-0.2/out)
|   | | | |     ->36.65% (797,360B) 0x401416: addNode (in /mnt/DRIVE_C/dev/open62541-0.2/out)
|   | | | |       ->36.65% (797,360B) 0x40159C: main (in /mnt/DRIVE_C/dev/open62541-0.2/out)
|   | | | |         
|   | | | ->00.01% (240B) in 1+ places, all below ms_print's threshold (01.00%)
|   | | | 
|   | | ->00.00% (0B) in 1+ places, all below ms_print's threshold (01.00%)
|   | | 
|   | ->00.33% (7,200B) in 1+ places, all below ms_print's threshold (01.00%)
|   | 
|   ->00.88% (19,200B) in 1+ places, all below ms_print's threshold (01.00%)
|   
->09.18% (199,631B) 0x405C29: UA_Array_copy (in /mnt/DRIVE_C/dev/open62541-0.2/out)
| ->09.16% (199,327B) 0x4058D7: copyNoInit (in /mnt/DRIVE_C/dev/open62541-0.2/out)
| | ->09.15% (199,120B) 0x405961: UA_copy (in /mnt/DRIVE_C/dev/open62541-0.2/out)
| | | ->09.15% (199,003B) 0x401A07: UA_String_copy (in /mnt/DRIVE_C/dev/open62541-0.2/out)
| | | | ->09.14% (198,890B) 0x403F48: NodeId_copy (in /mnt/DRIVE_C/dev/open62541-0.2/out)
| | | | | ->09.14% (198,870B) 0x4041EB: ExpandedNodeId_copy (in /mnt/DRIVE_C/dev/open62541-0.2/out)
| | | | | | ->09.14% (198,870B) 0x40585B: copyNoInit (in /mnt/DRIVE_C/dev/open62541-0.2/out)
| | | | | |   ->09.14% (198,870B) 0x405961: UA_copy (in /mnt/DRIVE_C/dev/open62541-0.2/out)
| | | | | |     ->09.14% (198,870B) 0x401BAB: UA_ExpandedNodeId_copy (in /mnt/DRIVE_C/dev/open62541-0.2/out)
| | | | | |       ->09.14% (198,870B) 0x41B025: addOneWayReference (in /mnt/DRIVE_C/dev/open62541-0.2/out)
| | | | | |         ->09.14% (198,870B) 0x417298: UA_Server_editNode (in /mnt/DRIVE_C/dev/open62541-0.2/out)
| | | | | |           ->09.14% (198,870B) 0x41B23C: Service_AddReferences_single (in /mnt/DRIVE_C/dev/open62541-0.2/out)
| | | | | |             ->04.57% (99,440B) 0x4188EE: Service_AddNodes_existing (in /mnt/DRIVE_C/dev/open62541-0.2/out)
| | | | | |             | ->04.57% (99,440B) 0x41A315: Service_AddNodes_single (in /mnt/DRIVE_C/dev/open62541-0.2/out)
| | | | | |             |   ->04.57% (99,440B) 0x41A587: __UA_Server_addNode (in /mnt/DRIVE_C/dev/open62541-0.2/out)
| | | | | |             |     ->04.57% (99,440B) 0x401231: UA_Server_addObjectNode (in /mnt/DRIVE_C/dev/open62541-0.2/out)
| | | | | |             |       ->04.57% (99,440B) 0x401416: addNode (in /mnt/DRIVE_C/dev/open62541-0.2/out)
| | | | | |             |         ->04.57% (99,440B) 0x40159C: main (in /mnt/DRIVE_C/dev/open62541-0.2/out)
| | | | | |             |           
| | | | | |             ->04.57% (99,430B) 0x41987A: instantiateObjectNode (in /mnt/DRIVE_C/dev/open62541-0.2/out)
| | | | | |               ->04.57% (99,430B) 0x418A7A: Service_AddNodes_existing (in /mnt/DRIVE_C/dev/open62541-0.2/out)
| | | | | |                 ->04.57% (99,430B) 0x41A315: Service_AddNodes_single (in /mnt/DRIVE_C/dev/open62541-0.2/out)
| | | | | |                   ->04.57% (99,430B) 0x41A587: __UA_Server_addNode (in /mnt/DRIVE_C/dev/open62541-0.2/out)
| | | | | |                     ->04.57% (99,430B) 0x401231: UA_Server_addObjectNode (in /mnt/DRIVE_C/dev/open62541-0.2/out)
| | | | | |                       ->04.57% (99,430B) 0x401416: addNode (in /mnt/DRIVE_C/dev/open62541-0.2/out)
| | | | | |                         ->04.57% (99,430B) 0x40159C: main (in /mnt/DRIVE_C/dev/open62541-0.2/out)
| | | | | |                           
| | | | | ->00.00% (20B) in 1+ places, all below ms_print's threshold (01.00%)
| | | | | 
[... cut ...]
| | ->00.01% (207B) in 1+ places, all below ms_print's threshold (01.00%)
| | 
| ->00.01% (304B) in 1+ places, all below ms_print's threshold (01.00%)
| 
->02.32% (50,470B) in 12 places, all below massif's threshold (01.00%)
  
--------------------------------------------------------------------------------
  n        time(i)         total(B)   useful-heap(B) extra-heap(B)    stacks(B)
--------------------------------------------------------------------------------
 93 24,920,817,563        2,164,968        1,872,436       292,532            0
```

nikita158, 2016-09-13:
```C
static UA_StatusCode
addOneWayReference(UA_Server *server, UA_Session *session,
                   UA_Node *node, const UA_AddReferencesItem *item) {
    size_t i = node->referencesSize;
    size_t refssize = (i+1) | 3; // so the realloc is not necessary every time

    printf("static UA_StatusCode addOneWayReference(..) browseName = %s i = %i noderefssize = %i\n",
           node->browseName.name.data, i, refssize);
...
}
```
output:
```console
...
static UA_StatusCode addOneWayReference(..) browseName = qualifiedName i = 1 noderefssize = 3
static UA_StatusCode addOneWayReference(..) browseName = BaseObjectType i = 13226 noderefssize = 13227
static UA_StatusCode addOneWayReference(..) browseName = qualifiedName i = 0 noderefssize = 3
static UA_StatusCode addOneWayReference(..) browseName = Objects i = 13209 noderefssize = 13211
static UA_StatusCode addOneWayReference(..) browseName = qualifiedName i = 1 noderefssize = 3
static UA_StatusCode addOneWayReference(..) browseName = BaseObjectType i = 13227 noderefssize = 13231
```

nikita158, 2016-09-13:
seems that #795 solved problem

closed 2016-09-13 by FlorianPalm