open62541/open62541 #803: How to modify the array value  of specified index?

ljia310, 2016-09-20:
There is an array of node(Demo.Static.Arrays.Int16 [10]), 
how to only set  Demo.Static.Arrays.Int16 [3]=16 ?

FlorianPalm, 2016-09-20:
Within a writeRequest you can specify the field 'indexRange'. You can specifiy a Range, for example:
"0:3" or "5" to read either the subarray from index 0 to 3 or the single element which is located at index 5. Same goes for reading arrays

ljia310, 2016-09-21:
I am using the code below:
```
    UA_Variant value;
    UA_Variant_init(&value);
    short v = 123;
    UA_Variant_setScalarCopy(&value,&v,&UA_TYPES[UA_TYPES_INT16]);

    UA_WriteValue wValue;
    UA_WriteValue_init(&wValue);
    wValue.nodeId = UA_NODEID_STRING(2,"Demo.Static.Arrays.Int16");
    wValue.attributeId = UA_ATTRIBUTEID_VALUE;
    wValue.value.value = value;
    wValue.value.hasValue = true;
    wValue.indexRange = UA_STRING("5");

    UA_WriteRequest wReq;
    UA_WriteRequest_init(&wReq);
    wReq.nodesToWrite = &wValue;
    wReq.nodesToWriteSize = 1;
    
    UA_WriteResponse wResp = UA_Client_Service_write(client, wReq);
    UA_StatusCode retval = wResp.responseHeader.serviceResult;
    UA_WriteResponse_deleteMembers(&wResp);
```
Finally return UA_STATUSCODE_GOOD , but in fact the values ​​in the array has not been modified

ljia310, 2016-09-21:
the capture data 
[opc_ua_pcap.zip](https://github.com/open62541/open62541/files/484100/opc_ua_pcap.zip)

ljia310, 2016-09-21:
the data of capture
[opc_ua_pcap.zip](https://github.com/open62541/open62541/files/484109/opc_ua_pcap.zip)

ljia310, 2016-09-21:
```
//UA_Variant_setScalarCopy(&value,&v,&UA_TYPES[UA_TYPES_INT16]);
UA_Variant_setArrayCopy(&value,&v,1,&UA_TYPES[UA_TYPES_INT16])
```

 thanks！！！！

closed 2016-09-21 by ljia310