open62541/open62541 #808: Matrix write error

ljia310, 2016-09-21:
```
    UA_Variant value;
    UA_Variant_init(&value);
    float v = 33.33;
    UA_Variant_setArrayCopy(&value,&v,1,&UA_TYPES[UA_TYPES_FLOAT])
    UA_WriteValue wValue;
    UA_WriteValue_init(&wValue);
    wValue.nodeId = UA_NODEID_STRING(2,"Demo.Static.Matrix.Float");
    wValue.attributeId = UA_ATTRIBUTEID_VALUE;
    wValue.value.value = value;
    wValue.value.hasValue = true;
    wValue.indexRange = UA_STRING("0,0,1");

    UA_WriteRequest wReq;
    UA_WriteRequest_init(&wReq);
    wReq.nodesToWrite = &wValue;
    wReq.nodesToWriteSize = 1;
    
    UA_WriteResponse wResp = UA_Client_Service_write(client, wReq);
    UA_StatusCode retval = wResp.responseHeader.serviceResult;
    UA_WriteResponse_deleteMembers(&wResp);
```
The value has not been modified.

the capture file:
[opc_ua_Matrix.zip](https://github.com/open62541/open62541/files/484326/opc_ua_Matrix.zip)

jpfr, 2016-09-21:
Hello @ljia310 ,

I just added
- two unit tests that cover your use case
https://github.com/open62541/open62541/blob/0.2/tests/check_services_attributes.c#L867
- a small boatload of log output for writing attributes

Can you retry with the current 0.2 branch and logging set to debug or trace level?

The Wireshark log shows a BADTYPEINVALID error. That is strange since you are writing float into a float array. And the unit tests that do the same run fine.

ljia310, 2016-09-22:
hi @jpfr 

**Add information:**
the opc ua client is developed by  open62541,opc ua server using the demo of www.unified-automation.com

Demo.Static.Matrix.Float =  3 * 3 * 3

[image]

I use the current 0.2 branch after the latest code to recompile the current debugging.
Enter the log is as follows:
```
$ ./opc_ua_test.exe
[09/22/2016 02:03:28.283] debug/network Sent HEL message
[09/22/2016 02:03:28.283] debug/network Received ACK message
[09/22/2016 02:03:28.283] debug/channel Requesting to open a SecureChannel
[09/22/2016 02:03:28.285] debug/channel SecureChannel opened
[09/22/2016 02:03:28.285] debug/client  Sending a request of type 426
[09/22/2016 02:03:28.286] debug/client  Received a response of type 431
[09/22/2016 02:03:28.286] debug/client  Sending a request of type 459
[09/22/2016 02:03:28.287] debug/client  Received a response of type 464
[09/22/2016 02:03:28.287] debug/client  Sending a request of type 465
[09/22/2016 02:03:28.288] debug/client  Received a response of type 470
[09/22/2016 02:03:28.288] debug/client  Sending a request of type 767
[09/22/2016 02:03:28.288] debug/client  Received a response of type 397

```

jpfr, 2016-09-26:
Aah. You implement a client. I though you had the server implemented with open62541.

I tested with the ANSI C example server from UnifiedAutomation.
The server does implement indexed writes into this variable.

Closing the issue as this is not on our side.
You should ask in the Unified Automation forum.

PS: open62541 based servers support indexed writes (of any dimension) by default… just saying.

closed 2016-09-26 by jpfr

ljia310, 2016-09-27:
@jpfr ,Thanks.