open62541/open62541 #810: Add features table to webpage

Pro, 2016-09-21:
What is your opinion about adding a table on the main webpage listing all the supported and non-supported features of open62541?

Additionally it may include information about the supported service sets, server profiles, multithreading support, ...

Currently it's not that easy for beginners to see what is supported and what not.

jpfr, 2016-09-21:
👍 

I can add a list of supported conformance units from the CTT.

We could take the list form node-opcua as the starting point:
https://github.com/node-opcua/node-opcua

Pro, 2016-09-21:
Uh, nice! node-opcua has exactly what I meant.
So probably it's also best to include the table into the README.md or a separate markup file to keep it consistent with the corresponding branch/code-base. Then on the webpage we may just include this file from current main branch.

I'll create a PR for the first draft.

Pro, 2016-09-21:
@jpfr I created the PR. You can directly push to my branch

closed 2016-09-22 by jpfr