open62541/open62541 #814: i cant find nodeset.h nodeset.c

xty19, 2016-09-23:
i use the command cmake ,but at the folder i cant find nodeset.h and nodeset.c ,could you tell me?please

FlorianPalm, 2016-09-23:
Try first to configure only UA_BUILD_EXAMPLES_NODESET_COMP within cmake, afterwards you can activate UA_BUILD_EXAMPLES

markusgraube, 2016-09-23:
``` bash
mkdir build
cd build
cmake -DUA_BUILD_EXAMPLES_NODESET_COMPILER=ON ..
make generate_informationmodel
```
Then you should find `nodeset.[c/h]` in `build/src_generated/`

xty19, 2016-09-26:
My system is raspberry3 debian.
i tried `cmke -DUA_BUILD_EXAMPLES_NODESET_COMPILER=ON .. 
           cmke -UA_BUILD_EXAMPLES_NODESET_COMPILER=ON ..
           cmke -D UA_BUILD_EXAMPLES_NODESET_COMPILER=ON ..
           make `

it informs me that
`fatal: Not a git repository (or any of the parent directories): .git
-- Git failed (not a repo, or no tags). Build will not contain git revision info
--Git version: unknown --git -commit -id -unknown
--Configuring done
--Generating done
--Build files have been written to:/path
--Build files have been written to :/path

make 
100 Build target open62541-object
100 build`target open62541
100 Build target open62541-static

xty19, 2016-09-26:
i try the below two friends ideas ,but it seems dont work ,i still cant find the node.c .h files.

markusgraube, 2016-09-26:
Hi @xty19, that's strange.
Are you on an up-to-date branch `0.2`?
Which version of CMake have you?

Can you provide the complete output, when starting the build in a new directory?

xty19, 2016-09-26:
@markusgraube 
hi friends.i use  it on raspberry 3, i will give you the detail soon,wait for minutes,and thanks for you reply.

xty19, 2016-09-26:
`i use the newest one 2.0
root@raspberrypi:/home/pi/open62541# cmake --version
cmake version 3.0.2

CMake suite maintained and supported by Kitware (kitware.com/cmake).
root@raspberrypi:/home/pi/open62541# 

result::
root@raspberrypi:/home/pi/open62541#  gcc -std=c99 open62541.c server_variable_xml_xx.c myNS.c -o server_xxx
myNS.c: In function ‘myNS’:
myNS.c:15:1: error: unknown type name ‘UA_ObjectTypeNode’
 UA_ObjectTypeNode *opcua_node_objecttype_t_ns_2_i_1001 = UA_NodeStore_newObjectTypeNode();
 ^
myNS.c:15:1: warning: implicit declaration of function ‘UA_NodeStore_newObjectTypeNode’ [-Wimplicit-function-declaration]
myNS.c:15:58: warning: initialization makes pointer from integer without a cast
 UA_ObjectTypeNode *opcua_node_objecttype_t_ns_2_i_1001 = UA_NodeStore_newObjectTypeNode();
                                                          ^
myNS.c:16:36: error: request for member ‘browseName’ in something not a structure or union
 opcua_node_objecttype_t_ns_2_i_1001->browseName = UA_QUALIFIEDNAME_ALLOC(1, "testType");
                                    ^
myNS.c:17:36: error: request for member ‘displayName’ in something not a structure or union
 opcua_node_objecttype_t_ns_2_i_1001->displayName = UA_LOCALIZEDTEXT_ALLOC("en_US", "testType");
                                    ^
myNS.c:18:36: error: request for member ‘description’ in somethingroot@raspberrypi:/home/pi/open62541# cmake --version
cmake version 3.0.2

CMake suite maintained and supported by Kitware (kitware.com/cmake).
root@raspberrypi:/home/pi/open62541# 
 not a structure or union
 opcua_node_objecttype_t_ns_2_i_1001->description = UA_LOCALIZEDTEXT_ALLOC("en_US", "");
                                    ^
myNS.c:19:36: error: request for member ‘nodeId’ in something not a structure or union
 opcua_node_objecttype_t_ns_2_i_1001->nodeId.namespaceIndex = 2;
                                    ^
myNS.c:20:36: error: request for member ‘nodeId’ in something not a structure or union
 opcua_node_objecttype_t_ns_2_i_1001->nodeId.identifier.numeric = 1001;
                                    ^
myNS.c:22:1: warning: implicit declaration of function ‘UA_RCU_LOCK’ [-Wimplicit-function-declaration]
 UA_RCU_LOCK();
 ^
myNS.c:23:1: warning: implicit declaration of function ‘UA_NodeStore_insert’ [-Wimplicit-function-declaration]
 UA_NodeStore_insert(server->nodestore, (UA_Node*) opcua_node_objecttype_t_ns_2_i_1001);
 ^
myNS.c:23:27: error: dereferencing pointer to incomplete type
 UA_NodeStore_insert(server->nodestore, (UA_Node*) opcua_node_objecttype_t_ns_2_i_1001);
                           ^
myNS.c:23:41: error: ‘UA_Node’ undeclared (first use in this function)
 UA_NodeStore_insert(server->nodestore, (UA_Node*) opcua_node_objecttype_t_ns_2_i_1001);
                                         ^
myNS.c:23:41: note: each undeclared identifier is reported only once for each function it appears in
myNS.c:23:49: error: expected expression before ‘)’ token
 UA_NodeStore_insert(server->nodestore, (UA_Node*) opcua_node_objecttype_t_ns_2_i_1001);
                                                 ^
myNS.c:24:1: warning: implicit declaration of function ‘UA_RCU_UNLOCK’ [-Wimplicit-function-declaration]
 UA_RCU_UNLOCK();
 ^
myNS.c:30:1: warning: implicit declaration of function ‘UA_alloca’ [-Wimplicit-function-declaration]
 UA_Variant *opcua_node_variable_t_ns_2_i_6001_variant = UA_alloca(sizeof(UA_Variant));
 ^
myNS.c:30:57: warning: initialization makes pointer from integer without a cast
 UA_Variant *opcua_node_variable_t_ns_2_i_6001_variant = UA_alloca(sizeof(UA_Variant));
                                                         ^
myNS.c:33:1: error: unknown type name ‘UA_VariableNode’
 UA_VariableNode *opcua_node_variable_t_ns_2_i_6001 = UA_NodeStore_newVariableNode();
 ^
myNS.c:33:1: warning: implicit declaration of function ‘UA_NodeStore_newVariableNode’ [-Wimplicit-function-declaration]
myNS.c:33:54: warning: initialization makes pointer from integer without a cast
 UA_VariableNode *opcua_node_variable_t_ns_2_i_6001 = UA_NodeStore_newVariableNode();
                                                      ^
myNS.c:34:34: error: request for member ‘browseName’ in something not a structure or union
[... 97 lines cut ...]
 opcua_node_variable_t_ns_2_i_6002->minimumSamplingInterval = (UA_Double) 0.0;
                                  ^
myNS.c:82:34: error: request for member ‘userAccessLevel’ in something not a structure or union
 opcua_node_variable_t_ns_2_i_6002->userAccessLevel = (UA_Int32) 3;
                                  ^
myNS.c:83:34: error: request for member ‘accessLevel’ in something not a structure or union
 opcua_node_variable_t_ns_2_i_6002->accessLevel = (UA_Int32) 3;
                                  ^
myNS.c:84:34: error: request for member ‘valueRank’ in something not a structure or union
 opcua_node_variable_t_ns_2_i_6002->valueRank = (UA_Int32) -1;
                                  ^
myNS.c:85:34: error: request for member ‘value’ in something not a structure or union
 opcua_node_variable_t_ns_2_i_6002->value.variant.value = *opcua_node_variable_t_ns_2_i_6002_variant;
                                  ^
myNS.c:86:34: error: request for member ‘valueSource’ in something not a structure or union
 opcua_node_variable_t_ns_2_i_6002->valueSource = UA_VALUESOURCE_VARIANT;
                                  ^
myNS.c:89:27: error: dereferencing pointer to incomplete type
 UA_NodeStore_insert(server->nodestore, (UA_Node*) opcua_node_variable_t_ns_2_i_6002);
                           ^
myNS.c:89:49: error: expected expression before ‘)’ token
 UA_NodeStore_insert(server->nodestore, (UA_Node*) opcua_node_variable_t_ns_2_i_6002);
                                                 ^
root@raspberrypi:/home/pi/open62541# 
`

xty19, 2016-09-26:
and here is the myNS.h and myNS.c files

myNS.h
`/* WARNING: This is a generated file.
 * Any manual changes will be overwritten.

 */

#include "ua_server_internal.h"
#include "ua_nodes.h"
#include "ua_util.h"
#include "ua_types.h"
#include "ua_types_encoding_binary.h"
#include "ua_types_generated_encoding_binary.h"
#include "ua_transport_generated_encoding_binary.h"

#include "open62541.h"
#define NULL ((void *)0)

#define UA_NS2ID_TESTTYPE 1001
extern void myNS(UA_Server *server);

`

myNS.c
`/* WARNING: This is a generated file.
 * Any manual changes will be overwritten.

 */
#include <stdio.h>
#include "myNS.h"
UA_INLINE void myNS(UA_Server *server) 
{
UA_Server_addNamespace(server, "http://yourorganisation.org/test/");
printf("i com the xml");

do {
// Node: opcua_node_objectType_t(ns=2;i=1001), 1:testType
UA_ObjectTypeNode *opcua_node_objecttype_t_ns_2_i_1001 = UA_NodeStore_newObjectTypeNode();
opcua_node_objecttype_t_ns_2_i_1001->browseName = UA_QUALIFIEDNAME_ALLOC(1, "testType");
opcua_node_objecttype_t_ns_2_i_1001->displayName = UA_LOCALIZEDTEXT_ALLOC("en_US", "testType");
opcua_node_objecttype_t_ns_2_i_1001->description = UA_LOCALIZEDTEXT_ALLOC("en_US", "");
opcua_node_objecttype_t_ns_2_i_1001->nodeId.namespaceIndex = 2;
opcua_node_objecttype_t_ns_2_i_1001->nodeId.identifier.numeric = 1001;
// Parent node does not exist yet. This node will be bootstrapped and linked later.
UA_RCU_LOCK();
UA_NodeStore_insert(server->nodestore, (UA_Node*) opcua_node_objecttype_t_ns_2_i_1001);
UA_RCU_UNLOCK();
} while(0);

/*
do {
UA_Variant *opcua_node_variable_t_ns_2_i_6001_variant = UA_alloca(sizeof(UA_Variant));
UA_Variant_init(opcua_node_variable_t_ns_2_i_6001_variant);
// Node: opcua_node_variable_t(ns=2;i=6001), 1:Var1
UA_VariableNode *opcua_node_variable_t_ns_2_i_6001 = UA_NodeStore_newVariableNode();
opcua_node_variable_t_ns_2_i_6001->browseName = UA_QUALIFIEDNAME_ALLOC(1, "Var1");
opcua_node_variable_t_ns_2_i_6001->displayName = UA_LOCALIZEDTEXT_ALLOC("en_US", "Var1");
opcua_node_variable_t_ns_2_i_6001->description = UA_LOCALIZEDTEXT_ALLOC("en_US", "");
opcua_node_variable_t_ns_2_i_6001->nodeId.namespaceIndex = 2;
opcua_node_variable_t_ns_2_i_6001->nodeId.identifier.numeric = 6001;
opcua_node_variable_t_ns_2_i_6001->minimumSamplingInterval = (UA_Double) 0.0;
opcua_node_variable_t_ns_2_i_6001->userAccessLevel = (UA_Int32) 3;
opcua_node_variable_t_ns_2_i_6001->accessLevel = (UA_Int32) 3;
opcua_node_variable_t_ns_2_i_6001->valueRank = (UA_Int32) -1;
opcua_node_variable_t_ns_2_i_6001->value.variant.value = *opcua_node_variable_t_ns_2_i_6001_variant;
opcua_node_variable_t_ns_2_i_6001->valueSource = UA_VALUESOURCE_VARIANT;
// Parent node does not exist yet. This node will be bootstrapped and linked later.
UA_RCU_LOCK();
UA_NodeStore_insert(server->nodestore, (UA_Node*) opcua_node_variable_t_ns_2_i_6001);
UA_RCU_UNLOCK();
} while(0);
*/

/*
do {
// Node: opcua_node_object_t(ns=2;i=5001), 1:testInstance
UA_ObjectNode *opcua_node_object_t_ns_2_i_5001 = UA_NodeStore_newObjectNode();
opcua_node_object_t_ns_2_i_5001->browseName = UA_QUALIFIEDNAME_ALLOC(1, "testInstance");
opcua_node_object_t_ns_2_i_5001->displayName = UA_LOCALIZEDTEXT_ALLOC("en_US", "testInstance");
opcua_node_object_t_ns_2_i_5001->description = UA_LOCALIZEDTEXT_ALLOC("en_US", "");
opcua_node_object_t_ns_2_i_5001->nodeId.namespaceIndex = 2;
opcua_node_object_t_ns_2_i_5001->nodeId.identifier.numeric = 5001;
opcua_node_object_t_ns_2_i_5001->eventNotifier = (UA_Byte) 0;
// Parent node does not exist yet. This node will be bootstrapped and linked later.
UA_RCU_LOCK();
UA_NodeStore_insert(server->nodestore, (UA_Node*) opcua_node_object_t_ns_2_i_5001);
UA_RCU_UNLOCK();
} while(0);
*/

/*

do {
UA_Variant *opcua_node_variable_t_ns_2_i_6002_variant = UA_alloca(sizeof(UA_Variant));
UA_Variant_init(opcua_node_variable_t_ns_2_i_6002_variant);
// Node: opcua_node_variable_t(ns=2;i=6002), 1:Var1
UA_VariableNode *opcua_node_variable_t_ns_2_i_6002 = UA_NodeStore_newVariableNode();
opcua_node_variable_t_ns_2_i_6002->browseName = UA_QUALIFIEDNAME_ALLOC(1, "Var1");
opcua_node_variable_t_ns_2_i_6002->displayName = UA_LOCALIZEDTEXT_ALLOC("en_US", "Var1");
opcua_node_variable_t_ns_2_i_6002->description = UA_LOCALIZEDTEXT_ALLOC("en_US", "");
opcua_node_variable_t_ns_2_i_6002->nodeId.namespaceIndex = 2;
opcua_node_variable_t_ns_2_i_6002->nodeId.identifier.numeric = 6002;
opcua_node_variable_t_ns_2_i_6002->minimumSamplingInterval = (UA_Double) 0.0;
opcua_node_variable_t_ns_2_i_6002->userAccessLevel = (UA_Int32) 3;
opcua_node_variable_t_ns_2_i_6002->accessLevel = (UA_Int32) 3;
opcua_node_variable_t_ns_2_i_6002->valueRank = (UA_Int32) -1;
opcua_node_variable_t_ns_2_i_6002->value.variant.value = *opcua_node_variable_t_ns_2_i_6002_variant;
opcua_node_variable_t_ns_2_i_6002->valueSource = UA_VALUESOURCE_VARIANT;
// Parent node does not exist yet. This node will be bootstrapped and linked later.
UA_RCU_LOCK();
UA_NodeStore_insert(server->nodestore, (UA_Node*) opcua_node_variable_t_ns_2_i_6002);
UA_RCU_UNLOCK();
} while(0);
*/

}
`

xty19, 2016-09-26:
the example nodeset.h nodeset.c cant be generated.
and i try to generate the server_nodeset.xml  server_nodeset.c server_nodeset.h 
but it seems that it cant be compiled.
the error is above.
anyone can pass the compiling ???

Pro, 2016-09-26:
in your `myNS.h` you should either include `open62541.h` or the list of files
```
#include "ua_server_internal.h"
#include "ua_nodes.h"
#include "ua_util.h"
#include "ua_types.h"
#include "ua_types_encoding_binary.h"
#include "ua_types_generated_encoding_binary.h"
#include "ua_transport_generated_encoding_binary.h"
```
But not both. You probably only want "open62541.h" if you are using amalgamation.

Can you check your "open62541.h" if it contains the following lines:
```
typedef struct {
    UA_NODE_BASEATTRIBUTES
    UA_Boolean isAbstract;

    /* Members specific to open62541 */
    UA_ObjectLifecycleManagement lifecycleManagement;
} UA_ObjectTypeNode;
```

And please use proper markdown formatting if you post code (I edited your comments):
https://guides.github.com/features/mastering-markdown/

xty19, 2016-09-26:
@Pro  thanks you , i try your comment,  remove "open62541.h" in the myNS.h file, 
 but it now implies that 

`root@raspberrypi:/home/pi/open3/open62541/tools/pyUANamespace#  gcc -std=c99 open62541.c server_variable_xml_xx.c myNS.c -o server_xxxx
In file included from server_variable_xml_xx.c:8:0:
myNS.h:23:18: error: unknown type name ‘UA_Server’
 extern void myNS(UA_Server *server);
                  ^
server_variable_xml_xx.c: In function ‘main’:
server_variable_xml_xx.c:49:2: warning: implicit declaration of function ‘myNS’ [-Wimplicit-function-declaration]
  myNS(server);
  ^
In file included from myNS.c:5:0:
myNS.h:23:18: error: unknown type name ‘UA_Server’
 extern void myNS(UA_Server *server);
                  ^
myNS.c:6:11: error: expected ‘=’, ‘,’, ‘;’, ‘asm’ or ‘__attribute__’ before ‘void’
 UA_INLINE void myNS(UA_Server *server) {`

Pro, 2016-09-26:
You are missing `#include "ua_server.h"`

Pro, 2016-09-26:
Can you post your XML which you used to create the `myNS.c/h` and the commands, step by step

xty19, 2016-09-26:
@Pro ok,

xty19, 2016-09-26:
my email is 915484383@qq.com and i am using opc-ua to contact for the work and study

xty19, 2016-09-26:
`<UANodeSet xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:uax="http://opcfoundation.org/UA/2008/02/Types.xsd" xmlns="http://opcfoundation.org/UA/2011/03/UANodeSet.xsd" xmlns:s1="http://yourorganisation.org/test/" xmlns:xsd="http://www.w3.org/2001/XMLSchema">
    <NamespaceUris>
        <Uri>http://yourorganisation.org/test/</Uri>
    </NamespaceUris>
    <Aliases>
        <Alias Alias="Double">i=11</Alias>
        <Alias Alias="Organizes">i=35</Alias>
        <Alias Alias="HasModellingRule">i=37</Alias>
        <Alias Alias="HasTypeDefinition">i=40</Alias>
        <Alias Alias="HasSubtype">i=45</Alias>
        <Alias Alias="HasComponent">i=47</Alias>
    </Aliases>
    <UAObjectType NodeId="ns=1;i=1001" BrowseName="1:testType">
        <DisplayName>testType</DisplayName>
        <References>
            <Reference ReferenceType="HasSubtype" IsForward="false">i=58</Reference>
            <Reference ReferenceType="HasComponent">ns=1;i=6001</Reference>
        </References>
    </UAObjectType>
    <UAVariable DataType="Double" ParentNodeId="ns=1;i=1001" NodeId="ns=1;i=6001" BrowseName="1:Var1" UserAccessLevel="3" AccessLevel="3">
        <DisplayName>Var1</DisplayName>
        <References>
            <Reference ReferenceType="HasTypeDefinition">i=63</Reference>
            <Reference ReferenceType="HasModellingRule">i=78</Reference>
            <Reference ReferenceType="HasComponent" IsForward="false">ns=1;i=1001</Reference>
        </References>
    </UAVariable>
    <UAObject NodeId="ns=1;i=5001" BrowseName="1:testInstance">
        <DisplayName>testInstance</DisplayName>
        <References>
            <Reference ReferenceType="Organizes" IsForward="false">i=85</Reference>
            <Reference ReferenceType="HasTypeDefinition">ns=1;i=1001</Reference>
            <Reference ReferenceType="HasComponent">ns=1;i=6002</Reference>
        </References>
    </UAObject>
    <UAVariable DataType="Double" ParentNodeId="ns=1;i=5001" NodeId="ns=1;i=6002" BrowseName="1:Var1" UserAccessLevel="3" AccessLevel="3">
        <DisplayName>Var1</DisplayName>
        <References>
            <Reference ReferenceType="HasTypeDefinition">i=63</Reference>
            <Reference ReferenceType="HasComponent" IsForward="false">ns=1;i=5001</Reference>
        </References>
        <Value>
            <uax:Double>42.0</uax:Double>
        </Value>
    </UAVariable>
</UANodeSet>
`

Pro, 2016-09-26:
Please also post the complete set of commands you are using to create the source code. Step by step instructions.

xty19, 2016-09-26:
@Pro ok wait a sec

`python ./generate_open62541CCode.py myNS.xml myNS`

add the header in my main.c  and put them in the same folder.
(open62541.c open62541.h myNS.c myNS.h server_variable_xml_xx.c [main class]),
and gcc them to complie
gcc -std=c99 open62541.c server_variable_xml_xx.c myNS.c -o server_xxxx
 and come out the error

Pro, 2016-09-26:
@xty19 Ok, I could reproduce the problem with following steps:
```
$ cd open62541
$ mkdir build && cd build && cmake -DUA_ENABLE_AMALGAMATION=ON .. && make
$ cp open62541.* ~/test/ns/
$ cd ~/test/ns/
$ ~/open62541/tools/pyUANamespace/generate_open62541CCode.py myNS.xml myNS
# create a simple main.c file
$ gcc -std=c99 open62541.c myNS.c main.c -o test
```

If you use the generateCCode.py like this it works (copied from examples):

```
~/open62541/tools/pyUANamespace/generate_open62541CCode.py  -i ~/open62541/tools/pyUANamespace/NodeID_Blacklist_FullNS0.txt ~/open62541/tools/schema/namespace0/Opc.Ua.NodeSet2.xml  myNS.xml myNS
```

For further description see:
http://open62541.org/doc/current/tutorial_noderelations.html

Please reopen the issue if this doesn't solve your problem.

closed 2016-09-26 by Pro

xty19, 2016-09-26:
@Pro  i am trying your solution ,i will leave the result later ,thanks

xty19, 2016-09-26:
@Pro 
i use the commend in the folder ,it works! create nodeset.c nodeset.h, but!!!
i create another folder containing open62541 
and i use comment "$ cd open62541
$ mkdir build && cd build && cmake -DUA_ENABLE_AMALGAMATION=ON .. && make"

the nodeset.c nodeset.h cant be created! why?

Pro, 2016-09-26:
nodeset.c and nodeset.h are only created if you enable `-DUA_BUILD_EXAMPLES_NODESET_COMPILER=ON` on cmake. But you don't really need that, do you?
You have your own nodeset in myNS.xml

So the cmake stuff for open62541 only needs to be executed once to build open62541.c/h

And yes, delete the build folder to be safe.

xty19, 2016-09-26:
@Pro  thanks ,i use the commend "cmake -DUA_BUILD_EXAMPLES_NODESET_COMPILER=ON .. "
it works ,create nodeset.c and nodeset.h.  i am unfamiliar to the commend.
i am testing the xml py commend, thanking for your patient soluting ,^_^.

xty19, 2016-09-26:
@Pro I use "~/open62541/tools/pyUANamespace/generate_open62541CCode.py  -i ~/open62541/tools/pyUANamespace/NodeID_Blacklist_FullNS0.txt ~/open62541/tools/schema/namespace0/Opc.Ua.NodeSet2.xml  myNS.xml myNS
”
and it can work!!!
thanks you very much ,you really give me a great favor!!!