open62541/open62541 #816: Question: Is someone using ENABLE_EXTERNAL_NAMESPACES?

markusgraube, 2016-09-23:
Hi, is someone there who is using `ENABLE_EXTERNAL_NAMESPACES` and can explain a little bit about the current state?

I have the feeling that this feature could be quite interesting. However I am not quite sure what exactly the purpose of this option.

* Is there a simple example showing the features?
* What is working and what is not?

I would be quite happy if I could get some hints where to start and how to proceed in developing.

jpfr, 2016-09-23:
@FlorianPalm developed this to plug external systems into open62541.
The current API requires external namespaces to implement the supported services themselves.
The multiplexing is done by mapping external-namespaces to namespace-indices.

My latest information is that ACPLT don't use this any longer and have a much smaller API where external systems are effectively "nodestores". So they return entire nodes and our internal services do the heavy lifting.

@FlorianPalm can tell you more.

FlorianPalm, 2016-09-26:
Yes, @jpfr  is right. 
At the moment, we have integrated an OPC UA interface with the external namespace functionality. We have created an own namespace for nodes which are managed by the ACPLT runtime environment (RTE). Though it is the fastest way to connect a "legacy" system with OPC UA, we have put a lot of effort into migration to newer versions of the stack.
We hope that we can reduce the overhead if we use the interface at "nodestore"-level. @ JGrothoff has worked with the "nodestore"-interface, I am sure that he will offer a pull request soon.

JGrothoff, 2016-09-26:
This is correct, i've worked with the "nodestore"-interface. We opened an OSIsoft PI Demosystem for the open62541 Stack via nodestore Interface. But it was a very prototypical implementation and there is some stuff todo, as I only used the getNode(...) for read access with the external nodestore.

Basicly the routing from all nodestore functions to the internel nodestore has to be established. This is done in my fork (https://github.com/JGrothoff/open62541/tree/nodestore_interface) by adding a pointer back to the server in the internal nodestore and overwriting the getNode(...) function to get the correct function handle. (https://github.com/JGrothoff/open62541/blob/nodestore_interface/src/server/ua_nodestore.c#L254)
Maybe we should rework the design of the nodestoreinterface at this point. Then we can use the interface at nodestore level for our ACPLT/RTE, as @FlorianPalm mentioned and bring together the external namespace and external nodestore.

JGrothoff, 2016-09-30:
So I just reworked the nodestore Interface (and merged it with the master branch) in my fork. --> Pull Request #830

There is still a lot todo, but i will test the implementation with our RuntimeEnvironment in the future.

closed 2016-10-04 by jpfr

markusgraube, 2016-10-05:
Thank you for the info