open62541/open62541 #818: existance of "networklayer_udp.h"

deyspandan, 2016-09-26:
Is there no "networklayer_udp.h" or am I missing something?

Pro, 2016-09-26:
Do you mean this?

https://github.com/open62541/open62541/blob/0.2/plugins/ua_network_udp.h

deyspandan, 2016-09-26:
yes!

my confusion arose from the file renaming,

although @jpfr did cleanup and renaming:
https://github.com/open62541/open62541/commit/54daafd37aeefb864e002393bbd5e58521190ff2

some more renaming remains I guess:
like this one -> https://github.com/open62541/open62541/blob/0.2/examples/server_udp.c

closed 2016-09-26 by deyspandan

Pro, 2016-09-26:
Yes, this should also be renamed. Will have a look at it. Thanks

deyspandan, 2016-09-26:
No problem, will raise a pull request once I get it up and running :)

Pro, 2016-09-26:
@deyspandan I created a PR where all the compilation errors are fixed.
Please continue discussion there if you have any additions/notes.

https://github.com/open62541/open62541/issues/820