open62541/open62541 #819: Option to make a VariableNode read-only?

caspervanzoest, 2016-09-26:
I am bulding my OPC server using the open62541.
All data is added by objects of my own 'Item' class.

However some of these 'items', are read-only.
Is there any option to give the variable nodes a read-only value?

Following code shows how I add a item to my server:
 ```
    void Server::addToServer(Item* item, UA_Server *UA_server, UA_ValueCallback callback)
     {
		this->disableCallbacks();
                UA_VariableAttributes attr = item->getAttributes();

		UA_NodeId myNodeId = UA_NODEID_STRING(1, (char*)(item->getFullName()).c_str());
		UA_QualifiedName myVarName = UA_QUALIFIEDNAME(1, (char*)(item->getName().c_str()));

		if (item->getName() == "Root")
		{
				UA_NodeId parentNodeId = UA_NODEID_NUMERIC(0,UA_NS0ID_OBJECTSFOLDER);
				UA_NodeId parentReferenceNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES);
				UA_Server_addVariableNode(UA_server, myNodeId, parentNodeId,
							parentReferenceNodeId, myVarName,
							UA_NODEID_NULL, attr, NULL, NULL);
		}
		else
		{
				UA_NodeId parentNodeId = UA_NODEID_STRING(1, (char*)(item->getParent().c_str()));
				UA_NodeId parentReferenceNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES);
				UA_Server_addVariableNode(UA_server, myNodeId, parentNodeId,
							parentReferenceNodeId, myVarName,
							UA_NODEID_NULL, attr, NULL, NULL);
		}
		UA_Server_writeValue(UA_server, myNodeId, attr.value);

        UA_Server_setVariableNode_valueCallback(UA_server, myNodeId, callback);
		this->enableCallbacks();
    }
```

Current situation:
When a client changes a value of a node (ie UAExert), the callback method checks if the item is read-only,
and if so, it undoes the change made to the node.

I want to improve this situation by:
- making it impossible to write such node anyway.
or:
- show a kind of message to the client, containing the performed action was not handled.

Any tips or suggestions would be great.
Thanks in advance!

FlorianPalm, 2016-09-26:
At the moment, we do not manage access rights. You could implement a little function that is called in whenever the a write access is performed and change the value only if the node is not in the "read"-only list ("hack"). 
It is planned to integrate the feature in a decent way but I am not sure how long it will take.

caspervanzoest, 2016-09-26:
Thanks for your answer, FlorianPalm. 
I already have a function which checks if the node is in de read only list. (as described under 'current situation'.)
Works fine, however, I would like to let the user know the action is not performed, like a messagebox or something. 

Now the only way to check as client is refreshing the node to see if the change of value is performed. 

Is it possible to notify the client in any way? 

Thanks in advance.

jpfr, 2016-09-26:
Sure. Every service returns status codes.
There is a status code for the entire read service and one for every read-item you send with it.

Presenting the results to the user is up to you.

caspervanzoest, 2016-10-03:
> Sure. Every service returns status codes.
> There is a status code for the entire read service and one for every read-item you send with it.
> 
> Presenting the results to the user is up to you.

Returning UA_StatusCode is only possible when using UA_DataSource, right?
(when working with the open62541.h file in my situation)

closed 2016-10-13 by caspervanzoest