open62541/open62541 #823: Using multiple .xml files with pyUANamespace compiler

Pro, 2016-09-28:
I have a few issues/questions about the pyUANamespaceCompiler especially to @Stasik0 and @ichrispa, but of course also others ;)

For your convenience I created a simple repo for this issue which contains all the described files and which you can use to test:
https://github.com/Pro/open62541_namespaces

Assume the following setup:
* I have a project which uses open62541 as a dependency with static linking
* `UA_ENABLE_GENERATE_NAMESPACE0=ON`
* In my project I want to use [Opc.Ua.Di.NodeSet2.xml](https://opcfoundation.org/UADevices/1.1/Opc.Ua.Di.NodeSet2.xml) as an additional base namespace
* The `Opc.Ua.Di.NodeSet2` namespace is extended with a custom ObjectType `PlcDeviceType` using UaModeler which generates `project1.NodeSet2.xml` by Company1
* The `Opc.Ua.Di.NodeSet2` is extended again with a custom type `MachineDeviceType` by Company2  which generates `project2.NodeSet2.xml`

So finally I have 4 .xml files:
- `Opc.Ua.NodeSet2.xml`
- `Opc.Ua.Di.NodeSet2.xml`
- `project1.NodeSet2.xml` depending on `Opc.Ua.Di.NodeSet2.xml` (see below for content)
- `project2.NodeSet2.xml` depending on `Opc.Ua.Di.NodeSet2.xml` (see below for content)

Now I want to compile everything together, i.e., I need to create .c/.h files for the .xml models. This can be done with the `generate_open62541CCode.py` script. In the tutorial it is mentioned that one should add additional .xml files to the corresponding line in the open62541 CMakeLists.txt file. I don't want to change the base files of open62541, thus I create a custom call to `generate_open62541CCode.py` like it is done for the examples.

Here is now my question:
How does the call to `generate_open62541CCode.py` have to look like, to generate the model which includes `Opc.Ua.Di.NodeSet2.xml`,  `project1.NodeSet2.xml`, `project2.NodeSet2.xml`

I tried it with the following, but (after fixing #822) it fails during gcc make. This error only occurs if **amalgamation is enabled**. If it is disabled, it works fine.

CMakeLists.txt:
```
# generated namespace for custom .xml models
add_custom_command(OUTPUT ${PROJECT_BINARY_DIR}/src_generated/test_generated.c
        ${PROJECT_BINARY_DIR}/src_generated/test_generated.h
        PRE_BUILD
        COMMAND ${PYTHON_EXECUTABLE} ${PROJECT_SOURCE_DIR}/deps/open62541/tools/pyUANamespace/generate_open62541CCode.py
        -i ${PROJECT_SOURCE_DIR}/deps/open62541/tools/pyUANamespace/NodeID_Blacklist_FullNS0.txt
        ${PROJECT_SOURCE_DIR}/deps/open62541/tools/schema/namespace0/Opc.Ua.NodeSet2.xml
        ${PROJECT_SOURCE_DIR}/model/Opc.Ua.Di.NodeSet2.xml
        ${PROJECT_SOURCE_DIR}/model/project1.NodeSet2.xml
        ${PROJECT_SOURCE_DIR}/model/project2.NodeSet2.xml
        ${PROJECT_BINARY_DIR}/src_generated/test_generated
        DEPENDS ${PROJECT_SOURCE_DIR}/deps/open62541/tools/schema/namespace0/Opc.Ua.NodeSet2.xml
        ${PROJECT_SOURCE_DIR}/model/Opc.Ua.Di.NodeSet2.xml
        ${PROJECT_SOURCE_DIR}/model/project1.NodeSet2.xml
        ${PROJECT_SOURCE_DIR}/model/project2.NodeSet2.xml
        ${PROJECT_SOURCE_DIR}/deps/open62541/tools/pyUANamespace/generate_open62541CCode.py)
```

Some errors during gcc/make:

```
...
/home/profanter/repos/open62541_namespaces/build/src_generated/test_generated.c: In function ‘test_generated’:
/home/profanter/repos/open62541_namespaces/build/src_generated/test_generated.c:307:34: warning: implicit declaration of function ‘malloc’ [-Wimplicit-function-declaration]
 inputArguments = (UA_Argument *) malloc(sizeof(UA_Argument) * 1);
...
/home/profanter/repos/open62541_namespaces/build/src_generated/test_generated.c:1483:1: warning: implicit declaration of function ‘UA_NodeStore_insert’ [-Wimplicit-function-declaration]
 UA_NodeStore_insert(server->nodestore, (UA_Node*) opcua_node_object_t_ns_2_i_6248);
 ^
...
/home/profanter/repos/open62541_namespaces/build/src_generated/test_generated.c:1483:41: error: ‘UA_Node’ undeclared (first use in this function)
 UA_NodeStore_insert(server->nodestore, (UA_Node*) opcua_node_object_t_ns_2_i_6248);
 ^
...
/home/profanter/repos/open62541_namespaces/build/src_generated/test_generated.c:1847:1: error: unknown type name ‘UA_ObjectNode’
 UA_ObjectNode *opcua_node_object_t_ns_2_i_6544 = UA_NodeStore_newObjectNode();
 ^
...
```

The problem here is that if amalgamation is disabled, some internal header files are included in the generated .h file. If amalgamation is enabled, it only includes "open62541.h" which does not provide all the necessary information for the generated model source.

Am I doing something wrong here?

ichrispa, 2016-09-28:
Hi @Pro,

this problem is known and cannot be fixed in userspace on the NS Compiler side atm. It is not even a namespace compiler issue (hence the tag is removed).

Amalgamation will only include *exported* functions in the header, hence internal function calls cannot be used by the generated code and compilations fails.

The namespace compiler cannot find a way to create a dependant-free graph. Apparently, nodes interdepend in such a way that at least some nodes will be created without parents. When that is the case, the namespace compiler can not use API code (UA_Server_addXYNode) anymore, as these require a manditory parent. Instead, the compiler will resort to bootstrapping mode, using non exported functions.

Given the way the API is written, this problem cannot be resolved otherwise. You are doing nothing wrong, but your models are created in such a way that there simply is no possible forward-only way to create the nodes. And amalgamated header simply do not include the necessary functions to get the job done.

How can you work around this?

You are obviously compiling the models into the stack, which does not require userspace APIs. So I would suggest the following workaround: Compile your shared object/DLL without amalgamation and save them. Then, recompile it with amalgamation, but without the namespace generation. You should now have a shared lib with your initialized namespace and a header usable for userspace linking.

As an alternative, you could "patch" open62541 to export the functions (adding UA_EXPORT to the headers) that are missing during linking.

Both fixes are dirty, I know, but the only solution to this quite specific problem I can think off. 

In any case, the generated code is correct and cannot be generated in any other way :/ Hence not really an error on the code generator side.

Best regards,

ichrispa

Pro, 2016-09-28:
Thanks for the work around ideas!

I'll close this issue for now.

closed 2016-09-28 by Pro