open62541/open62541 #824: Possibly wrong namespace IDs

Pro, 2016-09-28:
The pyUANamespace compiler creates a .c file with the following content:

```
/* WARNING: This is a generated file.
 * Any manual changes will be overwritten.

 */
#include "generated.h"
UA_INLINE void generated(UA_Server *server) {
UA_Server_addNamespace(server, "http://opcfoundation.org/UA/DI/");
UA_Server_addNamespace(server, "http://yourorganisation.org/project1/Types.xsd");
UA_Server_addNamespace(server, "http://yourorganisation.org/project2/Types.xsd");

do {
// Referencing node found and declared as parent: i=33/HierarchicalReferences using i=45/HasSubtype
// Node: opcua_node_referenceType_t(ns=2;i=6030), 1:ConnectsTo
UA_ReferenceTypeAttributes attr;
UA_ReferenceTypeAttributes_init(&attr);
attr.displayName = UA_LOCALIZEDTEXT("", "ConnectsTo");
attr.description = UA_LOCALIZEDTEXT("", "Used to indicate that source and target Node have a topological connection.");
UA_NodeId nodeId = UA_NODEID_NUMERIC(2, 6030);
UA_NodeId parentNodeId = UA_NODEID_NUMERIC(0, 33);
UA_NodeId parentReferenceNodeId = UA_NODEID_NUMERIC(0, 45);
UA_QualifiedName nodeName = UA_QUALIFIEDNAME(1, "ConnectsTo");
UA_Server_addReferenceTypeNode(server, nodeId, parentNodeId, parentReferenceNodeId, nodeName
       , attr, NULL, NULL);
} while(0);

// and many more
```

So it relies on the fact that the namespaces are not yet added to the server when calling `generated(server)` and then correspond to the following IDs:

- 2 = `http://opcfoundation.org/UA/DI/`
- 3 = `http://yourorganisation.org/project1/Types.xsd`
- 4 = `http://yourorganisation.org/project2/Types.xsd`

But what happens if before calling `generated(server)` some of the namespaces or any other are already added?
So the `main` method looks like this:

```
int main() {
	// initialization stuff ...
	UA_Server_addNamespace(server, "http://yourorganisation.org/project1/Types.xsd");
	// add some manually created nodes for this namespace ..
	// add the model from .xml
	generated(server)
	// ...
}
```

The namespace IDs don't correspond to the ones pyUANamespace expects:

- 3= `http://opcfoundation.org/UA/DI/`
- 2 = `http://yourorganisation.org/project1/Types.xsd`
- 4 = `http://yourorganisation.org/project2/Types.xsd`

Or another example, a completely unrelated namespace is added before, the IDs will be shifted:

- 3 = `http://opcfoundation.org/UA/DI/`
- 4 = `http://yourorganisation.org/project1/Types.xsd`
- 5 = `http://yourorganisation.org/project2/Types.xsd`

This causes an invalid generation of the model.
For a quick fix, there should be an assert in the generated .c file if the value returned by `UA_Server_addNamespace` is the expected namespace index.
For a better fix the returned index should be used for the following code.

Note: You can use my test repo for debugging this issue: https://github.com/Pro/open62541_namespaces

ichrispa, 2016-09-28:
Hi @Pro,

the namespace compiler - or rather its preprocessor - asserts the order of the namespaces and will reorder the IDs of the namespaces as required to allow all models to coexist. To achieve that, do *not* compile the models separately. Inform the namespace compiler about all the namespace models that exist (in any order), then let the preprocessor figure out how to reorder the namespace IDs. See #584 for the problem of merging namespaces from multiple sources.

If your only want a subset of the nodes to be printed, you can add the rest of the nodes to a respective ignore list and pass that with '-i ignoreTheseNodeIds.txt'. You can run the compiler multiple times with different ignore lists if you need to initialize the models separately.

I am aware of the fact that there is a multitude of modelling constellations on how one might attempt to merge various models. There is also a multitude of ways to break static code generation by adding various runtime components, such as partial models of NS0 and other antics. When models and usecases reach a certain degree of complexity, I would expect the user's expertise to be able to adapt to those circumstances, i.e. by running awk/sed over the generated, project/use-case specific code. Generating runtime-inspection code to derive/decafoumble namespace IDs and URIs imho certainly goes beyond what a code generator should include. After all, the generator is being broken once a month without such complex features.

Sorry @Pro, this one I can't fix. There is no real bug here beyond a specific use case with a mix of runtime initialization and static code generation.

Best regards,

ichrispa

closed 2016-09-28 by ichrispa

Pro, 2016-09-28:
@ichrispa ok, you handled the case of partial namespace initialization. I understand that that's quite a specific use-case which is not that common.

But what about the use-case, where someone adds another namespace before calling the generated method?

Imho it would be better to at least add a simple if check in the beginning and notify the developer of it's misbehaviour. Such errors are quite hard to find/debug.

I.E. I would suggest something like:

replace:
```
UA_Server_addNamespace(server, "http://opcfoundation.org/UA/DI/");
```
with:
```
if (UA_Server_addNamespace(server, "http://opcfoundation.org/UA/DI/") != 2) //the expected index
    // print some error message
```
or:
```
assert(UA_Server_addNamespace(server, "http://opcfoundation.org/UA/DI/") == 2);
```

If you want I can create a PR for that?

ichrispa, 2016-09-28:
Ok, that fix sound really simple. It doesn't fix anything, but at least you get a warning at startup...

Go ahead and create a PR :+1: