open62541/open62541 #825: Wrong namespace URI in pyUANamespace compiler

Pro, 2016-09-28:
I used UaModeler to create a new model which is based on NS0 and [Opc.Ua.Di.NodeSet2.xml](https://opcfoundation.org/UADevices/1.1/Opc.Ua.Di.NodeSet2.xml, called `project1.NodeSet2.xml`

In UaModeler I entered for the namespace URI:
`http://yourorganisation.org/project2/`

This generates the following .xml file:

```
<UANodeSet xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:uax="http://opcfoundation.org/UA/2008/02/Types.xsd" xmlns="http://opcfoundation.org/UA/2011/03/UANodeSet.xsd" xmlns:s1="http://yourorganisation.org/project1/Types.xsd" xmlns:s2="http://opcfoundation.org/UA/DI/Types.xsd" xmlns:ua="http://unifiedautomation.com/Configuration/NodeSet.xsd" xmlns:xsd="http://www.w3.org/2001/XMLSchema">
    <NamespaceUris>
        <Uri>http://yourorganisation.org/project1/</Uri>
        <Uri>http://opcfoundation.org/UA/DI/</Uri>
    </NamespaceUris>
    ...
```

Note:
* `xmlns:s1` is set to "http://yourorganisation.org/project1/Types.xsd" 
* `<NamespaceUris>` contains `http://yourorganisation.org/project1/`

And the generated .c file contains:
```
UA_Server_addNamespace(server, "http://yourorganisation.org/project1/Types.xsd");
```

This means that the namespace URI is wrong, i.e. it should be:
```
UA_Server_addNamespace(server, "http://yourorganisation.org/project1/");
```

This is caused by the following lines (https://github.com/open62541/open62541/blob/0.2/tools/pyUANamespace/open62541_XMLPreprocessor.py#L168)

```
    # Might be the more reliable method: Get the URI from the xmlns attributes (they have numers)
    if len(self.namespaceOrder) > 0:
     for el in self.namespaceOrder:
       if el[0] == nsId:
         return el[1]
```

The `namespaceOrder` contains the namespace with `Types.xsd`. This also causes that (https://github.com/open62541/open62541/blob/0.2/tools/pyUANamespace/open62541_XMLPreprocessor.py#L332)
```
if tgt.getReferencedNamespaceUri(1) == nsUri:
```
will never be true and thus it falls back to the stochastic method, which shouldn't happen.

If I remove the lines with `namespaceOrder` everything works as expected.

What's the purpose of the `namespaceOrder`?

ichrispa, 2016-09-28:
Within an XML nodes refer to namespace IDs implicitly referencing a model. This ID is only valid within a single XML. As an example, a new model using NS0 will always tend to call itself namespace 1.

This results in quite complex interdependencies when you use multiple models together. For example, you have three XML files, in which:

  * you use NS0, Di (which will call itself ns=2) and your own model (which would be ns=1). 
  * You want to use that with another model of your own which only uses NS0 and calls itself ns=1.  
  * A third model (ns=1) build on FDI (ns=2) and your previous DI model (ns=3).

So you have three files, each containing only its own model and referring to other models with a nsId. The challenge is to identify which model within an XML a certain nsID refers to when you attempt to merge these into a single server nodeset and make sure they don't collide.

There is no handy way to derive the information "I refer to Opc.Di.NodeSet2.xml as ns=2" from within an XML file. There are two ways to derive this information, of which I have seen both or either single one used in XML models:
 1. The namespace tag contains attributes called  ' xmlns:s1="http://yourorganisation.org/yourDeviceModel/" ', where N is the number of the namespace ID being referred to. 
 1. The XML Element 'NamespaceUris' appears to contain an ordered list of URI's just like xmlns:sN.

I have not yet encountered a model that set the xmlns attribute to an xsd file, hence the compiler assumes that the uri's given in either tag are identical (which is the case for the models I created with UA Modeller).

After deriving what model each numeric identifier refers to, the processor compares the various files and reorders them to be disjunctive (each namespace gets its own nsId). Then the nodes in each of the XMLs are being reassigned their respective unique namespace IDs before the actual namespace compiler turns them into code.

> will never be true and thus it falls back to the stochastic method, which shouldn't happen.

The preprocessor is not wrong in what it does. Your assumption is based on the premise that xmlns attributes always contain an XSD file - if that assumption does not hold true, the comparison does work. Here's how a DI based model is exported by my version of UA Modeller:

> xmlns:s1="http://yourorganisation.org/my_model_name/" xmlns:s2="http://opcfoundation.org/UA/DI/"

Note the complete absence of "xsd" references even in the DI model included by UA Modeller.

As a fix I would suggest having the NS Compiler's preprocessor removing any trailing '[:alnum:].xsd$'  from xmlns:sN attributes.

ichrispa, 2016-09-28:
@Pro: Can you rerun the preprocessor and see if that one fixes your problem?

Pro, 2016-09-28:
Thanks for your very nice explanation! I looked through the code and came to the same understanding :)

So the strange thing here is that my version of UaModeler adds the `Types.xsd` to the `xmlns:` Tags.
@ichrispa how did you include the `Opc.Ua.Di.NodeSet2.xml` into UaModeler? Which version are you using?

My Version is: 1.5.0 on Ubuntu 16.04
I added the `Di` by using the 'New Project' Wizard, Page 2 'Generate Code'  select modeling, and on Page 3 'Choose Base Model' I activated the `Di` nodeset. Then in the custom model just add a ObjectTypeNode under `DeviceType` and export the .xml
Do you maby use an older version?

I'll test your fix and let you know.

Pro, 2016-09-28:
Fixed with commit https://github.com/open62541/open62541/commit/dd29beee3d7bca9e6dbad7f297dad7d74260cbb1

closed 2016-09-28 by Pro

ichrispa, 2016-09-28:
Hi @Pro,

I am using version UA Modeller 1.3.4 on OpenSuse 42.1. I don't know if adding the 'xsd' is a strange thing, but I certainly never encountered it before. Maybe it's the way it is supposed to be and not adding it was a bug? Or I was missing some sort of extra option?

I always add DI by creating a new model, then selecting Di in "Additional models" in the wizard. Same as you. I use the same method for additional models such as FDI etc..

Anyway, it was an easy fix :)

Best regards,

ichrispa