open62541/open62541 #828: use  cleint.c   subscription error

xty19, 2016-09-29:
https://github.com/open62541/open62541/blob/33051eaed652a6b1f6d72d4ffb9f0479b6896a96/examples/client.c
the file above , i try , add the #define UA_ENABLE_SUBSCRIPTIONS

but  when  i compile it ,error make out ,
root@raspberrypi:/home/pi/open5/open62541/tests/example_project_nomethod# gcc -std=c99 open62541.c sub_client.c -o subclient_xx
sub_client.c: In function ‘main’:
sub_client.c:116:5: warning: implicit declaration of function ‘UA_Client_Subscriptions_new’ [-Wimplicit-function-declaration]
     UA_Client_Subscriptions_new(client, UA_SubscriptionSettings_standard, &subId);
     ^
sub_client.c:116:41: error: ‘UA_SubscriptionSettings_standard’ undeclared (first use in this function)
     UA_Client_Subscriptions_new(client, UA_SubscriptionSettings_standard, &subId);
                                         ^
sub_client.c:116:41: note: each undeclared identifier is reported only once for each function it appears in
sub_client.c:123:5: warning: implicit declaration of function ‘UA_Client_Subscriptions_addMonitoredItem’ [-Wimplicit-function-declaration]
     UA_Client_Subscriptions_addMonitoredItem(client, subId, monitorThis, UA_ATTRIBUTEID_VALUE,
     ^
sub_client.c:134:5: warning: implicit declaration of function ‘UA_Client_Subscriptions_manuallySendPublishRequest’ [-Wimplicit-function-declaration]
     UA_Client_Subscriptions_manuallySendPublishRequest(client);
     ^
sub_client.c:180:5: warning: implicit declaration of function ‘UA_Client_Subscriptions_remove’ [-Wimplicit-function-declaration]
     if(!UA_Client_Subscriptions_remove(client, subId))

could you please tell me the answer,thanks

Pro, 2016-09-29:
You need to build open62541 with the CMake switch `UA_ENABLE_SUBSCRIPTIONS`, just adding the define is not enough.

I.e.:
```
cd open62541
mkdir build
cd build
cmake -DUA_ENABLE_SUBSCRIPTIONS=ON ..
make
```

If you don't disable them manually, subscriptions are enabled by default.

If that's not the problem, please reopen and provide further details how you built open62541 and where exactly you added the define.

jpfr, 2016-09-29:
Hum.. it should be technically possible to enable / disable subscriptions just by setting the macro flag at the beginning of open62541.h.

Stasik0, 2016-09-29:
@jpfr: I think the cmake switch affects the amalgamated files, so @Pro has got the point. The reason was to keep amalgamated files as small as possible...

jpfr, 2016-09-29:
We currently include all source files in the amalgamation and switch code off with compiler switches (also in .c source files). And yes, there was more cmake magic in the past.

The upside is the single-file distribution is one-size-fits all with a single switch in the header.
The downside is the increased file size. But that's unproblematic for sources imho.

xty19, 2016-09-29:
@Pro thanks i will try later and tell you , the define  where and how i use

xty19, 2016-09-29:
the original code link is :
https://github.com/open62541/open62541/blob/33051eaed652a6b1f6d72d4ffb9f0479b6896a96/examples/client.c

the modified code is below

i define it in the header     UA_ENABLE_SUBSCRIPTIONS

/* This work is licensed under a Creative Commons CCZero 1.0 Universal License.
 * See http://creativecommons.org/publicdomain/zero/1.0/ for more information. */

#ifdef UA_NO_AMALGAMATION
# include "ua_types.h"
# include "ua_client.h"
# include "ua_client_highlevel.h"
# include "ua_nodeids.h"
# include "ua_network_tcp.h"
# include "ua_config_standard.h"
#else
# include "open62541.h"
# include <string.h>
# include <stdlib.h>
#endif

#include <stdio.h>
//i define in the header 
#define UA_ENABLE_SUBSCRIPTIONS

static void handler_TheAnswerChanged(UA_UInt32 monId, UA_DataValue *value, void *context) {
    printf("The Answer has changed!\n");
    return;
}

static UA_StatusCode
nodeIter(UA_NodeId childId, UA_Boolean isInverse, UA_NodeId referenceTypeId, void *handle) {
  UA_NodeId *parent = (UA_NodeId *) handle;

  if(!isInverse) {
    printf("%d, %d --- %d ---> NodeId %d, %d\n",
           parent->namespaceIndex, parent->identifier.numeric,
           referenceTypeId.identifier.numeric, childId.namespaceIndex,
           childId.identifier.numeric);
  }
  return UA_STATUSCODE_GOOD;
}

int main(int argc, char *argv[]) {
    UA_Client *client = UA_Client_new(UA_ClientConfig_standard);

    /* Listing endpoints */
    UA_EndpointDescription* endpointArray = NULL;
    size_t endpointArraySize = 0;
    UA_StatusCode retval = UA_Client_getEndpoints(client, "opc.tcp://localhost:16664",
                                                  &endpointArraySize, &endpointArray);
    if(retval != UA_STATUSCODE_GOOD) {
        UA_Array_delete(endpointArray, endpointArraySize, &UA_TYPES[UA_TYPES_ENDPOINTDESCRIPTION]);
        UA_Client_delete(client);
        return (int)retval;
    }
    printf("%i endpoints found\n", (int)endpointArraySize);
    for(size_t i=0;i<endpointArraySize;i++){
        printf("URL of endpoint %i is %.*s\n", (int)i,
               (int)endpointArray[i].endpointUrl.length,
               endpointArray[i].endpointUrl.data);
    }
    UA_Array_delete(endpointArray,endpointArraySize, &UA_TYPES[UA_TYPES_ENDPOINTDESCRIPTION]);

    /* Connect to a server */
    /* anonymous connect would be: retval = UA_Client_connect(client, "opc.tcp://localhost:16664"); */
    retval = UA_Client_connect_username(client, "opc.tcp://localhost:16664", "user1", "password");
    if(retval != UA_STATUSCODE_GOOD) {
[... 195 lines cut ...]
#endif

    UA_Client_disconnect(client);
    UA_Client_delete(client);
    return (int) UA_STATUSCODE_GOOD;
}

come out error code is :
oot@raspberrypi:/home/pi/open5/open62541/tests/example_project_nomethod# gcc -std=c99 open62541.c sub_clientx2.c -o subclient_xx2
sub_clientx2.c: In function ‘main’:
sub_clientx2.c:109:5: warning: implicit declaration of function ‘UA_Client_Subscriptions_new’ [-Wimplicit-function-declaration]
     UA_Client_Subscriptions_new(client, UA_SubscriptionSettings_standard, &subId);
     ^
sub_clientx2.c:109:41: error: ‘UA_SubscriptionSettings_standard’ undeclared (first use in this function)
     UA_Client_Subscriptions_new(client, UA_SubscriptionSettings_standard, &subId);
                                         ^
sub_clientx2.c:109:41: note: each undeclared identifier is reported only once for each function it appears in
sub_clientx2.c:115:5: warning: implicit declaration of function ‘UA_Client_Subscriptions_addMonitoredItem’ [-Wimplicit-function-declaration]
     UA_Client_Subscriptions_addMonitoredItem(client, subId, monitorThis, UA_ATTRIBUTEID_VALUE,
     ^
sub_clientx2.c:120:5: warning: implicit declaration of function ‘UA_Client_Subscriptions_manuallySendPublishRequest’ [-Wimplicit-function-declaration]
     UA_Client_Subscriptions_manuallySendPublishRequest(client);
     ^
sub_clientx2.c:165:5: warning: implicit declaration of function ‘UA_Client_Subscriptions_remove’ [-Wimplicit-function-declaration]
     if(!UA_Client_Subscriptions_remove(client, subId))

xty19, 2016-09-30:
I refer this https://github.com/open62541/open62541/issues/302
cmake -DBUILD_EXAMPLESERVER=On -DBUILD_EXAMPLECLIENT=On -DENABLE_AMALGAMATION=On -DENABLE_METHODCALLS=On -DENABLE_SUBSCRIPTIONS=On -DENABLE_AMALGAMATION=On -DENABLE_GENERATE_NAMESPACE0=On -DENABLE_ADDNODES=On ../
cmake -DUA_BUILD_EXAMPLESERVER=ON -DUA_BUILD_EXAMPLECLIENT=ON -DUA_ENABLE_AMALGAMATION=ON -DUA_ENABLE_METHODCALLS=ON -DUA_ENABLE_SUBSCRIPTIONS=ON -DUA_ENABLE_AMALGAMATION=ON -DUA_ENABLE_GENERATE_NAMESPACE0=ON -DUA_ENABLE_ADDNODES=ON ../
I use cmake - many parameters ,but it still does not work.  cant not compile the subscription instance,why???

xty19, 2016-09-30:
@Pro @jpfr @Stasik0

Pro, 2016-09-30:
Put the define `#define UA_ENABLE_SUBSCRIPTIONS` as the really first line before any other `#include` and please try again.
But as already mentioned this should not be needed because if you enable the corresponding CMake switch, it will be set in the `open62451/include/ua_config.h`.
Can you post the content of that file in your build folder (note: Not the `ua_config.h.in` but the generated `ua_config.h`) 

Here on github please use the buttons above the text area to format your input, e.g. for code postings select the lines and press the 'Code' button :)

xty19, 2016-09-30:
@Pro ok ，i put the define at the first side of open62541.h and it go through compile!!! and i test now

xty19, 2016-09-30:
@Pro  but when i run the client exe ,i found that. when i changed the variable it monitored ,but the handler_TheAnswerChanged does not run and print the log??？

xty19, 2016-09-30:
the client.c file link is https://github.com/open62541/open62541/blob/33051eaed652a6b1f6d72d4ffb9f0479b6896a96/examples/client.c

i add some circulation code.
`
after this code
UA_Client_Subscriptions_manuallySendPublishRequest(client);
 while(1)
{sleep(3);}
`

closed 2016-10-12 by xty19