open62541/open62541 #833: Use of UA_SessionManager with 'amalgamated' open62541.c/.h files

caspervanzoest, 2016-10-03:
Dear Community,

I am working on my own OPC server using the Open62541 API.
The program I am making has the following requirements:
- 1 user account with full read/write rights.
- 1 user acount with only read rights.

In order to implement this feature, I need to store the sessions of the accounts,
or something to distinguish in order to see wich client-user (session) is trying to read or write a node, right?

Unfortunately, The UA_SessionManager and it's methods are declared in the open62541.c file,
so they are unavailable when including 'open62541.h' in my .cpp files.

This leaves me with the following questions:
- What are the possibilities of distinguishing different users when they call my OnRead method. (or any other server method) ?
- Is the use of UA_SessionManager possible when using the open62541.h file?

Sorry for my 'noobness'.
Any suggestions are welcome!

Thanks in advance.

Regards,
Casper

jpfr, 2016-10-04:
> What are the possibilities of distinguishing different users when they call my OnRead method. (or any other server method) ?

This is currently not possible. See here for the definition of the callback.
https://github.com/open62541/open62541/blob/0.2/include/ua_server.h#L653

> Is the use of UA_SessionManager possible when using the open62541.h file?

The session manager is internal and has no public API. That's why it is not defined in open62541.h.

We were working out very fundamental features until recently.
Security / Encryption / Role Handling will be added in the 0.3 release cycle.

Would the following work for you?
- Callback to userland during CreateSession. The user can add custom user data to the session based on the login credentials
- The custom user data is given as part of method call and data source callbacks
- User-defined callbacks for node rights management (takes the nodeid, custom session data plus the operation in question)

We can add support for that rather quickly on the master branch. But that means you'd have to work with an unstable branch. Adding it to 0.2 is not possible as it would break the frozen API.

caspervanzoest, 2016-10-04:
Thank you for your comment.

The following solution would be great for my use:
> - The custom user data is given as part of method call and data source callbacks

Having an extra paramater on the Data-source callback, like Username or UserID would be great.

Current method or Data-source-callback:
`          UA_StatusCode (*read)(void *handle, const UA_NodeId nodeid,
                            UA_Boolean includeSourceTimeStamp, const UA_NumericRange *range,
                            UA_DataValue *value);
`

Would be great to have (write method as well):
`            UA_StatusCode (*read)(void *handle, const UA_NodeId nodeid,
                            UA_Boolean includeSourceTimeStamp, const UA_NumericRange *range,
                            UA_DataValue *value, UA_String username);
`

Note that anything to distinguish a user or session is good to me here,
so it does not have to be username, it might be something else to identify an 'user' or 'role'.

Thank you!

lpoujoulat, 2016-10-25:
I have a similar use case where I'd like to change the returned values according to the connection. This would fit my need. Having per session custom data that is provided to the other API callbacks as you proposed would be perfect.

caspervanzoest, 2016-10-27:
@jpfr No rush, but do we have someone available to implement the solution as you described?
Thanks in advance.

jpfr, 2016-10-31:
Hey @caspervanzoest,

that can be implemented in a matter of hours.

However, we will have to change the signature of the method and datasource callbacks.
If you need the feature fast, we can do it on the master branch.

However, I'd like the 0.2/master not diverge too much before we release 0.2.
Otherwise, it will become really difficult to back-port fixes on 0.2 that are still coming in.

caspervanzoest, 2016-10-31:
I see.
If you could send me the changed open62541.c and open62541.h to me,
including the features as described here, we could leave the branches as it is?

jpfr, 2016-10-31:
lol.
Developing a feature without adding it to a branch?

This is an effort to bundle capacity for a common need amongst the involved partners.
Open source projects are not a place where you find free capacity for your project-specific development.
Meaning, we only invest effort if it has a long-term benefit.

That being said, I might find a free slot end of this week.

jpfr, 2016-11-04:
Implemented in 860b800b82b5f8b60fa4cd70611897eaa3b510c8 on master.
You now get a session handle in the method callback. The session handle is a pointer you can assign during ActivateSession.

Feel free to add the session handle to the datasource callbacks and make a PR.

lpoujoulat, 2016-11-04:
Thank you very much.

caspervanzoest, 2016-11-04:
Thank you! 
I will keep you up to date on this,
I suggest to leave this issue open for more input.

jpfr, 2016-11-04:
@caspervanzoest 

> What are the possibilities of distinguishing different users when they call my OnRead method. (or any other server method) ?

Is it okay when you provide your custom `userAccessLevel` function instead of pushing the sessionHandle into the data source callback?

Checking the userAccessLevel for reads is still pending but straightfoward to implement.

caspervanzoest, 2016-11-04:
> Is it okay when you provide your custom userAccessLevel function instead of pushing the sessionHandle into the data source callback?

@jpfr 

I'm sorry, but I do not understand your question completely. (My English isn't very good) 
Are you suggesting to make my own method instead of using the existing data source methods?

Thanks for clarifying.

jpfr, 2016-11-04:
Currently, the data source callback is defined as
```c
UA_StatusCode (*read)(void *handle, const UA_NodeId nodeid,
                      UA_Boolean includeSourceTimeStamp,
                      const UA_NumericRange *range, UA_DataValue *value);
```

I suggest that we leave the signature as-is.

In order to implement access control, you can define your own implementation of
```c
UA_Byte (*getUserAccessLevel)(const UA_NodeId *sessionId, void *sessionHandle,
         const UA_NodeId *nodeId);
```

The functions returns the read-rights of the session on the node.
That is currently done with a trivial implementation in /plugins/ua_accesscontrol_default.c.
Then, we can use "proper" checking of user rights before calling the data source.

caspervanzoest, 2016-11-04:
I see. I will give it a try.
Thanks again.

G-Bossard, 2017-01-09:
Dear community,

First of all, I would like to say that you did a great job with this library !
I started using the 0.2 version, then I need some feature than have been implemented in master, so I move to it (session handle ...).
The UA_AccessControl is a very nice improvement.

I'm using open62541 in a **C++** application, and in my use case a small feature is missing in UA_AccessControl : 
A handle to pass to 'activateSession()' from 'struct UA_AccessControl' that will allow me to pass a object which manage user to 'activateSession' callback : usefull in OO (c++) or if multiple instance of UA_server coexist in the same application.

I did it in my local version of open62541.c/.h.
I suppose it's not a big work to move to master and that could be useful for others ...
I propose a patch (I'm not familiarly with git/github yet) : 
[open62541.diff.zip](https://github.com/open62541/open62541/files/693168/open62541.diff.zip)

**What do you think about it ?**

Some idea around that (use of open62541 in C++ context) : 
- Maybe this handle could be pass to the others callback of 'struct UA_AccessControl' ?
- Maybe the handle should not be in 'struct UA_AccessControl' but in 'UA_server' and pass to all callback ?

Sorry for my English.
Thanks in advance.
Regards,

Guillaume

jpfr, 2017-01-09:
Hello Guillaume,

welcome to this platform!

Adding the handle is a good idea.
And yes, adding the handle to the server would be even more general / useful.

I'm a bit short on time. So this is up for grabs for somebody to convert the patch into a Github Pull Request and move the handle into the general server config.
Maybe somebody wants to wet his feet as a developer.

jpfr, 2017-01-09:
And thank you @G-Bossard for the contribution. 😄

wollud1969, 2017-05-17:
Hello all together,

last message in this thread is from January, is this thread still "hot" or are the topics mentioned here are all solved in between?
I ask because I'm working on a somewhat similar problem and simply modified the signature of (*read) and (*write) in the UA_DataSource structure to also have the session pointer:

```C
    UA_StatusCode (*read)(void *handle, const UA_Session *session,
                          const UA_NodeId nodeid,
                          UA_Boolean includeSourceTimeStamp,
                          const UA_NumericRange *range, UA_DataValue *value);
```    
```C
    UA_StatusCode (*write)(void *handle, const UA_Session *session,
                           const UA_NodeId nodeid,
                           const UA_Variant *data, const UA_NumericRange *range);
```

To have `UA_Session` available in the userland I put a forward declaration in `include/ua_server.h`:
```C
struct UA_Session;
typedef struct UA_Session UA_Session;
```

This modification propagations through several files, however, in the end I've the session pointer in my callbacks and can perform fine-grained checks based on the session before executing actual read or write operations for authorization purposes.

If this is interesting for anyone else I could file a pull request.

Cheers,
Wolfgang

wollud1969, 2017-05-18:
Forget my last comment, the accesscontrol stuff in the master branch fits perfect for me needs.

wollud1969, 2017-06-09:
Concerning an earlier message here:

> In order to implement access control, you can define your own implementation of
> 
> UA_Byte (*getUserAccessLevel)(const UA_NodeId *sessionId, void *sessionHandle,
>          const UA_NodeId *nodeId);
> 
> The functions returns the read-rights of the session on the node.
>  That is currently done with a trivial implementation in /plugins/ua_accesscontrol_default.c.
>  Then, we can use "proper" checking of user rights before calling the data source.

This `getUserAccessLevel` callback is only considered when writing something into a node, I did I got something wrong? I found it only in `copyAttributeIntoNode`.
There is no authorization support for read access to nodes, right? Are there already plans to add it?

Cheers, Wolfgang

jpfr, 2017-06-16:
@wollud1969 
See https://github.com/open62541/open62541/pull/1077

More thorough access control checking will be added in the 0.3 dev cycle.

jpfr, 2018-09-01:
This was solved in the AccessLevel plugin API for 0.3.
You can now add a user-defined pointer to sessions during creation.
This is forwarded to the callbacks using the session.

closed 2018-09-01 by jpfr