open62541/open62541 #834: Compilation failure: 'malloc.h' not found

escherstair, 2016-10-03:
I downloaded the single pair of [header](https://github.com/open62541/open62541/releases/download/v0.2.0-RC1/open62541.h) and [source](https://github.com/open62541/open62541/releases/download/v0.2.0-RC1/open62541.c) files and I tried to compile the client example using a compiler which is a porting of CLANG 3.3 under Windows 7.
The compilation fails because of <malloc.h> inclusion.

I'm not an expert, but I did a quick research and it seems that <malloc.h> is somehow not-C-standard, but linux-specific.
Is it possible to compile the library under Windows with a standard-C compiler?

caspervanzoest, 2016-10-03:
Try this:
`Refer to 'std::malloc'`
or 
`#include <stdlib.h>`

caspervanzoest, 2016-10-03:
Include `<stdlib.h>` instead of `malloc.h`.

escherstair, 2016-10-03:
It's quite strange, because in open62541.c I find
#include <stdlib.h> // malloc, free
#ifdef _WIN32
# include <malloc.h>
#endif
and so it seems that including <malloc.h> is useless (because <stdlib.h> is already included in the line above).
But if I delete the <malloc.h> inclusion the compilation fails with a different error:
implicit declaration of function '__alloca' is invalid in C99 (expanded from macro 'UA_alloca')

jpfr, 2016-10-03:
> The compilation fails because of inclusion.

Inclusion of what? Can you give the error messages?
Output in the command line can be redirected to a file also in Windows, for example `dir > files.txt` to get a list of file names.

open62541 compiles fine on Windows with gcc (mingw), cygwin and MS Visual Studio.
Usually, clang aims for compatibility with gcc.

>  implicit declaration of function '__alloca' is invalid in C99 (expanded from macro 'UA_alloca')

Just redefine with the builtin alloca name of clang in your distribution.

escherstair, 2016-10-04:
First of all, thank you to everyone for your quick support. This is the first time I post to github so I didn't realize that if I don't include the code sections inside "code" some of the words are not shown.
I edited my first message to fix this problem. Sorry.

The original failure came from `<malloc.h>` inclusion.
`fatal error: 'malloc.h' file not found`
As far as I know malloc.h header is not available (and not necessary) in clang.
I fixed the error adding
```
#ifndef __clang__
# include <malloc.h>
#endif
```

For the implicit declaration of '__alloca' the solution for clang is to apply the same redefinition valid for gcc
```
# if defined __GNUC__ || defined __clang__
#  define UA_alloca(size) __builtin_alloca (size)
# elif defined(_WIN32)
...
```
In this way I was able to fix the build errors

jpfr, 2016-10-04:
Its strange you need
```
#ifndef __clang__
# include <malloc.h>
#endif
```
For the following table, clang exports the `_WIN32` macro on windows. So malloc.h should be included anyway.
http://nadeausoftware.com/articles/2012/01/c_c_tip_how_use_compiler_predefined_macros_detect_operating_system

The latter redefinition is now applied to the 0.2 branch.

closed 2016-10-04 by jpfr

escherstair, 2016-10-04:
To compile on clang, I need to skip `malloc.h`, for this reason I add "if not defined __clang__".
You're right, clang exports `_WIN32` macro and so the code searches for `malloc.h`, which is not available on clang.

escherstair, 2016-10-05:
After a deeper investigation, seems that the malloc.h header file is not one of ANSI C standard header files, but it is associated with the Microsoft C Run-Time Libraries (CRT). Probably under Windows it is available if you compile using MS Visual Studio, but it's not if you compile using CLANG.
I think that the fix for the original issue of this thread ('malloc.h' not found) is to skip the inclusion of malloc.h if CLANG is used.
I know that this thread has been marked as closed, but do you think it would be possible adding a new commit with
```
#ifndef __clang__
# include <malloc.h>
#endif
```
to skip the inclusion if CLANG is used?

Thanks in advance

commit 1e883d19 (2016-10-04): fix: alloca with clang on windows (fixes #834)
commit dfc313f7 (2016-10-17): fix: do not include malloc.h in clang (fix #834)