open62541/open62541 #835: Dependency ordering in namespace compiler

Pro, 2016-10-05:
The pyUANamespace compiler does not yet order the nodes based on their dependency, i.e., it just prints out the order from the XML file.

This causes the resulting source code to fail, e.g., for Namespace0 the XML contains on line 1183:

```
<UAObject NodeId="i=2009" BrowseName="ServerCapabilities" ParentNodeId="i=2004">
    <DisplayName>ServerCapabilities</DisplayName>
    <Description>Describes capabilities supported by the server.</Description>
    <References>
      <Reference ReferenceType="HasProperty">i=3086</Reference>
      <Reference ReferenceType="HasProperty">i=3087</Reference>
      <Reference ReferenceType="HasProperty">i=3088</Reference>
      <Reference ReferenceType="HasProperty">i=3089</Reference>
      <Reference ReferenceType="HasProperty">i=3090</Reference>
      <Reference ReferenceType="HasProperty">i=3091</Reference>
      <Reference ReferenceType="HasProperty">i=3092</Reference>
      <Reference ReferenceType="HasComponent">i=3093</Reference>
      <Reference ReferenceType="HasComponent">i=3094</Reference>
      <Reference ReferenceType="HasTypeDefinition">i=2013</Reference>
      <Reference ReferenceType="HasModellingRule">i=78</Reference>
      <Reference ReferenceType="HasComponent" IsForward="false">i=2004</Reference>
    </References>
  </UAObject>

```

Which has the referene `HasTypeDefinition=2013`, but the type definition for `i=2013` is on line 1794:
```
  <UAObjectType NodeId="i=2013" BrowseName="ServerCapabilitiesType">
    <DisplayName>ServerCapabilitiesType</DisplayName>
    <Description>Describes the capabilities supported by the server.</Description>
    <References>
      ...
    </References>
  </UAObjectType>
```

So the namespace initialization fails and creates an invalid information model.

@ichrispa, @Stasik0 
Is there already some code which builds the dependency graph and makes sure the order is correct?
If not, does it make sense to implement a reordering of the nodes based on the dependency graph.
The following fields indicate dependencies between nodes:
- ParentNodeId
- References

ichrispa, 2016-10-11:
Hi @Pro,

> The pyUANamespace compiler does not yet order the nodes based on their dependency, i.e., it just prints out the order from the XML file.

That statement is just plain wrong (sry Pro). We are using node order deriviation.

The problem you found is "new". So far, we only needed to create individual nodes and references (that's all what's described by the XML files). Since a couple of commits ago, we use instantiation. Someone deemed that if an instatiation failes, the entire node creation failes. Which is where your issue comes in.

The namespace compiler *does* check for subtypes and typedefinitions and weighs them. But so far, it was not necessary to assign a high weight to them. 

Since now we actually mandate that the entire typedef must exist when creating a node, that was changed. Hence, I have increased the significance of type/subtype references by an order of 100. With f6e79fb994644228334f1b9cdc2eb8506f4bfc90 the "type must exist for node to be created" problem should be fixed.

Best regards,

ichrispa

closed 2016-10-11 by ichrispa

Pro, 2016-10-11:
@ichrispa 

> That statement is just plain wrong (sry Pro). We are using node order deriviation.

After digging further into the code I found that part too :+1: . Thanks for the clarification anyways!